ljalves/linux_media #63: What OS to avoid 'tainted kernel' messages and no driver kernel loading

750430, 2015-01-04:
Hi there,

I wonder if you can help?

I had (!) a Mythbuntu system and tried to implement these drivers but got a message in dmesg about the kernel being tainted and these drivers being unsigned.

I found some build instructions which mentioned Mageia Linux but I have way more experience with Debian based systems and very very little with some seemingly esoteric spin-off of Mandriva.

I tried with a fresh build of Ubuntu 14.10 and I installed a vanilla kernel using this guide (http://mapopa.blogspot.co.uk/2009/01/compiling-2.html) and still got the same error.

So, I wondered what you would suggest using. Ideally Debian flavoured. And if you have any thoughts on what OS to use. I'd like to use Myth again but would be happy to give TV Headend a whirl.

Many thanks,
Jonathan

trsqr, 2015-01-04:
Can you paste the actual error here?

750430, 2015-01-04:
Hmm, sorry, no, I trashed the OS.

After a: lspci -v | grep --after-context=10 7160 
I got:Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
       Subsystem: Device 6280:0011
       Flags: bus master, fast devsel, latency 0, IRQ 18
       Memory at e0300000 (64-bit, non-prefetchable) [size=1M]
       Capabilities: [40] MSI: Enable- Count=1/32 Maskable- 64bit+
       Capabilities: [50] Express Endpoint, MSI 00
       Capabilities: [74] Power Management version 2
       Capabilities: [80] Vendor Specific Information: Len=50 <?>
       Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
       Kernel driver in use: SAA716x Budget

But where i expected to see Kernel module, I see no entry. 

Looking at dmesg I saw:
WARNING: You are using an experimental version of the media stack.
        As the driver is backported to an older kernel, it doesn't offer
        enough quality for its usage in production.
        Use it with care.

And I'm pretty confident I saw: 
module verification failed: signature and/or required key missing - tainting kernel

The vanilla kernel i updated to was 3.18 and when i ssh'd into the box, i could see that's what was being used.

what distro do you use and how did the installation of the open source drivers go?

many thanks
Jonathan

ljalves, 2015-01-06:
As far as I know "tainting kernel" means that you are loading closed source drivers in the kernel.
It should be completely unrelated to the linux distro you use.

closed 2015-01-06 by ljalves