ljalves/linux_media #64: TBS6285 no signal

epon, 2015-01-04:
Hi,

With a bit of effort I managed to get the drivers built for the TB2 6285. However, when I start TVHeadend 3.4.27 I can see all the adapters and it will appear to tune to a frequency, but it doesn't find any channels. I've tried both int_type=0 and int_type=1 without success.

The firmware is in /lib/firmware and is taken from bas-t's repository.

I'm using the latest commit from both repositories.

Using the official drivers I can get scan successfully, but they cause my server to crash after a day or so even using int_type=1 (MSI/MSI-X).

Any ideas?

RayCic, 2015-01-04:
Unfortunately it looks driver is incapable to correctly initialize some versions (all?) of card.

I had same problem and I fixed it by addiding following commands to boot scripts:
```
/usr/bin/dvb-fe-tool -a 0 -d DVBT
/usr/bin/dvb-fe-tool -a 1 -d DVBT
/usr/bin/dvb-fe-tool -a 2 -d DVBT
/usr/bin/dvb-fe-tool -a 3 -d DVBT
```
You should change numbers with your adapters numbers

Theoretically you can use any program capable of _FULL_ initialization of card, for example w_scan.

BTW for statistics can you post you revision of card (mine is V21, it is written on front side of card)

750430, 2015-01-04:
Hi, sorry for the thread hijack, what distro did you use to successfully build the drivers? When you run a : lspci -v | grep --after-context=10 7160  do you see 'kernel module' loaded?

ljalves, 2015-01-15:
@epon 
You can paste dmesg. Are you sure the firmware is located in the right place and you have the correct versions?
The ideia of having to do a "full initialization of the card" is weird and should not be necessary.

epon, 2015-01-17:
Since issue #66 describes the same problem I'm seeing in this issue, I think we should close this one down and mark it as a duplicate of #66 if that's possible. Better to keep the information in one place.

ljalves, 2015-01-28:
Closing and marking duplicate of #66

closed 2015-01-28 by ljalves