ljalves/linux_media #65: TBS 6285 open source driver with other cards

epon, 2015-01-04:
Hi,

I have two TV cards in my system and noticed that when you build the open source driver for the TBS 6285 you also end up building and installing the drivers for all other V4L cards.

The problem I've got is the driver that's built for my WinTV-NOVA-HD-S2 (Conexant Systems, Inc. CX23880) card is very unstable and in fact won't load up without crashing when using TVHeadend. The in-Kernel driver however is fine.

My quesiton is how can I get both of these cards working on my system simultaneously?

RayCic, 2015-01-04:
Welcome to the dark side of the Source! I share your pain.

Short answer: you can not
Long answer: we should wait when some body add TBS6285 driver to main kernel (unfortunately it may never happen)

ant-thomas, 2015-01-04:
Are you suffering with similar issues to mine with the TBS 8920? (https://github.com/ljalves/linux_media/issues/52)

I've currently got my 2x8920s disabled in TVH

epon, 2015-01-05:
Well it looks like it's back to the unstable "official" drivers.

There's no winning with this card...

trsqr, 2015-01-05:
Indeed. I would not recommend TBS cards at the moment...

However, this is what Luis and the other contributors are doing:

- they take the kernel media tree
- they add their TBS-specific code on top of that

Thsu basically everything supported by the kernel media_tree should be supported by these as well. So if these drivers are not stable and the kernel drivers are, then it's a question if these have changed something that impacts your card in a negative way. I can see that Luis has recently updated his drivers from the main kernel media-tree. Is your WinTV-NOVA stable with the latest media_build?

epon, 2015-01-05:
>> Indeed. I would not recommend TBS cards at the moment...

Unfortunately the bulk of "decent" channels (esp HD channels) in the UK are only available on DVB-T2 based Freeview-HD so there's very little other choice for cards :( The New Hauppauge HVR-2205 doesn't have *any* support for Linux.

>>  Is your WinTV-NOVA stable with the latest media_build?

Building the driver from media_build @ (SHA-1) 37d29613f4e5ca397090c77fc559476c5494c745 and as soon as I launch TVHeadend I get all sorts of problems displayed in dmesg.

The in-Kernel driver@ 3.16.5 however is very stable.

trsqr, 2015-01-05:
DVBSky T982 is a dual tuner DVB-T2 card supported by standard kernel drivers. There's also DVBSky T9580 (1x DVB-T2/C + 1x DVB-S2), DVBSky T980C and TechnoTrend CT2-4500 (both 1x DVB-T2/C + CI slot) that are supported. There are also several USB DVB-T2 adapters these days, but I understand that you might not be interested in those.

HVR-2205 looks interesting. My guess is that adding support for that should be relatively easy.

However, if the standard media_build will not work for you, then there's no chance that these drivers would work either. Also: the standard media_tree is what's going to be distributed with the Linux kernel in future, so unless that code is corrected, the support for that device will break in future for everyone upgrading to more recent kernels.

epon, 2015-01-05:
Thanks Olli,

I'll have a look at those options.

Cheers.

epon, 2015-01-05:
>> DVBSky T982 is a dual tuner DVB-T2 card supported by standard kernel drivers

So by that do you mean it's supported in-Kernel? If so, do you know which version of the Kernel has support? Does anyone have any experience of using this card with TVHeadend?

trsqr, 2015-01-05:
The manufacturer submitted a patch to add support in November: http://www.spinics.net/lists/linux-media/msg83096.html

This is in kernel 3.19: http://www.mail-archive.com/linux-media@vger.kernel.org/msg82976.html 

Use media-build to compile drivers for earlier kernels as usual...

Last question I cannot answer. :)

epon, 2015-01-05:
Thanks again - really helpful :)

bas-t, 2015-01-05:
I've got a couple of those DVBSky T982 adapters, working nice for me. However, I'm using MythTV/FFdecsawrapper.

epon, 2015-01-05:
The problem I've got with those is they don't have a passthrough. I currently have a requirement for four tuners and so would need to buy two of these cards and set up some kind of two way distribution amplifier.

I really wish all card manufacturers would switch to using F connectors now instead of Belling-Lee.

bas-t, 2015-01-05:
Yeah, you're right, but that is the only flaw I discovered.
@trsqr :
are you willing to maintain the status of DVBSky T982 as "working very nicely"?
If so, I can send you one. (donate)

trsqr, 2015-01-05:
@epon, if the signal level is enough, just split your signal into four with a simple passive 1-to-4 splitter? After all, the signal has to be split somewhere - be it inside or outside the card. Of course on the card there's a chance it is amplified or so...

@bas-t Thanks for the offer, but I do this for my own enjoyment and I'd prefer not to make too much commitments here. After all, I'd feel bad if you were saying it's broken and I'd have my hands full already with something else or just didn't feel like spending time ironing out the bug from the code...

bas-t, 2015-01-05:
@trsqr OK, I understand.

trsqr, 2015-01-05:
Anyway, I've got already a T9580 which is ~half the same card as yours using pretty much the same code, so at least the I can keep the other half of your T982 working. :D

And seriously, the other tuner on T982 is just copy-paste of the first one with slight changes. Besides, I've understood that now that DVBSky have upstreamed the code to Linux kernel they'll terminate the proprietary drivers and will provide support of the open-source ones.

bas-t, 2015-01-05:
Ok, so DVBSky is supposed to maintain the driver, that's nice. And even I can copy & paste!
Thanks for your reply.

epon, 2015-01-05:
>> Besides, I've understood that now that DVBSky have upstreamed the code to Linux kernel they'll terminate the proprietary drivers and will provide support of the open-source ones.

I wish TBS would learn a lesson and do the same!

bas-t, 2015-01-05:
I guess that thy are forced to learn. If not, they will loose a substantial amount of customers eventualy.
BTW, I've got some 6285/6281 cards for sale, interested? (just kidding)

bas-t, 2015-01-06:
@epon : forgot to mention that I build the 6285 driver in-tree most times. Got it up and running with linux-3.19-rc3 atm. Maybe your WinTV-NOVA-HD-S2 will coexist with the 6285 that way.
You could try following the readme at

https://github.com/bas-t/saa716x-intree

Of course this is not ideal, but guess what, I can't upstream saa716x.

epon, 2015-01-06:
@ bas-t

Thanks for the suggestion - I might give that a try. I've already got the source for all my Kernel versions since I'm using gentoo.

ljalves, 2015-01-15:
@epon 
Like I referred in #52 the WinTV-NOVA-HD-S2 (Conexant Systems, Inc. CX23880) uses the cx88 driver that was also converted to vb2 (commit 0b6b6302d983236f8b5d6d6602b91a6d1e144896)
In the case of cx23885 @RayCic discovered that a similar commit (453afdd9ce33293f640e84dc17e5f366701516e8) also broke some cards.

@epon you told before that the in-kernel driver works, but have you tested with the latest official media tree? I would suspect you will have the same issue in the official media tree... if that is the case you should report this bug in the media mailing list.

epon, 2015-01-15:
@ljalves

Yep, I will have another attempt with these drivers and see if I can get things rolling.

Cheers.

ljalves, 2015-01-15:
@epon not with my tree, but with http://git.linuxtv.org/cgit.cgi/media_tree.git/tree/

epon, 2015-01-15:
@ljalves

Yep, that's what I meant. I was able to get your driver to run but I ran into the issues being discussed in #66

epon, 2015-01-16:
@ljalves Out of interest, are you planning on getting 6285 support into the kernel anytime or would that require donations? :)

ljalves, 2015-01-22:
It would require that a significant amount of people would mail Manu Abraham (the saa716x driver author and owner) to submit it to the linuxtv media tree... :)

bas-t, 2015-01-22:
mailto: abraham.manu@gmail.com

bas-t, 2015-01-22:
Maybe even better: bomb the media_tree mailing list and CC Manu Abraham.

so mailto: linux-media@vger.kernel.org
CC abraham.manu@gmail.com
Subject: [saa716x] upstream request

bas-t, 2015-01-22:
If you lack inspiration, this is what I've sent to the mailing list and cc'd to Manu:
(but please don't jus copy this, make it a personal request..)

Hello Mr Abraham,

Thank you very much for coding the saa716x driver. It's highly appreciated!

I own a couple of TBS DVB-T/T2/C cards that use those chips, so your driver is of great value to me.
Of course official drivers from TBS suck bigtime, but fortunately Luis Alves thinks so too.
So he coded support for many TBS cards that use saa716x supported chips. And I thank him very much for doing so too.

However, at some point even Luis is likely to stop supporting this development for some reason.
That brings me to you. You're the original coder and owner of the driver.
So my questions to you are:

Would you please please please upstream your driver?
What will it take you to upstream your driver?

Maybe others can be of some assistance, take some of the workload from your back.
That is, in the event that you are willing to upstream it. Are you?

You realize of course that if you honour my humble request, you will, to say the least, enter into the top level of the Eternal Hall of Fame, that most TBS card owners will send you flowers, burn candles for you and think of you in their prayers?
And as a bonus: TBS itself will probably be 'not amused', that's nice too, isn't it?
After all, as you said: they didn't even have the common courtesy to thank you..

Well, I do! And I do it also on behalf of the vast majority of the (future) owners of saa716x supported chips driven DVB adapters.

Kind regards,

Tycho Lürsen.

ljalves, 2015-05-02:
Closing this non-issue.

closed 2015-05-02 by ljalves