ljalves/linux_media #66: TBS 6285 (V21) & saa716x_budget module parameters

RayCic, 2015-01-06:
I played with TBS 6285 (V21) &  saa716x_budget module parameters.

I used following methodology:
1. set module parameters in file:  /etc/modprobe.d/my-dvb.conf
2. warm reboot system
3. run: w_scan -F -t 3
4. if w_scan did not find any channels, then rerun w_scan
5. repeat steps 2.-4. three times
6. write down results
7. goto step 1.

What I found:

CASE 1: int_type=0 verbose=any value
in this case first w_scan run did not find any transponders,
in second run it found all transponders & channels

CASE 2: int_type=1 verbose=0
in this case in first & second run w_scan found all transponders
but was unable to receive channels data

CASE 3: int_type=1 verbose>0
in this case w_scan behaves randomly:
sometimes it behaves like in CASE 2 but sometimes
it did not find any transponders on first run and
found all transponders on second run, but was unable
to receive channels data.

Conclusions:
1) TBS 6285 (V21) is usable only with module parameter:  int_type=0
2) card requires some kind of initialization by programs like w_scan or dvb-fe-tool
3) IMHO driver have timing issues, especially with parameter  int_type=1

ljalves, 2015-01-06:
A lot of issues with the TBS6285 card (#54 #61 #64 #65 and this one) are showing up.
I've initially added support to this card by remotely using someone's machine (if I remember correctly it was @bas-t machine).
As far as I know his card is still working ok.

One thing that could be happening is that TBS made a new hardware revision of that card and switched things around a bit... the first thing that comes to my mind is TS mode setting or even newer chip revisions.

Important stuff when comparing cards are:
- saa716x revision: TBS uses at least 2 revisions (rev2 and rev3) is it happens that they use different revisions for the same card series, I'll need to change the i2c code.
- hardware revision: @RayCic has v21 what other versions exist out there?

@bas-t, if possible post your card HW revision and saa716x rev.

bas-t, 2015-01-06:
@ljalves : I'll have to take a look. Anyway, I've got 2 different revisions of the board and the one that I use now (the newer one) is not the one you accessed while coding the driver.
Both revisions have different phisycal layouts.

On the newer board is printed:
V20

lspci for the newer revision says:

04:00.0 Multimedia controller [0480]: Philips Semiconductors SAA7160 [1131:7160] (rev 02)
        Subsystem: Device [6285:0001]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 32 bytes
        Interrupt: pin A routed to IRQ 32
        Region 0: Memory at f7e00000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
                Address: 00000000fee0300c  Data: 4192
        Capabilities: [50] Express (v1) Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <256ns, L1 <1us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset-
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
                LnkCap: Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
                        ClockPM- Surprise- LLActRep- BwNot-
                LnkCtl: ASPM Disabled; RCB 128 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
        Capabilities: [74] Power Management version 2
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [80] Vendor Specific Information: Len=50 <?>
        Capabilities: [100 v1] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
        Kernel driver in use: SAA716x Budget

For the older one:

Printed on the board:

V12M

lspci:

02:00.0 Multimedia controller [0480]: Philips Semiconductors SAA7160 [1131:7160] (rev 02)
        Subsystem: Device [6285:0001]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 32 bytes
        Interrupt: pin A routed to IRQ 30
        Region 0: Memory at f7e00000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
                Address: 00000000fee0300c  Data: 4162
        Capabilities: [50] Express (v1) Endpoint, MSI 00
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <256ns, L1 <1us
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset-
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 128 bytes
                DevSta: CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
                LnkCap: Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
                        ClockPM- Surprise- LLActRep- BwNot-
                LnkCtl: ASPM Disabled; RCB 128 bytes Disabled- CommClk-
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
        Capabilities: [74] Power Management version 2
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot-,D3cold-)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [80] Vendor Specific Information: Len=50 <?>
        Capabilities: [100 v1] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
        Kernel driver in use: SAA716x Budget

bas-t, 2015-01-06:
I did not use the V12M boards for a while, so just to confirm: they are still working great!

bpbastos, 2015-01-06:
Mine has v21 printed on the board.

RayCic, 2015-01-06:
Just for fun I searched Internet for pictures of TBS 6285.
I found two additional revisions: V11 & V12M

BTW how can I get revision of saa716x?
lspci show
```
06:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 02)
```
Is 'rev 02' its revision?

bas-t, 2015-01-06:
@RayCic V12M is not additional, it is the board Luis used to code the initial driver on my machine.

crazycat69, 2015-01-06:
Maybe V2.x is HW with Pericom PCI-E switch before saa7160 ? Like used in tbs 6983 - http://wxsatpicture.free.fr/tbs6925/tbs6983.htm.

lspci --n:
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor DRAM Controller [8086:0150] (rev 09)
00:1c.0 PCI bridge [0604]: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 1 [8086:1e10] (rev c4)
00:1c.1 PCI bridge [0604]: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 2 [8086:1e12] (rev c4)
00:1c.2 PCI bridge [0604]: Intel Corporation 7 Series/C210 Series Chipset Family PCI Express Root Port 3 [8086:1e14] (rev c4)
00:1c.3 PCI bridge [0604]: Intel Corporation 82801 PCI Bridge [8086:244e] (rev c4)
00:1d.0 USB controller [0c03]: Intel Corporation 7 Series/C210 Series Chipset Family USB Enhanced Host Controller #1 [8086:1e26] (rev 04)
04:00.0 PCI bridge [0604]: Pericom Semiconductor Device [12d8:2304] (rev 05)
05:01.0 PCI bridge [0604]: Pericom Semiconductor Device [12d8:2304] (rev 05)
05:02.0 PCI bridge [0604]: Pericom Semiconductor Device [12d8:2304] (rev 05)
06:00.0 Multimedia controller [0480]: Philips Semiconductors SAA7160 [1131:7160] (rev 03)

lspci -t:
-[0000:00]-+-00.0
           +-1c.2-[04-07]----00.0-[05-07]--+-01.0-[06]----00.0
           |                               \-02.0-[07]--

bpbastos, 2015-01-06:
lspci -nn

00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v3 Processor DRAM Controller [8086:0c08] (rev 06)
00:02.0 VGA compatible controller [0300]: Intel Corporation Xeon E3-1200 v3 Processor Integrated Graphics Controller [8086:041a] (rev 06)
00:03.0 Audio device [0403]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller [8086:0c0c] (rev 06)
00:14.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI [8086:8c31] (rev 05)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
00:16.3 Serial controller [0700]: Intel Corporation 8 Series/C220 Series Chipset Family KT Controller [8086:8c3d] (rev 04)
00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection I217-LM [8086:153a] (rev 05)
00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 05)
00:1b.0 Audio device [0403]: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller [8086:8c20] (rev 05)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d5)
00:1c.3 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #4 [8086:8c16] (rev d5)
00:1c.4 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 [8086:8c18] (rev d5)
00:1c.5 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #6 [8086:8c1a] (rev d5)
00:1c.6 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #7 [8086:8c1c] (rev d5)
00:1c.7 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #8 [8086:8c1e] (rev d5)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 05)
00:1f.0 ISA bridge [0601]: Intel Corporation C226 Series Chipset Family Server Advanced SKU LPC Controller [8086:8c56] (rev 05)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c02] (rev 05)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 05)
00:1f.6 Signal processing controller [1180]: Intel Corporation 8 Series Chipset Family Thermal Management Controller [8086:8c24] (rev 05)
02:00.0 Ethernet controller [0200]: Intel Corporation I210 Gigabit Network Connection [8086:1533] (rev 03)
03:00.0 PCI bridge [0604]: Pericom Semiconductor Device [12d8:2304] (rev 05)
04:01.0 PCI bridge [0604]: Pericom Semiconductor Device [12d8:2304] (rev 05)
04:02.0 PCI bridge [0604]: Pericom Semiconductor Device [12d8:2304] (rev 05)
05:00.0 Multimedia controller [0480]: Philips Semiconductors SAA7160 [1131:7160] (rev 02)
07:00.0 PCI bridge [0604]: Tundra Semiconductor Corp. Device [10e3:8113] (rev 01)
08:03.0 FireWire (IEEE 1394) [0c00]: Texas Instruments TSB43AB22A IEEE-1394a-2000 Controller (PHY/Link) [iOHCI-Lynx] [104c:8023]
09:00.0 USB controller [0c03]: Renesas Technology Corp. uPD720202 USB 3.0 Host Controller [1912:0015] (rev 02)
0a:00.0 SATA controller [0106]: ASMedia Technology Inc. ASM1062 Serial ATA Controller [1b21:0612] (rev 01)

crazycat69, 2015-01-06:
so you have pericom IC on PCB ?

bpbastos, 2015-01-06:
Don't know, how can I see if it has the pericom IC?

crazycat69, 2015-01-06:
http://wxsatpicture.free.fr/tbs6925/tbs6983_fichiers/image015.jpg

bas-t, 2015-01-06:
My V20 has a pericim IC, and it's working fine

bas-t, 2015-01-06:
But the bottom line reads: 1338GT

bpbastos, 2015-01-06:
Yes it has: http://i.imgur.com/SBZdRHH.jpg

RayCic, 2015-01-06:
@bas-t: do you use any workarounds?
For example:
* saa716x_budget module with parameter int_type=1
* card initialization with program like  dvb-fe-tool

bas-t, 2015-01-06:
I set it permanently to DVB-C like this (so no dvb-fe-tool):
sed -i 's/SYS_DVBT, SYS_DVBT2, SYS_DVBC_ANNEX_A/SYS_DVBC_ANNEX_A/' drivers/media/dvb-frontends/si2168.c

and I set:

options saa716x_budget int_type=1

Sometimes I use Antti's silabs branch as wel, but that's optional.

bas-t, 2015-01-06:
My current setup is as in the README:
https://github.com/bas-t/saa716x-intree

bpbastos, 2015-01-07:
I think I have found the reason of my problems. I had to disable intel_iommu=on from my server, the tuner is working flawless since them.

The main problem about disabling iommu is that I need it for virtualization with kvm.

RayCic, 2015-01-07:
__[pedant mode: on]__
@ljalves:
> The pericom chip is just a pci-e switch (similar function to an ethernet switch) and should be fully transparent to the kernel (no driver needed).

Pericom PCIe Switch FAQs:
__Do Pericom's packet switches require the device driver?__
Yes, Pericom's Packet Switches require the standard PCI-to-PCI Bridge device driver in order to work, which is built in most of the OS's. That is, most of the OS's automatically install the standard PCI-to-PCI Bridge device driver when a Pericom's packet switch is plugged in the system for first time.

__[pedant mode: off]__

I am trying to understand what have in common @bpbastos Intel systems and mine AMD system and how they differ from @bpbastos AMD based HP Microserver N40L.
And IMHO it is IOMMU
1. mine system have IOMMU enabled
2. supermicro x10sae have IOMMU
3. GA-B85M according to manual have IOMMU
4. AFAIK HP Microserver N40L don't have IOMMU

@bpbastos, to be 100% sure, can you post result of following command on all of your boxes (TBS 6285 card is not needed):
```
dmesg | grep -i -E 'amd.?vi|io.?mmu'
```

bas-t, 2015-01-07:
Hi, output of dmesg | grep -i -E 'amd.?vi|io.?mmu' on my backend:

[    0.411098] AMD IOMMUv2 driver by Joerg Roedel <joerg.roedel@amd.com>
[    0.411100] AMD IOMMUv2 functionality not available on this system

I guess that settles it then, but let's wait for bpbastos' reply.

Next step I guess: get this thing to work properly with  IOMMU enabled

timmydog, 2015-01-07:
For what it's worth, I have started having problems tuning into channels but on a TBS6280, using a recent-ish build of the drivers, see https://github.com/ljalves/linux_media/issues/43 for some more detail.

It also has a rev 03 SAA7160,

I can post more info and lspci output if you think that it's relevant.

bpbastos, 2015-01-07:
In my case the problem is iommu, I have disabled it and the tuner worked flawless overnight. The main problem is I can't disable iommu in this server, I need it for virtualization.

ljalves, 2015-01-07:
Please someone clarify this: Do TBS official drivers work ok with iommu enabled?
If yes it should be really easy to fix since the saa716x driver is open source in TBS package.

bpbastos, 2015-01-07:
I didn't test with TBS official drivers but I'll test and return to you ASAP.

bpbastos, 2015-01-07:
@ljalves
The TBS official driver has presented an different problem with iommu enable. It didn't stoped to get signal like unofficial driver, but tvheadend started to show a lot continuity errors and image pixelation on the client.

I don't know if the continuity errors are an common problem when using tbs-official + tvheadend.

Here are my logs:
<pre>
#cat /proc/cmdline 
BOOT_IMAGE=/boot/vmlinuz-3.16.0-28-generic root=UUID=9d11b733-b11d-41d9-85d2-1c14eb506154 ro quiet intel_iommu=on nomdmonddf nomdmonisw
</pre>

#lspci -nnvv -s 05:00
05:00.0 Multimedia controller [0480]: Philips Semiconductors SAA7160 [1131:7160] (rev 02)
	Subsystem: Device [6285:0001]
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort+ <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 67
	Region 0: Memory at f7a00000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
		Address: 00000000fee00598  Data: 0000
	Capabilities: [50] Express (v1) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <256ns, L1 <1us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset-
		DevCtl:	Report errors: Correctable+ Non-Fatal- Fatal+ Unsupported-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- UncorrErr+ FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
			ClockPM- Surprise- LLActRep- BwNot-
		LnkCtl:	ASPM Disabled; RCB 128 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
	Capabilities: [74] Power Management version 2
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [80] Vendor Specific Information: Len=50 <?>
	Capabilities: [100 v1] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
	Kernel driver in use: SAA716x TBS

Looking at dmesg I got this trace when iommu is enable	
#dmesg
[    0.581279] PCI-DMA: Intel(R) Virtualization Technology for Directed I/O
[    0.581363] ------------[ cut here ]------------
[    0.581368] WARNING: CPU: 0 PID: 1 at /build/buildd/linux-3.16.0/drivers/pci/search.c:131 pci_find_upstream_pcie_bridge+0x65/0x90()
[    0.581368] Modules linked in:
[    0.581371] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 3.16.0-28-generic #38-Ubuntu
[    0.581372] Hardware name: Supermicro X10SAE/X10SAE, BIOS 2.0a 05/09/2014
[    0.581373]  0000000000000009 ffff88040955fd30 ffffffff81781eaa 0000000000000000
[    0.581375]  ffff88040955fd68 ffffffff8106fedd ffff880408f11098 ffff880408f11000
[    0.581376]  ffff880408f11098 0000000000000000 0000000000000000 ffff88040955fd78
[    0.581377] Call Trace:
[    0.581382]  [<ffffffff81781eaa>] dump_stack+0x45/0x56
[    0.581385]  [<ffffffff8106fedd>] warn_slowpath_common+0x7d/0xa0
[    0.581387]  [<ffffffff8106ffba>] warn_slowpath_null+0x1a/0x20
[    0.581388]  [<ffffffff813e6e05>] pci_find_upstream_pcie_bridge+0x65/0x90
[    0.581391]  [<ffffffff8165154e>] intel_iommu_add_device+0x3e/0x210
[    0.581394]  [<ffffffff81646ea0>] ? bus_set_iommu+0x50/0x50
[    0.581395]  [<ffffffff81646eca>] add_iommu_group+0x2a/0x60
[    0.581398]  [<ffffffff814cf70b>] bus_for_each_dev+0x6b/0xb0
[    0.581400]  [<ffffffff81646e98>] bus_set_iommu+0x48/0x50
[    0.581402]  [<ffffffff81d9bde5>] intel_iommu_init+0x504/0x58a
[    0.581404]  [<ffffffff81d4e209>] ? memblock_find_dma_reserve+0x13c/0x13c
[    0.581405]  [<ffffffff81d4e21b>] pci_iommu_init+0x12/0x3c
[    0.581408]  [<ffffffff81002148>] do_one_initcall+0xd8/0x210
[    0.581411]  [<ffffffff81092b6d>] ? parse_args+0x21d/0x4e0
[    0.581413]  [<ffffffff81d44115>] kernel_init_freeable+0x190/0x218
[    0.581416]  [<ffffffff81774fd0>] ? rest_init+0x80/0x80
[    0.581417]  [<ffffffff81774fde>] kernel_init+0xe/0xf0
[    0.581420]  [<ffffffff81789e3c>] ret_from_fork+0x7c/0xb0
[    0.581421]  [<ffffffff81774fd0>] ? rest_init+0x80/0x80
[    0.581425] ---[ end trace 4d55fb55f6790e5d ]---

ljalves, 2015-01-07:
@RayCic,
Just 1 test missing:
int_type=1 (MSI interrupts), TBS driver, IOMMU on / off

RayCic, 2015-01-07:
@ljalves , it is first case in my findings:
1. int_type=1 (MSI interrupts), any driver (_tested with both drivers_), IOMMU on or off (_tested with IOMMU on and off_): ... 

Strange...
Due ticket #62 I installed https://github.com/bas-t/saa716x-intree
and now in case "int_type=0 (ordinary interrupts), this driver, IOMMU on"
w_scan fully works on first try.

RayCic, 2015-01-07:
Unfortunately after some time w_scan loses ability read channel data.

ljalves, 2015-01-07:
@RayCic
Sorry missed the "any driver".

On a first thought, If it works without IOMMU then the driver should fine because the dma code is seamless to the kernel... but it can also be a bad methodology of dma coding in the driver and Manu Abraham (the author) doesn't touch it for a long long time...

Then, looking at @RayCic stack dumps, it can also be an issue on the dvb core itself (but my theory above also applies).

Since it's broken for amd AND intel iommu the iommu driver is less likely to be broken.

I don't have a machine with IOMMU but I'll take a look at the saa716x dma code and see if I can do anything about it.

Does anyone have other cards (pci-e) to test on an IOMMU machine?

bpbastos, 2015-01-07:
I have an pcie intel nic card and it is working just fine in this same machine with iommu enabled.

ljalves, 2015-01-07:
@bpbastos I mean a dvb card
The test was to see if the issue is in the dvb core itself and not in the saa716x driver.

bpbastos, 2015-01-07:
Sorry @ljalves. 

If you want I can give you access to my intel box.

timmydog, 2015-01-08:
@ljalves
> Does anyone have other cards (pci-e) to test on an IOMMU machine?

I have a TBS6280 in an AMD A6-7400K (Kaveri) box that has IOMMU enabled; it's running Debian Jessie.

I'm not running the very latest code, but I couldn't get a lock in mythTV with "int_type=0 verbose=1" or "int_type=1 verbose=1". I didn't try running w_scan first, just went straight into mythTV.

Yesterday, with IOMMU switched off in the BIOS, I didn't get any transports found by w_scan or mythTV, with int_type=0 or 1 and verbose=1.

I don't get much time on the box when it's not in use, but let me know if there's anything else you'd like doing and I'll try and make time.

bpbastos, 2015-01-08:
@timmydog Could you please try to modify your cmd boot line and add amd_iommu=on and run the tests with int_type=1 again?

timmydog, 2015-01-08:
When I ran

> dmesg | grep -i -E 'amd.?vi|io.?mmu'

it reported that the IOMMU was enabled, but I had no luck getting a channel lock in mythTV with int_type=1 set in my modprobe conf file.

I'll have another go with it added to the command line though if you think it will make a difference.

bpbastos, 2015-01-08:
Just to check, vt-d was enable in BIOS when I ran my tests.  The problems with int_type=1 only happens if I add intel_iommu=on at cmd boot line.

RayCic, 2015-01-11:
I wrote question to linux-kernel mailing list about PCIe switch & IOMMU.
https://lkml.org/lkml/2015/1/6/1181

Short version of this thread:
1) PCIe switches must be transparent
2) interesting comment from Manu Abraham:
>You shouldn't use MSI mode on most systems unless you are sure that your machine works perfectly well with MSI mode.
>
>You should simply use standard interrupt mode, if you are unsure.

ljalves, 2015-01-12:
@RayCic,
I've found an issue in the saa716x driver and its dma code.
I'll fix it as soon as possible and announce here for testing.

Regards,
Luis

ljalves, 2015-01-12:
On a second look, I don't think the issue is there... I'll keep looking.

epon, 2015-01-15:
I'd be happy to help where possible as I've been using the "official" drivers with int_mode=1 and enable_ir=0 and my system will randomly freeze to the point where I have to perform a hard reset. This is far from ideal...

ljalves, 2015-01-15:
@RayCic,
Can you please test again with 54362743588bc8271e7988dd2823f818c00bf43d

RayCic, 2015-01-15:
@ljalves ,
I will do it tomorrow.

RayCic, 2015-01-17:
Test results:

CASE 1: int_type=0 verbose=any value
in this case first w_scan run did not find any transponders,
in second run it found all transponders & channels

CASE 2: int_type=1 verbose=any value
in this case first w_scan run did not find any transponders,
in second run it found all transponders but was unable to
receive  channel data.

All tests done with IOMMU: on

Environment: kernel: 3.18.1 / tree: this of 2015-01-16

IMHO differences from old tests can be explained by differences in environments

IMHO I found why w_scan is unable to find any transponder on first run: according to dmesg firmware for tuner is loaded on first run, but firmware for demod is loaded only on second run

RayCic, 2015-01-18:
IMHO problem with firmware is related to tuner and/or demod driver(s), because  https://github.com/bas-t/saa716x-intree do not have this problem and only important difference with this tree is silabs-3.19.patch which contain tuner & demod driver fixes.

ljalves, 2015-01-28:
Merged latest official tree (which should include those patches)

bas-t, 2015-01-28:
Yeah, it should indeed, but it does not. Mauro is about to merge those patches, probably today.

bas-t, 2015-01-28:
Just a heads up: when those patches get merged, they don't cleanly apply against your 'latest' branch.
I synced si2157 with upstream before merging antti's silabs branch:
https://github.com/bas-t/media_tree/commit/39bda003365032befe10aba0ac0784533f77378c

epon, 2015-02-13:
So what is the status of the 6285?

I just tried again with the latest repos, using firmware from both bas-t and OE repo with int_type=1 and int_type=0 but I only get partial success as in I can sometimes get some of the services in TVHeadend but not all. The channel information doesn't always come through either.

I'm not using IOMMU and I'm also doing a "full initialization" of the card to DVBT2 using dvb-fe-tool.

Nothing appears to be working :(

epon, 2015-02-15:
I just managed to get into my server with difficulty and found that the card I have is a V12M - so now I'm really surprised it isn't working!

ljalves, 2015-05-02:
No activity for some time here - closing this issue.
Feel free to reopen if needed.

closed 2015-05-02 by ljalves