ljalves/linux_media #67: Make fails when compiling on kernel 3.11

donaghhorgan, 2015-01-09:
Hi, I've been trying to build the drivers, but I'm hitting a problem:

    Applying patches for kernel 3.11.0-19-generic
    patch -s -f -N -p1 -i ../backports/api_version.patch
    patch -s -f -N -p1 -i ../backports/pr_fmt.patch
    patch -s -f -N -p1 -i ../backports/debug.patch
    patch -s -f -N -p1 -i ../backports/drx39xxj.patch
    patch -s -f -N -p1 -i ../backports/v3.16_netdev.patch
    patch -s -f -N -p1 -i ../backports/v3.16_wait_on_bit.patch
    patch -s -f -N -p1 -i ../backports/v3.16_void_gpiochip_remove.patch
    patch -s -f -N -p1 -i ../backports/v3.11_dev_groups.patch
    2 out of 3 hunks FAILED
    make[2]: *** [apply_patches] Error 1
    make[2]: Leaving directory `/home/donagh/donagh/media_build/linux'
    make[1]: *** [default] Error 2
    make[1]: Leaving directory `/home/donagh/donagh/media_build/v4l'
    make: *** [all] Error 2

The problem seems to be with the patch file `v3.11_dev_groups.patch`. It's supposed to patch `v4l2-dev.c`, but the patch fails because it's expecting, for instance, a line with `debug_show` but  `dev_debug_show`. Any ideas?

epon, 2015-01-10:
There's not much to work with here but if it helps I encountered this issue when running a parallel make which it doesn't like.

Try building with "MAKEOPTS=-j1" and see if this works for you.

donaghhorgan, 2015-01-10:
I tried setting MAKEOPTS to "j1", but it doesn't make any difference. Thanks for the suggestion though.

It looks to me like the patch file is expecting a different version of `v4l2-dev.c`. For instance, in the version that is supplied with this git repo (media/drivers/media/v4l2-core/v4l2-dev.c), at line 37 I have:

```
/*
 *	sysfs stuff
 */

static ssize_t index_show(struct device *cd,
			  struct device_attribute *attr, char *buf)
{
	struct video_device *vdev = to_video_device(cd);

	return sprintf(buf, "%i\n", vdev->index);
}
static DEVICE_ATTR_RO(index);

static ssize_t dev_debug_show(struct device *cd,
			  struct device_attribute *attr, char *buf)
{
	struct video_device *vdev = to_video_device(cd);

	return sprintf(buf, "%i\n", vdev->dev_debug);
}
```

However, the patch file, which is included in the official V4L media build repo (media_build/backports/v3.11_dev_groups.patch), expects this:
```
@@ -38,25 +38,24 @@
  *     sysfs stuff
  */
 
-static ssize_t index_show(struct device *cd,
-                         struct device_attribute *attr, char *buf)
+static ssize_t show_index(struct device *cd,
+                        struct device_attribute *attr, char *buf)
 {
        struct video_device *vdev = to_video_device(cd);
 
        return sprintf(buf, "%i\n", vdev->index);
 }
-static DEVICE_ATTR_RO(index);
 
-static ssize_t debug_show(struct device *cd,
-                         struct device_attribute *attr, char *buf)
+static ssize_t show_debug(struct device *cd,
+                        struct device_attribute *attr, char *buf)
 {
        struct video_device *vdev = to_video_device(cd);
 
        return sprintf(buf, "%i\n", vdev->debug);
 }
```

Because these are not compatible, applying the patch fails, which appears to be fatal. I have tried disabling the patch by editing the backports.txt file and commenting out the kernel 3.11 section, and this gets me past this point, but then `make` fails at a later point anyway (presumably because I didn't apply the required patch).

All suggestions are welcome!

bas-t, 2015-01-10:
This is an upstream issue and you should report it upstream.
Meanwhile, don't just omit the patch, update it. (and send it upstream)
If you don't know how to update the patch, pastebin your v4l2-dev.c, I can do it for you.

donaghhorgan, 2015-01-10:
Great, thanks! I can report the bug, but I don't know how to update my v4l2-dev.c. I've created a pastebin here: http://pastebin.com/negK9Rxq.

bas-t, 2015-01-10:
Your v4l2-dev.c differs in an odd way from what's expected.
However, you can try this patch, it should apply clean:

http://pastebin.com/raw.php?i=tfdQ5Hqe

bas-t, 2015-01-10:
I stand corrected, if the patch applies clean, it is not an upstream issue. In that case it is an issue of this repo.

bas-t, 2015-01-10:
Just updated the patch, sorry.
http://pastebin.com/raw.php?i=8RinXiix

bas-t, 2015-01-11:
The issue is introduced in: http://git.linuxtv.org/cgit.cgi/media_tree.git/commit/drivers/media/v4l2-core/v4l2-dev.c?id=17028cdb74bf8bb593aebe5550bc90325fa9af88

ljalves, 2015-01-12:
Yes, this errors are coming from the media_build not being updated.
I've noticed that they submitted new stuff today.
Please use the latest media_build and retry.

I might start cloning the media_build repo here to match my current media_tree.

Same think applies to #68

donaghhorgan, 2015-01-13:
Sorry, I ended up updating my kernel, borking my install and then completely reinstalling Lubuntu. I'm running 3.17 now, so this error doesn't occur anymore, although I am now affected by #68.

ljalves, 2015-01-15:
Closing this issue since it's related to media_build.
This type of errors should be reported to the linux-media mailing list (after confirming that they also occur in the official media tree).

closed 2015-01-15 by ljalves