ljalves/linux_media #68: Make fails when compiling for kernel 3.17

CvH, 2015-01-10:
After the latest merge of the media_tree i get this error if i build for kernel 3.17.7
I have tried different media_build rev with no luck. Any suggestion ?

```
CC [M]  /v4l/saa716x_ff_main.o
CC [M]  /v4l/saa716x_ff_cmd.o
/v4l/saa716x_rom.c: In function 'saa716x_eeprom_data':
/v4l/saa716x_rom.c:1070:1: warning: the frame size of 1120 bytes is larger than 1024 bytes [-Wframe-larger-than=]
 }
 ^
  CC [M]  /v4l/saa716x_ff_ir.o
  CC [M]  /v4l/radio-shark2.o
  CC [M]  /v4l/radio-tea5777.o
  CC [M]  /v4l/smiapp-core.o
  CC [M]  /v4l/smiapp-regs.o
  CC [M]  /v4l/smiapp-quirk.o
  CC [M]  /v4l/smiapp-limits.o
  CC [M]  /v4l/smsdvb-main.o
  CC [M]  /v4l/smsdvb-debugfs.o
  CC [M]  /v4l/smscoreapi.o
  CC [M]  /v4l/sms-cards.o
  CC [M]  /v4l/smsendian.o
  CC [M]  /v4l/smsir.o
  CC [M]  /v4l/bt87x.o
  CC [M]  /v4l/solo6x10-core.o
  CC [M]  /v4l/solo6x10-i2c.o
/v4l/smiapp-core.c: In function 'smiapp_get_pdata':
/v4l/smiapp-core.c:3061:3: error: implicit declaration of function 'of_read_number' [-Werror=implicit-function-declaration]
   pdata->op_sys_clock[i] = of_read_number(val + i * 2, 2);
   ^
cc1: some warnings being treated as errors
scripts/Makefile.build:257: recipe for target '/v4l/smiapp-core.o' failed
make[4]: *** [/v4l/smiapp-core.o] Error 1
make[4]: *** Warte auf noch nicht beendete Prozesse...
Makefile:1373: recipe for target '_module_/v4l' failed
make[3]: *** [_module_/v4l] Error 2
make[3]: Leaving directory '/linux-3.17.7'
Makefile:51: recipe for target 'default' failed
make[2]: *** [default] Error 2
make[2]: Leaving directory '/v4l'
Makefile:26: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory ''
make: *** [release] Fehler 2
```

bas-t, 2015-01-12:
This is introduced in: http://git.linuxtv.org/cgit.cgi/media_tree.git/commit/drivers/media/i2c/smiapp/smiapp-core.c?id=390a5fa5bd07fd683f8f8a2b2959400242e0b76f

ljalves, 2015-01-13:
Just do this until media_build gets fixed:
1) Edit v4l/.config (assuming you are right after you got the error, in media_build folder)

2) Change:
CONFIG_VIDEO_SMIAPP_PLL=m
to:
CONFIG_VIDEO_SMIAPP_PLL=n
and
CONFIG_VIDEO_SMIAPP=m
to:
CONFIG_VIDEO_SMIAPP=n

This should allow you to compile the media_tree.

CvH, 2015-01-13:
it "works" for the meantime, tx

ljalves, 2015-01-15:
Closing this issue since it's related to media_build.
This type of errors should be reported to the linux-media mailing list (after confirming that they also occur in the official media tree).

Last build log also reports that error:
http://hverkuil.home.xs4all.nl/logs/Thursday.log

closed 2015-01-15 by ljalves