ljalves/linux_media #71: Error when compiling on cubox-ipro with archlinux

JaquierM, 2015-01-14:
Hi,

I try to compiling for my cubox, but I have the following error:

/root/media_build/v4l/coda-common.c: In function 'coda_probe':
/root/media_build/v4l/coda-common.c:2085:2: error: implicit declaration of function 'devm_reset_control_get_optional' [-Werror=implicit-function-declaration]
  dev->rstc = devm_reset_control_get_optional(&pdev->dev, NULL);
  ^
/root/media_build/v4l/coda-common.c:2085:12: warning: assignment makes pointer from integer without a cast
  dev->rstc = devm_reset_control_get_optional(&pdev->dev, NULL);
            ^
cc1: some warnings being treated as errors
scripts/Makefile.build:308: recipe for target '/root/media_build/v4l/coda-common.o' failed
make[3]: *** [/root/media_build/v4l/coda-common.o] Error 1

My kernel info:
Linux alarm 3.14.27-2-ARCH #1  armv7l GNU/Linux

ljalves, 2015-01-15:
Hi @JaquierM,
Do you have your kernel compiled with CONFIG_RESET_CONTROLLER ?

ljalves, 2015-01-15:
Forget it... that config option was only introduced in kernel 3.15.
You should try to use kernel >=3.15

Also, these error are usually related to backport patching (you will also get them if you try to compile the official media tree) so you can try send a message to the linux-media mailing list.

closed 2015-01-22 by ljalves