ljalves/linux_media #72: Make errors

danielhitch, 2015-01-16:
Hi.

I have used this driver with my TBS 6982 without fault over the last few months on Ubuntu.

My machine updated to the latest firmware a couple of days ago which usually wipes out certain parts of this driver (correct??)

I tried re-installing using the wiki guidance and got to the make process.

A few errors have appeared and now make install does nothing.

Please help!

Output from make (successful files removed for clarity):

make -C /home/daniel/media_build/v4l 
make[1]: Entering directory `/home/daniel/media_build/v4l'
creating symbolic links...
make -C firmware prep
make[2]: Entering directory `/home/daniel/media_build/v4l/firmware'
make[2]: Leaving directory `/home/daniel/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory `/home/daniel/media_build/v4l/firmware'
make[2]: Nothing to be done for `default'.
make[2]: Leaving directory `/home/daniel/media_build/v4l/firmware'
Kernel build directory is /lib/modules/3.13.0-44-generic/build
make -C ../linux apply_patches
make[2]: Entering directory `/home/daniel/media_build/linux'
Syncing with dir ../../media
Patches for 3.13.0-44-generic already applied.
make[2]: Leaving directory `/home/daniel/media_build/linux'
make -C /lib/modules/3.13.0-44-generic/build SUBDIRS=/home/daniel/media_build/v4l  modules
make[2]: Entering directory `/usr/src/linux-headers-3.13.0-44-generic'
  CC [M]  /home/daniel/media_build/v4l/smiapp-core.o
/home/daniel/media_build/v4l/smiapp-core.c: In function 'smiapp_get_pdata':
/home/daniel/media_build/v4l/smiapp-core.c:3061:3: error: implicit declaration of function 'of_read_number' [-Werror=implicit-function-declaration]
   pdata->op_sys_clock[i] = of_read_number(val + i * 2, 2);
   ^
cc1: some warnings being treated as errors
make[3]: *** [/home/daniel/media_build/v4l/smiapp-core.o] Error 1
make[2]: *** [_module_/home/daniel/media_build/v4l] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-3.13.0-44-generic'
make[1]: *** [default] Error 2
make[1]: Leaving directory `/home/daniel/media_build/v4l'
make: *** [all] Error 2

ljalves, 2015-01-16:
Pleaase read last comment on #68

ljalves, 2015-01-20:
Hans has now fixed media_build.
Closing.

closed 2015-01-20 by ljalves