ljalves/linux_media #73: Compile error for Amlogic Openelec linux

BTopbas, 2015-01-21:
Hi;
I have m8 amlogic box and i use codesnake's openelec port for this device. I try compiling with openelec but this is give me error in soc_camera.o 
Here is compiling error:
  CC [M]  /home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l/soc_camera.o
  CC [M]  /home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l/soc_mediabus.o
  CC [M]  /home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l/soc_scale_crop.o
/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l/soc_camera.c: In function 'soc_of_bind':
/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l/soc_camera.c:1660:2: error: implicit declaration of function 'of_find_i2c_device_by_node' [-Werror=implicit-function-declaration]
  client = of_find_i2c_device_by_node(remote);
  ^
/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l/soc_camera.c:1660:9: warning: assignment makes pointer from integer without a cast
  client = of_find_i2c_device_by_node(remote);
         ^
cc1: some warnings being treated as errors
scripts/Makefile.build:314: recipe for target '/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l/soc_camera.o' failed
make[4]: *** [/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l/soc_camera.o] Error 1
make[4]: *** Waiting for unfinished jobs....
Makefile:1239: recipe for target '_module_/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l' failed
make[3]: *** [_module_/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l] Error 2
make[3]: Leaving directory '/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/linux-amlogic-3.10-24e850b-oe'
Makefile:51: recipe for target 'default' failed
make[2]: *** [default] Error 2
make[2]: Leaving directory '/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/batuhan/Desktop/OpenELEC.tv/build.OpenELEC-Amlogic.M8.arm-devel/ljalves-12012015/media_build'
make: *** [release] Error 2

BTopbas, 2015-01-22:
Linux kernel 3.10.61 maybe kernel is old?

ljalves, 2015-01-28:
Yes, you will need kernel >= 3.12 or you can try the following hack:

Add in "/include/linux/i2c.h" at the end before the #endif
```c
static inline struct i2c_client *of_find_i2c_device_by_node(struct device_node *node)
{
        return NULL;
}

static inline struct i2c_adapter *of_find_i2c_adapter_by_node(struct device_node *node)
{
        return NULL;
}
```

closed 2015-03-26 by ljalves