ljalves/linux_media #74: TBS6981/vb2-cx23885 kernel oops

nickb937, 2015-01-24:
The following are 3 samples of kernel messages that I am getting on a fairly regular basis, which seem to have started occurring since the cx23885 vb2 conversion.

I have a TBS6985 and a TBS6981 card present in the machine.  The TBS6985 is working fine.  The kernel messages are triggered by VDR's idle scanning which it done using the TBS6981.

[16795.690383] BUG: Bad page state in process vb2-cx23885[0]  pfn:36bd1
[16795.690387] page:ffffea0000daf440 count:-1 mapcount:0 mapping:          (null) index:0x0
[16795.690389] flags: 0x1ffff0000000000()
[16795.690390] page dumped because: nonzero _count
[16795.690391] Modules linked in: rfcomm bluetooth iptable_mangle xt_DSCP ip_tables x_tables ir_jvc_decoder(OE) ir_lirc_codec(OE) lirc_dev(OE) ir_xmp_decoder(OE) ir_sony_decoder(OE) ir_sanyo_decoder(OE) ir_sharp_decoder(OE) ir_rc6_decoder(OE) ir_rc5_decoder(OE) ir_nec_decoder(OE) ir_mce_kbd_decoder(OE) rc_tbs_nec(OE) cx25840(OE) binfmt_misc rc_dib0700_rc5(OE) dib7000p(OE) cx23885(OE) altera_ci(OE) tda18271(OE) altera_stapl(OE) snd_hda_codec_hdmi videobuf2_dvb(OE) tveeprom(OE) snd_hda_codec_realtek av201x(OE) intel_rapl cx2341x(OE) x86_pkg_temp_thermal intel_powerclamp videobuf2_dma_sg(OE) videobuf2_memops(OE) coretemp videobuf2_core(OE) kvm snd_hda_codec_generic v4l2_common(OE) snd_hda_intel snd_hda_controller snd_hda_codec videodev(OE) crct10dif_pclmul crc32_pclmul saa716x_budget(OE) ghash_clmulni_intel snd_hwdep media(OE) snd_pcm i915 aesni_intel dvb_usb_dib0700(OE) tas2101(OE) aes_x86_64 dib9000(OE) dib7000m(OE) dib0090(OE) dib0070(OE)
[16795.690421] vb2: counters for queue ffff880227c55828: UNBALANCED!
[16795.690422]  cxd2820r(OE) dib3000mc(OE) lrw video
[16795.690425] vb2:     setup: 1 start_streaming: 1 stop_streaming: 1
[16795.690426] vb2:     wait_prepare: 10464 wait_finish: 10465
[16795.690427] vb2:   counters for queue ffff880227c55828, buffer 31: UNBALANCED!
[16795.690429] vb2:     buf_init: 1 buf_cleanup: 1 buf_prepare: 109 buf_finish: 110
[16795.690429] vb2:     buf_queue: 109 buf_done: 109
[16795.690430] vb2:     alloc: 1 put: 1 prepare: 109 finish: 109 mmap: 0
[16795.690431] vb2:     get_userptr: 0 put_userptr: 0
[16795.690432] vb2:     attach_dmabuf: 0 detach_dmabuf: 0 map_dmabuf: 0 unmap_dmabuf: 0
[16795.690433] vb2:     get_dmabuf: 0 num_users: 0 vaddr: 110 cookie: 110
[16795.690434]  gpio_ich drm_kms_helper snd_seq_midi snd_seq_midi_event snd_rawmidi gf128mul dibx000_common(OE) mb86a16(OE) snd_seq glue_helper snd_seq_device lp cx24117(OE) mei_me snd_timer dvb_usb(OE) drm ablk_helper rc_core(OE) parport saa716x_core(OE) lpc_ich snd mei dvb_core(OE) soundcore i2c_algo_bit i2c_mux serio_raw cryptd mac_hid psmouse ahci r8169 libahci mii
[16795.690452] CPU: 1 PID: 16771 Comm: vb2-cx23885[0] Tainted: G           OE  3.18.1-ickle75+ #1
[16795.690453] Hardware name: Shuttle SH67H3/FH67, BIOS 4.6.4 03/22/2011
[16795.690454]  ffffffff81aad259 ffff880231493a48 ffffffff817781fd 0000000000000000
[16795.690456]  ffffea0000daf440 ffff880231493a78 ffffffff81774bcc ffffea0000daf440
[16795.690458]  0000000000000141 0000000000000200 ffff88023fdf4780 ffff880231493b68
[16795.690459] Call Trace:
[16795.690465]  [<ffffffff817781fd>] dump_stack+0x46/0x58
[16795.690468]  [<ffffffff81774bcc>] bad_page.part.50+0xe0/0xfe
[16795.690471]  [<ffffffff81175113>] get_page_from_freelist+0x873/0x9e0
[16795.690474]  [<ffffffff810a2a3b>] ? select_idle_sibling+0x2b/0x120
[16795.690498]  [<ffffffff811753df>] __alloc_pages_nodemask+0x15f/0xa50
[16795.690501]  [<ffffffff8109cf64>] ? try_to_wake_up+0x1e4/0x330
[16795.690505]  [<ffffffff81019db4>] dma_generic_alloc_coherent+0xa4/0x160
[16795.690508]  [<ffffffff810597e1>] x86_swiotlb_alloc_coherent+0x21/0x50
[16795.690513]  [<ffffffffc068c822>] cx23885_risc_databuffer+0xb2/0x170 [cx23885]
[16795.690519]  [<ffffffffc01a2221>] ? dvb_dmx_swfilter_section_copy_dump+0x1f1/0x250 [dvb_core]
[16795.690522]  [<ffffffffc068d6af>] cx23885_buf_prepare+0x7f/0xd0 [cx23885]
[16795.690526]  [<ffffffffc068ed49>] buffer_prepare+0x19/0x20 [cx23885]
[16795.690529]  [<ffffffffc04607ee>] __buf_prepare+0x27e/0x390 [videobuf2_core]
[16795.690532]  [<ffffffffc046203b>] vb2_internal_qbuf+0x7b/0x240 [videobuf2_core]
[16795.690535]  [<ffffffffc046230e>] vb2_thread+0x10e/0x480 [videobuf2_core]
[16795.690538]  [<ffffffffc0462200>] ? vb2_internal_qbuf+0x240/0x240 [videobuf2_core]
[16795.690540]  [<ffffffff8108ff42>] kthread+0xd2/0xf0
[16795.690542]  [<ffffffff8108fe70>] ? kthread_create_on_node+0x180/0x180
[16795.690545]  [<ffffffff817805bc>] ret_from_fork+0x7c/0xb0
[16795.690547]  [<ffffffff8108fe70>] ? kthread_create_on_node+0x180/0x180
[16795.690548] Disabling lock debugging due to kernel taint

[129577.894551] BUG: unable to handle kernel NULL pointer dereference at 0000000000000058
[129577.896227] IP: [<ffffffffc046237a>] vb2_thread+0x17a/0x480 [videobuf2_core]
[129577.897858] PGD 0 
[129577.899451] Oops: 0002 [#6] SMP 
[129577.901033] Modules linked in: rfcomm bluetooth iptable_mangle xt_DSCP ip_tables x_tables ir_jvc_decoder(OE) ir_lirc_codec(OE) lirc_dev(OE) ir_x
mp_decoder(OE) ir_sony_decoder(OE) ir_sanyo_decoder(OE) ir_sharp_decoder(OE) ir_rc6_decoder(OE) ir_rc5_decoder(OE) ir_nec_decoder(OE) ir_mce_kbd_dec
oder(OE) rc_tbs_nec(OE) cx25840(OE) binfmt_misc rc_dib0700_rc5(OE) dib7000p(OE) cx23885(OE) altera_ci(OE) tda18271(OE) altera_stapl(OE) snd_hda_code
c_hdmi videobuf2_dvb(OE) tveeprom(OE) snd_hda_codec_realtek av201x(OE) intel_rapl cx2341x(OE) x86_pkg_temp_thermal intel_powerclamp videobuf2_dma_sg
(OE) videobuf2_memops(OE) coretemp videobuf2_core(OE) kvm snd_hda_codec_generic v4l2_common(OE) snd_hda_intel snd_hda_controller snd_hda_codec video
dev(OE) crct10dif_pclmul crc32_pclmul saa716x_budget(OE) ghash_clmulni_intel snd_hwdep media(OE) snd_pcm i915 aesni_intel dvb_usb_dib0700(OE) tas210
1(OE) aes_x86_64 dib9000(OE) dib7000m(OE) dib0090(OE) dib0070(OE) cxd2820r(OE) dib3000mc(OE) lrw video gpio_ich drm_kms_helper snd_seq_midi snd_seq_
midi_event snd_rawmidi gf128mul dibx000_common(OE) mb86a16(OE) snd_seq glue_helper snd_seq_device lp cx24117(OE) mei_me snd_timer dvb_usb(OE) drm ab
lk_helper rc_core(OE) parport saa716x_core(OE) lpc_ich snd mei dvb_core(OE) soundcore i2c_algo_bit i2c_mux serio_raw cryptd mac_hid psmouse ahci r81
69 libahci mii
[129577.911617] CPU: 0 PID: 11837 Comm: vb2-cx23885[0] Tainted: G    B D    OE  3.18.1-ickle75+ #1
[129577.913452] Hardware name: Shuttle SH67H3/FH67, BIOS 4.6.4 03/22/2011
[129577.915308] task: ffff880228249e00 ti: ffff8802310d0000 task.ti: ffff8802310d0000
[129577.917139] RIP: 0010:[<ffffffffc046237a>]  [<ffffffffc046237a>] vb2_thread+0x17a/0x480 [videobuf2_core]
[129577.918983] RSP: 0018:ffff8802310d3e48  EFLAGS: 00010246
[129577.920817] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 000000000000000b
[129577.922654] RDX: 0000000000000058 RSI: ffff880228249e00 RDI: 0000000000000058
[... 61 lines cut ...]
[132180.785765] cx23885[0]:   ptr1_reg: 0x00006310
[132180.785767] cx23885[0]:   ptr2_reg: 0x000105f8
[132180.785769] cx23885[0]:   cnt1_reg: 0x00000002
[132180.785771] cx23885[0]:   cnt2_reg: 0x00000007

00:00.0 Host bridge: Intel Corporation 2nd Generation Core Processor Family DRAM Controller (rev 09)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200/2nd Generation Core Processor Family PCI Express Root Port (rev 09)
00:02.0 VGA compatible controller: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller (rev 09)
00:16.0 Communication controller: Intel Corporation 6 Series/C200 Series Chipset Family MEI Controller #1 (rev 04)
00:1a.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #2 (rev 05)
00:1b.0 Audio device: Intel Corporation 6 Series/C200 Series Chipset Family High Definition Audio Controller (rev 05)
00:1c.0 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 1 (rev b5)
00:1c.1 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 2 (rev b5)
00:1c.2 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 3 (rev b5)
00:1c.3 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 4 (rev b5)
00:1c.4 PCI bridge: Intel Corporation 6 Series/C200 Series Chipset Family PCI Express Root Port 5 (rev b5)
00:1d.0 USB controller: Intel Corporation 6 Series/C200 Series Chipset Family USB Enhanced Host Controller #1 (rev 05)
[... cut ...]
00:1f.3 SMBus: Intel Corporation 6 Series/C200 Series Chipset Family SMBus Controller (rev 05)
01:00.0 Multimedia video controller: Conexant Systems, Inc. CX23885 PCI Video and Audio Decoder (rev 04)
03:00.0 USB controller: ASMedia Technology Inc. ASM1042 SuperSpeed USB Host Controller
04:00.0 USB controller: ASMedia Technology Inc. ASM1042 SuperSpeed USB Host Controller
05:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (rev 06)
06:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 02)

typingArtist, 2015-01-25:
I can confirm these issues.
TBS closed-source driver doesn’t exhibit this behavior.

ljalves, 2015-01-28:
This issue is happening on the official media tree (check issue #62)
Please report this to the media mailing list (linux-media@vger.kernel.org)

ljalves, 2015-05-02:
I believe that this has been fixed some time ago in the official media tree.
Closing this issue.

closed 2015-05-02 by ljalves