ljalves/linux_media #75: TBS6280 fails to attach

ep0ch2, 2015-01-31:
Hi,

I'm trying to get the latest branch to work but i'm having an issue i can't get round. So I'd like some help please.

```
dmesg output
[    6.028478] WARNING: You are using an experimental version of the media stack.
[    6.028479]  As the driver is backported to an older kernel, it doesn't offer
[    6.028480]  enough quality for its usage in production.
[    6.028481]  Use it with care.
[    6.028482] Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
[    6.028483]  fe7dd40dbf03b4a9c0c5fe7533270d5da111771f TBS6991SE fix atempt: switch i2c buses
[    6.071223] SAA716x Budget 0000:01:00.0: setting latency timer to 64
[    6.088249] DVB: registering new adapter (SAA716x dvb adapter)
[    6.697821] SAA716x Budget 0000:01:00.0: TurboSight TBS 6280 frontend 0 tuner attach failed
[    6.698786] SAA716x Budget 0000:01:00.0: TurboSight TBS 6280 frontend 0 attach failed
[    6.699768] DVB: registering new adapter (SAA716x dvb adapter)
[    6.701855] SAA716x Budget 0000:01:00.0: TurboSight TBS 6280 frontend 1 tuner attach failed
[    6.702825] SAA716x Budget 0000:01:00.0: TurboSight TBS 6280 frontend 1 attach failed
```
Showing it can't attach.
```
lspvi -v
01:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
        Subsystem: Device 6280:0011
        Physical Slot: 0
        Flags: bus master, fast devsel, latency 0, IRQ 16
        Memory at fce00000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] MSI: Enable- Count=1/32 Maskable- 64bit+
        Capabilities: [50] Express Endpoint, MSI 00
        Capabilities: [74] Power Management version 2
        Capabilities: [80] Vendor Specific Information: Len=50 <?>
        Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
        Kernel driver in use: SAA716x Budget
```

```devices
crw-rw---T 1 root video 212, 5 Jan 31 10:38 net0
crw-rw---T 1 root video 212, 4 Jan 31 10:38 dvr0
crw-rw---T 1 root video 212, 3 Jan 31 10:38 demux0

/dev/dvb/adapter0:
total 0
crw-rw---T 1 root video 212, 2 Jan 31 10:38 net0
crw-rw---T 1 root video 212, 1 Jan 31 10:38 dvr0
crw-rw---T 1 root video 212, 0 Jan 31 10:38 demux0
```

If i build the closed source driver it works fine except for some bugs.

Is there anything else I can do to find out why it's not attaching?

Thanks for your help

ep0ch2, 2015-02-07:
seems to be tda18212 related? any workaround i can do

crazycat69, 2015-02-11:
TBS 6220 work ok with latest tda18212-related changes.

ep0ch2, 2015-02-12:
6220 or 6280?

ljalves, 2015-05-02:
@ep0ch2,
Did you solve your issue?

ep0ch2, 2015-05-02:
Hi, no i didn't. The official TBS drivers worked, CrazyCat drivers worked okay, but I didn't manage to get these working. I'll try again in a week. thanks.
I should note i'm running under proxmox.

ep0ch2, 2015-05-06:
Currently i can't build:

/ljalves/media_build/v4l/adp1653.c:39:33: fatal error: linux/gpio/consumer.h: No such file or directory

running:
 3.2.0-4-amd64 #1 SMP Debian 3.2.68-1+deb7u1 x86_64 GNU/Linux

ep0ch2, 2015-05-21:
latest now builds but still fails to attach

    5.111885] media: Linux media interface: v0.10
[    5.114422] WARNING: You are using an experimental version of the media stack.
[    5.114431]  As the driver is backported to an older kernel, it doesn't offer
[    5.114433]  enough quality for its usage in production.
[    5.114434]  Use it with care.
[    5.114435] Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
[    5.114436]  ca628c878f1a2fbe28eac7b51c81732921d8dd40 Merge pull request #97 from bas-t/fixes
[    5.186512] snd_hda_intel 0000:00:1b.0: irq 55 for MSI/MSI-X
[    5.191975] snd_hda_intel 0000:00:1b.0: setting latency timer to 64
[    5.201088] SAA716x Budget 0000:06:10.0: setting latency timer to 64
[    5.208938] hda-intel: no codecs found!
[    5.220389] DVB: registering new adapter (SAA716x dvb adapter)
[    5.661490] input: ImExPS/2 Generic Explorer Mouse as /devices/platform/i8042/serio1/input/input3
[    5.830481] SAA716x Budget 0000:06:10.0: TurboSight TBS 6280 frontend 0 tuner attach failed
[    5.832508] SAA716x Budget 0000:06:10.0: TurboSight TBS 6280 frontend 0 attach failed
[    5.836359] DVB: registering new adapter (SAA716x dvb adapter)
[    5.839257] SAA716x Budget 0000:06:10.0: TurboSight TBS 6280 frontend 1 tuner attach failed
[    5.841283] SAA716x Budget 0000:06:10.0: TurboSight TBS 6280 frontend 1 attach failed

ep0ch78, 2015-09-18:
you can close this now.
all good

closed 2015-09-19 by ljalves

timmydog, 2015-09-19:
Confirmed - works on my 6280 too.