ljalves/linux_media #78: TBS6281 - HDTV - Boxer

ClaesBG, 2015-02-02:
Hi,

I just compiled the latest version of the opensource drivers successfully on a Ubuntu 14.04.1 LTS. All the required firmware has been downloaded and the card looks OK in DMESG.

[    6.286824] si2168 1-0064: Silicon Labs Si2168 successfully attached
[    6.376895] si2157 2-0060: si2157: Found a Si2158-A20 rev51
[    6.376900] si2157 2-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.538357] si2168 0-0064: Silicon Labs Si2168 successfully attached
[    6.596886] si2157 10-0060: si2157: Found a Si2158-A20 rev51
[    6.596890] si2157 10-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.777724] si2168 12-0064: Silicon Labs Si2168 successfully attached
[    6.832886] si2157 13-0060: si2157: Found a Si2158-A20 rev51
[    6.832890] si2157 13-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[    6.995374] si2168 11-0064: Silicon Labs Si2168 successfully attached
[    7.052894] si2157 14-0060: si2157: Found a Si2158-A20 rev51
[    7.052896] si2157 14-0060: Silicon Labs Si2147/2148/2157/2158 successfully attached
[  124.317990] si2157 2-0060: found a 'Silicon Labs Si2146/2147/2148/2157/2158' in cold state
[  124.334114] si2157 2-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[  124.556546] si2168 1-0064: found a 'Silicon Labs Si2168' in cold state
[  124.566594] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  124.710936] si2168 1-0064: found a 'Silicon Labs Si2168' in warm state

When I do a scan the card does not seems to be able to locate the HDTV channels on channel 33 (570MHz). This might not be the driver itself but the cards (I have two installed). 

Any idea what it can be? If I connect a normal DVB-T2 box it has no problems at all to find the channels on 570MHz.

Thanks in advance!

Compulsion, 2015-02-08:
There does seem to be an issue with DVB-T2. Been playing this evening:
Arch Linux
Kernel 3.18.6
TBS6281
Compiled drivers from this repository today.
int_type=1
Lastest v4l-utils from git.
dvbv5-scan works fine, and creates a channels file with all the DVB-T and DVB-T2 channels available on my local transmitter.
```
lspci -vvnn
01:00.0 Multimedia controller [0480]: Philips Semiconductors SAA7160 [1131:7160] (rev 03)
	Subsystem: Device [6281:0001]
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 93
	Region 0: Memory at d0800000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] MSI: Enable+ Count=1/32 Maskable- 64bit+
		Address: 00000000fee0f00c  Data: 4152
	Capabilities: [50] Express (v1) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s <256ns, L1 <1us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE- FLReset-
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 2.5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <4us, L1 <64us
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp-
		LnkCtl:	ASPM Disabled; RCB 128 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk- DLActive- BWMgmt- ABWMgmt-
	Capabilities: [74] Power Management version 2
		Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [80] Vendor Specific Information: Len=50 <?>
	Capabilities: [100 v1] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
	Kernel driver in use: SAA716x Budget
	Kernel modules: saa716x_budget
```
On the first zap attempt on each adapter it seems to load the tuner firmware but fails to lock:
```
dvbv5-zap -c dvb_channel.conf -t 20 -o bbc2hd.ts 'BBC TWO HD'
using demux '/dev/dvb/adapter0/demux0'
reading channels from file 'dvb_channel.conf'
service has pid type 05:  150
tuning to 770000000 Hz
video pid 101
  dvb_set_pesfilter 101
audio pid 102
  dvb_set_pesfilter 102
       (0x00) C/N= 30.25dB
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
       (0x00)
[... 48 lines cut ...]
### zap failed adapter 0 ###
Feb 08 20:59:08 shottower kernel: si2157 10-0060: found a 'Silicon Labs Si2146/2147/2148/2157/2158
Feb 08 20:59:08 shottower kernel: si2157 10-0060: downloading firmware from file 'dvb-tuner-si2158

### zap failed adapter 1 ###
Feb 08 20:59:33 shottower kernel: si2157 11-0060: found a 'Silicon Labs Si2146/2147/2148/2157/2158
Feb 08 20:59:33 shottower kernel: si2157 11-0060: downloading firmware from file 'dvb-tuner-si2158

### zap success adapter 0 ###
Feb 08 21:00:14 shottower kernel: si2168 8-0064: found a 'Silicon Labs Si2168' in cold state
Feb 08 21:00:14 shottower kernel: si2168 8-0064: downloading firmware from file 'dvb-demod-si2168-
Feb 08 21:00:14 shottower kernel: si2168 8-0064: found a 'Silicon Labs Si2168' in warm state

### zap success adapter 1 ###
Feb 08 21:02:27 shottower kernel: si2168 9-0064: found a 'Silicon Labs Si2168' in cold state
Feb 08 21:02:27 shottower kernel: si2168 9-0064: downloading firmware from file 'dvb-demod-si2168-
Feb 08 21:02:28 shottower kernel: si2168 9-0064: found a 'Silicon Labs Si2168' in warm state

### all subsequent DVB-T2 zap attempts fail but DVB-T works ###
```
@ClaesBG 
It's worth noting that not all apps support DVB-T2 properly:
http://blog.palosaari.fi/2014/09/linux-dvb-t2-tuning-problems.html
MythTV is one of these, but a patch has been submitted, so I guess it will be fixed in the next release:
https://github.com/MythTV/mythtv/pull/90

ljalves, 2015-03-26:
Can you retry building the drivers from the latest stuff?
Fixes for the demod/tuner drivers have been submitted to the media tree and merged back here.

Compulsion, 2015-05-01:
Hi there, sorry for slow response. Built the latest today.

DVB-T still works fine at all times.
I can only get a lock on DVB-T2 on the first zap after boot:
```
May 01 23:18:22 shottower kernel: si2157 11-0060: firmware version: 2.1.6
May 01 23:18:21 shottower kernel: si2157 11-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
May 01 23:18:21 shottower kernel: si2157 11-0060: found a 'Silicon Labs Si2158-A20'
May 01 23:18:21 shottower kernel: si2168 8-0064: firmware version: 4.0.4
May 01 23:18:21 shottower kernel: si2168 8-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
May 01 23:18:21 shottower kernel: si2168 8-0064: found a 'Silicon Labs Si2168-B40'
```
The successful T2 zaps result in perfect quality audio + video, no sign of the pixelation I reported in a related issue, which I'm guessing could be due to a change in ffmpeg (used to get the zap output into a playable state), rather than the driver.
.

Xxyzzy, 2015-05-02:
Hi !

Somewhat similar TBS6281 problem here. 

DVB-C (comhem) works, DVB-T (boxer) works. DVB-T2 (boxer) does not work at all. Completely dead. 

I'm using channel files produced using a nanoStick T2 290e (the nanoStick works flawlessly for DVB-C, DVB-T and DVB-T2). This is on a 4.0.1 kernel.

Firmware seem to load as expected:
[  770.038310] si2168 0-0064: found a 'Silicon Labs Si2168-B40'
[  770.038336] si2168 0-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[  770.182572] si2168 0-0064: firmware version: 4.0.4
[  770.185025] si2157 3-0060: found a 'Silicon Labs Si2158-A20'
[  770.185037] si2157 3-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[  770.405907] si2157 3-0060: firmware version: 2.1.6

I have also tried out the proprietary driver (v150411) on a 3.18.12 kernel.
With this driver I get a lock for a short while on the DVB-T2 frequencies when running w_chan, followed by "no data from PAT". Using this driver the card is reported only to support DVB-T, not DVB-T2 (by dvb-fe-tool).

Tests made on two different machines, one Intel-based and one AMD-based.

ljalves, 2015-05-02:
The si2168/si2157 driver is comming from the official linux media tree.
Any of you will have better luck by reporting any related issue to this driver in the corresponding mailing list (linux-media@vger.kernel.org).

Xxyzzy, 2015-05-04:
A bit off topic... I managed to get the proprietary driver to work using a hacked version of dvbv5-zap.
Dvb-fe-tool reports only DVBT, but using proper DVBT2 config seem to work.

It seems as if the hw has to be "trained" though.
After reboot I get some "Sync" lines from dvbv5-zap, switching to a channel on another DVBT2 transponder normally results in a "Lock".
And from now on, it works without any problems, until next reboot when the "training" must be repeated.
Ie, switchnig back/fort beetween transpoders. Normally one switch seem to be enough.

Btw, the dvbv5-zap hack is for PMT (the -p switch). If no PMT is found, the PMT is assumed to have the same value as the SID.
Seem to work for all channels, at least here (teracom, sweden).

Compulsion, 2015-07-12:
Good news, I have T2 working now!

Was testing with si2168 firmware versions 4.0.4, and 4.0.11 previously, and debug output suggested that a command was timing out in si2168_set_frontend, in the bit where it selects the PLP for T2.

4.0.19 firmware has fixed the issue for me:
http://www.mail-archive.com/linux-media%40vger.kernel.org/msg89283.html

Was having problems when using both tuners simultaneously with the official TBS drivers. Occasional picture breakup, which was especially noticeable when one tuner was EPG scanning.

I've now been able to switch my server away from TBS to Luis's drivers, and Initial testing looks good, not noticed any breakup even when EPG scanning on the spare tuner.

Thanks for your work Luis.

Xxyzzy, 2015-07-12:
Hi !

Sorry, no such luck for me (using a 4.0.8 kernel).

Jul 12 18:50:11 nas kernel: [   71.104742] si2168 1-0064: found a 'Silicon Labs Si2168-B40'
Jul 12 18:50:11 nas kernel: [   71.106111] si2168 1-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
Jul 12 18:50:11 nas kernel: [   71.461328] si2168 1-0064: firmware version: 4.0.19
Jul 12 18:50:11 nas kernel: [   71.464001] si2157 2-0060: found a 'Silicon Labs Si2158-A20'
Jul 12 18:50:11 nas kernel: [   71.464944] si2157 2-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
Jul 12 18:50:12 nas kernel: [   71.688258] si2157 2-0060: firmware version: 2.1.6

[hakan@nas ~]$ /usr/local/dvbv5/bin/dvbv5-zap -a 0 -r -c .tzap/dvb_channel.conf SVT2HD
using demux '/dev/dvb/adapter0/demux0'
reading channels from file '.tzap/dvb_channel.conf'
tuning to 722000000 Hz
video pid 2309
  dvb_set_pesfilter 2309
audio pid 2307
  dvb_set_pesfilter 2307
       (0x00)
       (0x00)
       (0x00)
       (0x00)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
       (0x00)
       (0x00)
       (0x00)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
Carrier(0x03)
       (0x00)
<...>

However, the Official TBS v150525 drivers seem to work rather well for me using a 4.0.4 kernel (does not work with 4.0.8), also when both channels are in simultaneous use.

ljalves, 2016-06-24:
Closing old issue.
Reopen if needed

closed 2016-06-24 by ljalves