ljalves/linux_media #81: Compiling the driver on Ubuntu 12.04

zougoulou59, 2015-02-06:
Hi,
I'm trying to compil the driver following those instructions 
```
git clone https://github.com/ljalves/media_build.git'
git clone --depth=1 https://github.com/ljalves/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make distclean
make
```
but i've got an error when compiling rtl2832.o when using the `make` command
```
  CC [M]  /home/zougoulou59/media_build/v4l/rtl2832.o
/home/zougoulou59/media_build/v4l/rtl2832.c: In function 'rtl2832_probe':
/home/zougoulou59/media_build/v4l/rtl2832.c:1222:4: warning: passing argument 3 of 'i2c_add_mux_adapter' makes integer from pointer without a cast [enabled by default]
include/linux/i2c-mux.h:35:21: note: expected 'u32' but argument is of type 'struct rtl2832_priv *'
/home/zougoulou59/media_build/v4l/rtl2832.c:1222:4: error: too many arguments to function 'i2c_add_mux_adapter'
include/linux/i2c-mux.h:35:21: note: declared here
/home/zougoulou59/media_build/v4l/rtl2832.c:1235:4: warning: passing argument 3 of 'i2c_add_mux_adapter' makes integer from pointer without a cast [enabled by default]
include/linux/i2c-mux.h:35:21: note: expected 'u32' but argument is of type 'struct rtl2832_priv *'
/home/zougoulou59/media_build/v4l/rtl2832.c:1235:4: warning: passing argument 5 of 'i2c_add_mux_adapter' makes pointer from integer without a cast [enabled by default]
include/linux/i2c-mux.h:35:21: note: expected 'int (*)(struct i2c_adapter *, void *, u32)' but argument is of type 'int'
/home/zougoulou59/media_build/v4l/rtl2832.c:1235:4: error: too many arguments to function 'i2c_add_mux_adapter'
include/linux/i2c-mux.h:35:21: note: declared here
make[3]: *** [/home/zougoulou59/media_build/v4l/rtl2832.o] Error 1
make[2]: *** [_module_/home/zougoulou59/media_build/v4l] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-3.2.0-76-generic'
make[1]: *** [default] Error 2
make[1]: Leaving directory `/home/zougoulou59/media_build/v4l'
make: *** [all] Error 2
```
I've got the TBS6281

```
zougoulou59@nas:~/media_build$ sudo lspci -v | grep --after-context=10 7160
06:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
	Subsystem: Device 6281:0001
	Flags: bus master, fast devsel, latency 0, IRQ 10
	Memory at f7800000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] MSI: Enable- Count=1/32 Maskable- 64bit+
	Capabilities: [50] Express Endpoint, MSI 00
	Capabilities: [74] Power Management version 2
	Capabilities: [80] Vendor Specific Information: Len=50 <?>
	Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
```

ljalves, 2015-03-26:
What kernel version are you using?
Can you retry with the latest stuff...

ljalves, 2015-05-02:
No response. Closing this.

closed 2015-05-02 by ljalves