ljalves/linux_media #84: TBS6285 V12M no signal

epon, 2015-02-15:
I've re-opened this and removed posts from other issues to prevent them being brough off-topic.

However I have the following:

V12M TBS 6285 which I can't get any signal with.

My motherboard block diagram is like this:

Notice that the card is connected on PCIE4 through a PCI-E siwtch (PLX PEX 8747).

I've tried the card in PCIE3 nad PCI7 i.e. not though a switch in case it was that that was causing the problems but the card isn't even recognized by the BIOS on those sockets!

<table style="width:auto;"><tr><td>[image]</td></tr><tr><td style="font-family:arial,sans-serif; font-size:11px; text-align:right">From <a href="https://picasaweb.google.com/101120527538649978903/Forum?authuser=0&feat=embedwebsite">Forum</a></td></tr></table>

I just tried with the latest repos, using firmware from both bas-t and OE repo with int_type=1 and int_type=0:

int_type=0: I can scan but get only some channel information but can't play and of the channels. TVHeadend reports a transport error:

<pre>
tvheadend[4335]: TS: Silicon Labs Si2168/522,000 kHz/BBC TWO: Transport error indicator
tvheadend[4335]: TS: Silicon Labs Si2168/522,000 kHz/BBC TWO: MPEG2VIDEO @ #201: Continuity counter error
</pre>

int_type=1: I can scan but get no channel information at all and can't play any channels.

I'm not using IOMMU and I'm also doing a "full initialization" of the card to DVBT2 using dvb-fe-tool.

Nothing appears to be working :(

Dmesg:
<pre>
[  535.825804] si2168 2-0064: found a 'Silicon Labs Si2168' in cold state
[  535.825824] si2168 2-0064: downloading firmware from file 'dvb-demod-si2168-a20-01.fw'
[  537.254066] si2168 2-0064: found a 'Silicon Labs Si2168' in warm state
[  537.256039] si2157 3-0062: found a 'Silicon Labs Si2146/2147/2148/2157/2158' in cold state
[  537.256050] si2157 3-0062: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
</pre>

bas-t, 2015-02-15:
Silicon Labs Si2168 should be loading dvb-demod-si2168-b40-01.fw afaik.

epon, 2015-02-15:
Is that an issue on my end or the driver end? It seems like the driver is specifically looking for that version of the firmware...

bas-t, 2015-02-15:
Can't tell, Luis did not merge latest patch series for Si2168 and SI2157 yet. Are you running Debian Jessie?

epon, 2015-02-15:
I'm running Gentoo (gentoo-sources-3.17.7)

bas-t, 2015-02-15:
Well, you can try compiling from my repos, works for me.
Do:

git clone git://github.com/bas-t/media_build.git -b arch
git clone --depth=1 git://github.com/bas-t/media_tree.git -b arch && cd media_build
make dir DIR=../media_tree
make distclean
make allyesconfig
sed -i 's/SYS_DVBT, SYS_DVBT2, SYS_DVBC_ANNEX_A/SYS_DVBT/' drivers/media/dvb-frontends/si2168.c
make -j3 && make install
cp -f /data/v4l/media_tree/firmware_extra/* /lib/firmware/
echo "options saa716x_budget int_type=1" > /etc/modprobe.d/tbs_opensource.conf

And now, poweroff your system completely for a minute or so. After boot, it should work.
You don't need the dvb-fe-tool anymore, as your adapter is set to DVB-T permanently.
Best of luck!

bas-t, 2015-02-15:
Ok, the sed line is in two lines, should be on one line

epon, 2015-02-15:
Thanks bas-t. But how does your repo differ from Luis' ?

bas-t, 2015-02-15:
It is updated with the patch series that Luis does not have, also I synced 2157 with upstream, so it differs from Luis' branch. (I've sent Luis a PR for that, but he did not merge it yet)
and last but not least, it installs the drivers in the 'update'dir of your kernel. So it is easy to remove when you are not satisfied.

epon, 2015-02-15:
Ok thanks. I've currently got my own Gentoo ebuild which installs into a folder I've named "dvb-os" which gives the same result.

@ljalves Is it possible to merge the above patches?

epon, 2015-02-15:
FYI the output of w_scan with FE set to DVBT2 using dvb-fe-tool:

<pre>
....
....
Info: no data from SDT(actual)
Info: no data from PMT
Info: no data from PMT
Info: no data from PMT
Info: no data from PMT
Info: no data from PMT
Info: no data from PMT
Info: no data from PMT
Info: no data from PMT
Info: no data from PMT
Info: no data from PMT
Info: no data from NIT(actual)
dumping lists (12 services)
service_id 8258;(null):498000:B8:T:27500:1701=2:1702=eng@4,1703=eng:0;1731:0:8258:0:0:0
service_id 8325;(null):498000:B8:T:27500:1801=2:1802=eng@4,1803=eng:0;1831:0:8325:0:0:0
service_id 8357;(null):498000:B8:T:27500:1901=2:1902=eng@4,1903=eng:0;1931:0:8357:0:0:0
service_id 8448;(null):498000:B8:T:27500:1301=2:1302=eng@4,1303=eng:0;1331:0:8448:0:0:0
service_id 8452;(null):498000:B8:T:27500:1201=2:1202=eng@4,1203=eng:0;1231:0:8452:0:0:0
service_id 8500;(null):498000:B8:T:27500:1601=2:1602=eng@4,1603=eng:0;1631:0:8500:0:0:0
service_id 4287;(null):522000:B8:T:27500:201=2:202=eng@3,206=eng:0;205:0:4287:0:0:0
service_id 5632;(null):522000:B8:T:27500:0:1402=eng@3:0:0:5632:0:0:0
service_id 5760;(null):522000:B8:T:27500:0:1602=eng@3:0:0:5760:0:0:0
service_id 5888;(null):522000:B8:T:27500:0:1802=eng@3:0:0:5888:0:0:0
service_id 6848;(null):522000:B8:T:27500:0:1202=eng@3:0:0:6848:0:0:0
service_id 6912;(null):522000:B8:T:27500:0:1302=eng@3:0:0:6912:0:0:0
</pre>

Compulsion, 2015-02-15:
Looks like you're in the UK epon. I've got a 6281. TBS drivers work mostly okay for me with int_type=1, only issue I have is occasional pixelation when both tuners are in use. Did you have the same issue bas-t? (I vaguely remember you mentioning it somewhere)

With ljalves drivers, DVBT seems to work okay, but not DVBT2. First dvbv5-zap failed to lock, second worked, but all subsequent attempts fail (but DVBT continues to work). I posted in another issue:
https://github.com/ljalves/linux_media/issues/78

I tried your arch branch today bas-t. Much the same result, only difference was that it was the first T2 zap attempt that locked.

The T2 steams I did manage to capture suffer constant pixelation (you'll have to open the full size version to see!):
[image]

bas-t, 2015-02-15:
@Compulsion
@epon
You both seem to suffer from DVB-T/T2 issues, I can't reproduce this. I'm using DVB-C
No issues whatsoever over here (afaik)

bas-t, 2015-02-15:
@epon : try the sed method in favour of the dvb-fe-thingy. You just might get lucky!

epon, 2015-02-15:
@bas-t Hmm, should I set it to DVB-T or DVB-T2?

bas-t, 2015-02-15:
Dunno, try one. Educated guess: DVB-T

epon, 2015-02-15:
@bas-t It's a tricky one because here in the UK we use both DVB-T and DVB-T2. The latter are HD muxes only.

bas-t, 2015-02-15:
@epon However, you still have to load the right firmware. You may or may not remember: I've got a V12M too. In fact, that was the board Luis used to get this 6285 thing going. And I must say: thanks to Luis, I'm still going strong!

I don't know for sure, but you can try (in one line) :

sed -i 's/SYS_DVBT, SYS_DVBT2, SYS_DVBC_ANNEX_A/SYS_DVBT, SYS_DVBT2/' drivers/media/dvb-frontends/si2168.c

This should support both DVB-T and DVB-T2

bas-t, 2015-02-15:
I realy don't know if the driver supports both DVB-T and DVB-T2 at the same time. I'm guessing it doesn't. So, if you need both systems, you best set one or two of the si2168 chips to DVB-T and the others to DVB-T2. Of course, that brings you back to the dvb-fe-thingy.
Well, happy experimenting!

epon, 2015-02-17:
>> However, you still have to load the right firmware
@bas-t Looking at the image in my first post I'm thinking the firmware is correct.

>> I realy don't know if the driver supports both DVB-T and DVB-T2 at the same time. I'm guessing it doesn't. 

@bas-t If this is true then it's a real shame and this driver won't be any good for me. It's simply too inefficient to have tuners restricted to either DVB-T or DVB-T2... :(

ant-thomas, 2015-02-17:
I don't really know much (or anything) about these things but should it be much different to how all the DVB-S2 cards can handle both S and S2 broadcasts with the @ljalves drivers?

CkNoSFeRaTU, 2015-03-16:
I'm also have V12M 6285 board with Si2168-20 chip and tried today using DVB-T2 without any success.
Tried both this tree latest branch and bas-t arch branch with gentoo 3.19.1 kernel.
Tried both tuner firmwares 2.1.6 and 2.1.9.
Modules and firmwares looks like loading fine, except there are some warnings about sysfs which cannot create symlinks and duplicate filenames.
Tuning also seems work fine, but there are no usable data. I have two frequencies in town:
 - on 618Mhz/8Mhz there are some data going on, even can scan channels and read tables, but channel bitstream is completely broken.
 - on 674Mhz/8Mhz some data goes only first 50-100ms after tune. After that complete silence.

TBS proprietary drivers work fine(I'm using CrazyCat's tree https://bitbucket.org/CrazyCat/linux-tbs-drivers). If I load proprietary drivers after opensource ones there are some warnings about demod command is not responding during load, but card starts working fine.

I'm also have 6985 V11 board on the same machine, not tested it much but looks like working fine with this tree.

ljalves, 2015-05-02:
I don't have access to a DVB-T2 mux so I can't help much...
But If I were one of you guys having issues with dvb-t2 I would first ask for help in the official linux media mailing list.
The si2168/si2158 drivers are comming from there.

I would first ask if anyone using a card with those chips works ok with dvb-t2.

CkNoSFeRaTU, 2015-05-03:
I'll give it a try with mailing list.
Meanwhile if you are interested I can setup test machine and provide remote access to it if that helps.

CkNoSFeRaTU, 2015-09-01:
Just for FIY. Yesterday with latest branch I successfully tunned 6285 V12M board to our DVB-T2 muxes. No issues since then. Dunno whats changed in codebase from May.

Interesting if it'll be more stable than proprietary driver with which I had "no data" stucks after 1-2 weeks of uptime. According to beholders forum silabs made new demod firmware patch specifically for Russian DVB-T2 muxes. But it is not clear if it even exist for si2168-a20, they are not using those chips and provide fix only for A30/B40.

ljalves, 2016-06-24:
Closing - reopen if needed.

closed 2016-06-24 by ljalves