ljalves/linux_media #86: Support for TBS 5220

adaasch, 2015-03-07:
Is there support for TBS 5220 USB Stick? 
If not I could help with device and hacking if someone gives me an entry point.

ljalves, 2015-03-26:
If I'm not mistaken, the 5220 uses de silabs chips already open source'ed.
It should be very easy to add support to card.

adaasch, 2015-03-28:
Yes, it has an Cypress CY7C68013A USB bridge and a SiLabs Si2168 DVB Chip on it.

ljalves, 2015-05-02:
@adaasch, still interested on adding support for that card?

adaasch, 2015-05-03:
@ljalves, yes, I am. What are the steps?

nereeb, 2015-08-15:
does this stick have the same components as the geniatech t230?
Can I try echo 734c 5220 > /sys/bus/usb/drivers/??/new_id?
which module should I modprobe to do this?

chipsugar, 2015-09-20:
It doesn't work. cxusb is the module to modprobe but that fails to do anything. I'm no coder but I think actual support needs to be added for this device, rather than just a new id.

nereeb, 2015-09-20:
I had figured that out also the dvb_usb_dvbsky for tchnotrend 4400 did not work. So we have to wait and see if someone can add support.

nereeb, 2016-02-25:
The TBS5220 is working with https://github.com/crazycat69/linux_media

ljalves, 2016-03-01:
PR #112 merged: adds support for tbs5220.
Thanks CrazyCat.

closed 2016-03-01 by ljalves