ljalves/linux_media #87: Support for TBS6205 card

bpbastos, 2015-03-11:
Hello @ljalves,

I've received the 6205 card as a replacement for my 6285,this card is going to replace the TBS 6285.

According to TBS, this card uses the sames chips as 6285 they only have changed the pcie chip.

I'd to know if there is any chance of this card be added in your drivers and if there is anything I could do to help.

lspci
<pre>
02:02.0 Multimedia controller [0480]: Device [544d:6178]
	Subsystem: Device [6205:0001]
	Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast &#62;TAbort- &#60;TAbort- &#60;MAbort- &#62;SERR- &#60;PERR- INTx-
	Interrupt: pin A routed to IRQ 46
	Region 0: Memory at fe640000 (32-bit, non-prefetchable) [size=256K]
	Capabilities: [50] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [70] MSI: Enable+ Count=1/1 Maskable- 64bit+
		Address: 00000000fee00000  Data: 4022
	Capabilities: [90] Express (v1) Endpoint, MSI 00
		DevCap:	MaxPayload 128 bytes, PhantFunc 0, Latency L0s &#60;64ns, L1 &#60;1us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	Report errors: Correctable+ Non-Fatal+ Fatal+ Unsupported-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 2.5GT/s, Width x1, ASPM L0s, Exit Latency L0s unlimited, L1 &#60;1us
			ClockPM- Surprise- LLActRep- BwNot-
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 2.5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
	Capabilities: [100 v1] Device Serial Number 00-00-00-00-00-00-00-00
	Kernel driver in use: TBS PCIE
</pre>

modinfo
<pre>
filename:       /lib/modules/3.16.0-30-generic/kernel/drivers/media/common/tbspcie/tbs_pcie-dvb.ko
version:        1.0
license:        GPL
author:         Konstantin Dimitrov <kosio.dimitrov@gmail.com>
description:    TBS PCIE driver
alias:          pci:v0000544Dd00006178sv00006704sd00000001bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006814sd00001131bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006205sd00000001bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006205sd00001131bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006905sd00000001bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006908sd00001132bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006908sd00000001bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006908sd00001131bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006904sd00001131bc*sc*i*
depends:        dvb-core
vermagic:       3.16.0-30-generic SMP mod_unload modversions 
parm:           adapter_nr:DVB adapter numbers (array of short)
parm:           int_type:force Interrupt Handler type: 0=INT-A, 1=MSI. default INT-A mode (int)
</pre>

bpbastos, 2015-03-16:
More information about the drivers I am currently using.

Picture of the card: http://i.imgur.com/z1DM8uL.jpg

Drivers:

Kconfig
http://pastebin.com/DUTp3Mkv

Makefile
http://pastebin.com/WDaJegEL

tbs_pcie.c
http://pastebin.com/VvLJwMJ1

tbs_pcie.h
http://pastebin.com/AHS3zapQ

tbs_pcie-reg.h
http://pastebin.com/vZq4y7eP

Many thanks to @bas-t helping me identifying the drivers.

ljalves, 2015-03-26:
Wierd that they have replaced the saa7160 with an FPGA (usually FPGAs are more expensive and have higher power consumptions that ASICs).

Well, in theory it should be very easy to add this card.

bpbastos, 2015-03-26:
Thank you so much @ljalves.

I'm not a developer, but please let me know if I can be of any further help.

malakudi, 2015-04-24:
As far as I know, SAA716X had initialization problems on some newer PCI-E 3.0 motherboards (some were fixed if your BIOS supported setting the PCI-E slot to Gen1 but not all boards support this), so they decided to replace it with a different chip.
Same pci-e chip on TBS6905, the new TBS6985.

crazycat69, 2015-04-24:
TBS 6905 have STV6120/STV910BDB tuner/demod chips.
Maybe later will be relesead TBS 6904 with 4x AV2011/TAS2101.

ljalves, 2015-05-02:
@bpbastos,
I'm closing this issue for now.
If someone decides to donate any of those cards I can then work on adding new drivers.

closed 2015-05-02 by ljalves