ljalves/linux_media #89: Resume problems - TBS6280 - cxd2820r module?

cotman, 2015-04-15:
Hi,

I'm using the latest cut of the drivers, but I'm having some issues with operation on resume.

My (newly installed) tvheadend system has a TBS6981, HVR4000 and a TBS6280, although I'm only using the TBS6280, at the moment.

The drivers themselves seem to work perfectly after a fresh reboot - really impressed with performance.  However, I use S3 when inactive, and on resume the TBS6280 isn't working:

    Apr 15 18:32:25 tv tvheadend[1016]: subscription: 023F: "192.168.1.104 [ generic | XBMC Media Center ]" subscribing on channel "BBC ONE East E", weight: 150, adapter: "Sony CXD2820R : DVB-T #0", network: "Freeview", mux: "745.833MHz", service: "BBC ONE East E", profile="htsp", hostname="192.168.1.104", username="generic", client="XBMC Media Center"
    Apr 15 18:32:30 tv tvheadend[1016]: linuxdvb: Sony CXD2820R : DVB-T #0 - poll TIMEOUT
 
...repeated.

Now - I do have a `sleep.d` custom script, and that currently has the following:

    case "$1" in
    suspend|hibernate)
    service tvheadend stop
    sleep 3
    modprobe -r cx88_dvb cx88_alsa cx8800 cx88xx cx23885 saa716x_budget saa716x_core cx25840
    ;;
    resume|thaw)
    modprobe cx88_dvb cx88_alsa cx23885 saa716x_budget
    sleep 3
    service tvheadend start
    ;;
    esac

Now, from the error, I'm guessing it would be good to add the `cxd2820r` (and maybe `dvb_core` ? ) to that module removal list before sleep.

However, I can't...

With the above modules removed, `lsmod` lists the following:

    root@tv:~# lsmod | grep cxd2820
    Module                  Size  Used by
    cxd2820r               31562  -1 
    dvb_core              126812  1 cxd2820r

However, if I try and remove {{cxc2820r}}, I get this:

    root@tv:~# rmmod cxd2820r 
    rmmod: ERROR: Module cxd2820r is in use

Bit stuck - as I can't see anything else using it.
Trying to force the removal isn't any more successful:

    root@tv:~# rmmod -f cxd2820r 
    rmmod: ERROR: ../libkmod/libkmod-module.c:769 kmod_module_remove_module() could not remove 'cxd2820r': Resource temporarily unavailable
    rmmod: ERROR: could not remove module cxd2820r: Resource temporarily unavailable

Any advice?

The drivers work really well in normal operation, post a cold boot, so this is the one thing stopping me switching this to be my live system...

cotman, 2015-04-18:
OK - to remove one potential issue, I've removed my HVR4000 card.

That means my system is now a TBS6280 and a TBS6981, although I'm only using the TBS6280 at the moment.

As a result, I've updated my `sleep.d` script as follows:

    case "$1" in
    suspend|hibernate)
    service tvheadend stop
    sleep 3
    modprobe -r cx23885 saa716x_budget saa716x_core cx25840
    ;;
    resume|thaw)
    modprobe cx23885 saa716x_budget
    sleep 3
    service tvheadend start
    ;;
    esac

However, I still get the same situation as above - it's not possible to remove `cxd2820r` and `dvb_core`.
`lsmod` output is the same as above.

Just to replicate formally, end to end, I did the following:

* 20:55 BST - rebooted system
* 21:00 BST - watched TV, briefly, to confirm working - all good
* 21:15 BST - system went to sleep via cron/script
* 21:20 BST - system brought out of standby by WOL from XBMC client
* 21:21 BST - tried to watch TV using the TBS6280, but I just get time-outs that repeat

...if I reboot, then everything works fine again, but that seems the only way to resolve the issue.

Logs are below, but there's nothing that jumps out (to my untrained eye):

dmesg: https://gist.github.com/cotman/c8f7ec180edbfb909ed2
syslog: https://gist.github.com/cotman/328ee8b8502a04064ba2

Anything I can do to dig into this further?
Would it be an issue with trying to run two TBS cards?

cotman, 2015-04-18:
I am a tool - please disregard :smile: .

So - it turns out, in all of this, that my sleep script wasn't actually being called at all (I assumed `/var/log/pm-suspend.log` didn't output their names...although now I don't know why I thought that);  I got suspicious, however, when I added a `-v` to the `modprobe` calls and still didn't see anything in any logs.

Now - what I had been doing was creating a symbolic link from the actual sleep script in my home directory to `/etc/pm/sleep.d/` - that seemed a nice idea at the time, as in my home directory the script sat in git repo, that I pushed for safe-keeping (along with other configuration scripts).  Turns out that plain doesn't work though - it just gets ignored.

For reference, the final script I have (actually copied to `/etc/pm/sleep.d/`) just contains the following:

    case "$1" in
    suspend|hibernate)
    service tvheadend stop
    sleep 3
    modprobe -v -r cx23885 saa716x_budget cx25840
    ;;
    resume|thaw)
    modprobe -v cx23885 
    modprobe -v saa716x_budget
    sleep 3
    service tvheadend start
    ;;
    esac

...best of all, it works :smile: 

Apologies for the noise.

ljalves, 2015-05-02:
This kind of noise is welcome! We all learn from our mistakes. 
Glad you solved your issues.
Closing.

closed 2015-05-02 by ljalves