ljalves/linux_media #90: Support 5922 SE

beam, 2015-04-28:
Hello a tried make support for 5922SE based on QBOX22
Commit: https://github.com/beam/linux_media/commit/16500bd7ccd8bdaacc4d52bbd1d6493aac509680

```
[  384.412023] usb 1-3: new high-speed USB device number 9 using ehci-pci
[  384.544762] usb 1-3: New USB device found, idVendor=734c, idProduct=5923
[  384.544766] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  384.544768] usb 1-3: Product: TBS5922S
[  384.544771] usb 1-3: Manufacturer: TBS-Tech
[  384.545201] dvb-usb: found a 'TBS 5922SE DVBS2 USB2.0' in cold state, will try to load a firmware
[  384.545205] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5923.fw'
[  384.545207] tbs5922se: start downloading TBS5922SE firmware
[  384.664007] dvb-usb: found a 'TBS 5922SE DVBS2 USB2.0' in warm state.
[  384.664056] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[  384.664211] DVB: registering new adapter (TBS 5922SE DVBS2 USB2.0)
[  384.664250] usb 1-3: media controller created
[  384.760277] dvb-usb: MAC address: 00:22:ab:c0:39:78
[  384.760472] dvb_register_media_device: media device 'dvb-demux' registered.
[  384.760512] dvb_register_media_device: media device 'dvb-dvr' registered.
[  384.760567] dvb_register_media_device: media device 'dvb-net' registered.
[  384.760604] i2c i2c-4: Added multiplexed i2c bus 5
[  384.760632] i2c i2c-4: Added multiplexed i2c bus 6
[  384.876224] i2c i2c-6: av201x: Airoha Technology AV201x successfully attached
[  384.876226] TBS: TBS5922SE attached.
[  384.876592] TBS: frontend attached
[  384.876596] usb 1-3: DVB: registering adapter 0 frontend 0 (Tmax TAS2101)...
[  384.876636] dvb_register_media_device: media device 'Tmax TAS2101' registered.
[  384.876892] input: IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:1a.7/usb1/1-3/input/input26
[  384.876974] dvb-usb: schedule remote query interval to 150 msecs.
[  384.876977] dvb-usb: TBS 5922SE DVBS2 USB2.0 successfully initialized and connected.
```

Device is attached, but i can't scan any channel..
```
beam@goduntu:media $ w_scan -fs -s S13E0 -D0c
w_scan version 20130331 (compiled for DVB API 5.10)
using settings for 13.0 east Hotbird 6/7A/8
scan type SATELLITE, channellist 69
output format vdr-2.0
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter0/frontend0 -> SATELLITE "Tmax TAS2101": very good :-))

Using SATELLITE frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_
Using DVB API 5.a
frontend 'Tmax TAS2101' supports
INVERSION_AUTO
DVB-S
DVB-S2
FREQ (0.95GHz ... 2.15GHz)
SRATE (1.000MSym/s ... 45.000MSym/s)
using LNB "UNIVERSAL"
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_
trying 'S  f = 10719 kHz V SR = 27500  5/6 0,35  QPSK'
(time: 00:01)
trying 'S  f = 10723 kHz H SR = 29900  3/4 0,35  QPSK'
(time: 00:04)
trying 'S  f = 10758 kHz V SR = 27500  5/6 0,35  QPSK'
(time: 00:08)
trying 'S2 f = 10775 kHz H SR = 29900  3/4 0,35  8PSK'
(time: 00:11) ^C
ERROR: interrupted by SIGINT, dumping partial result...
dumping lists (0 services)
Done.
```

With TBS drivers i can scan, any idea what i can do?
Thanks for any help or hint.

OS: Ubuntu 14.04 (Linux goduntu 3.13.0-24-generic #47-Ubuntu SMP Fri May 2 23:30:00 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux)

ljalves, 2015-04-28:
Hi there! Nice to seem more people contributing to this project.

Your problem should just be the init sequence.
Try this init and let me know if it works:

static struct tas2101_config tbs5922_cfg = {
	.i2c_address   = 0x68,
	.id            = ID_TAS2100,
	.reset_demod   = NULL,
	.lnb_power     = NULL,
	.init          = {0x67, 0x45, 0xba, 0x23, 0x01, 0x98, 0x33},
};

beam, 2015-04-28:
Nope, this doesn't help :/

beam, 2015-04-28:
FYI, after testing opensource drivers i have to tuner unplug from power to make tbs drivers works, without tuner even original drivers doesn't work.

ljalves, 2015-04-28:
Try to set the .id to ID_TAS2101

.id = ID_TAS2101,

beam, 2015-04-28:
Wau, much better!
beam@goduntu:~ $  w_scan -fs -s S13E0 -D0c
w_scan version 20130331 (compiled for DVB API 5.10)
using settings for 13.0 east Hotbird 6/7A/8
scan type SATELLITE, channellist 69
output format vdr-2.0
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
	/dev/dvb/adapter0/frontend0 -> SATELLITE "Tmax TAS2101": very good :-))

Using SATELLITE frontend (adapter /dev/dvb/adapter0/frontend0)
-_-_-_-_ Getting frontend capabilities-_-_-_-_ 
Using DVB API 5.a
frontend 'Tmax TAS2101' supports
INVERSION_AUTO
DVB-S
DVB-S2
FREQ (0.95GHz ... 2.15GHz)
SRATE (1.000MSym/s ... 45.000MSym/s)
using LNB "UNIVERSAL"
-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_-_ 
trying 'S  f = 10719 kHz V SR = 27500  5/6 0,35  QPSK'
(time: 00:02) (time: 00:04) signal ok:
	S  f = 10719 kHz V SR = 27500  5/6 0,35  QPSK
Info: no data from NIT(actual)
trying 'S  f = 10723 kHz H SR = 29900  3/4 0,35  QPSK'
(time: 00:17) 
trying 'S  f = 10758 kHz V SR = 27500  5/6 0,35  QPSK'
(time: 00:20) (time: 00:22) signal ok:
	S  f = 10758 kHz V SR = 27500  5/6 0,35  QPSK
^C
ERROR: interrupted by SIGINT, dumping partial result...
dumping lists (0 services)
Done.

beam, 2015-04-28:
Looks like Network search doesn't work, but got LOCK is awsome!

ljalves, 2015-04-28:
Great!
It would be nice if you could send a pull request to merge your work in here.

beam, 2015-04-28:
PR after some test

ljalves, 2015-04-28:
By the way, the init serquence that I sent earlier is the correct one for your card.
You have to use it also.

beam, 2015-04-28:
I do

beam, 2015-04-28:
https://github.com/beam/linux_media/commit/13b812b5bee44e08c6bb842e72fb03c0ea9d7eb5

beam, 2015-04-28:
Got lock, but no data :/
```
2015-04-28 22:39:56.413 [  DEBUG]:mpegts: 23.5 - adding mux Multiplex [onid:FFFF tsid:FFFF] in 23.5 to scan queue weight 5 flags 0100
2015-04-28 22:40:06.000 [  DEBUG]:mpegts: 12525V in 23.5 - add raw service
2015-04-28 22:40:06.000 [  DEBUG]:service: 2: 12525V in 23.5 si 0x14a9f90 Tmax TAS2101 : DVB-S #0 weight 0 prio 0 error 0
2015-04-28 22:40:06.000 [   INFO]:mpegts: 12525V in 23.5 - tuning on Tmax TAS2101 : DVB-S #0
2015-04-28 22:40:06.340 [  DEBUG]:linuxdvb: Tmax TAS2101 : DVB-S #0 - starting 12525V in 23.5
2015-04-28 22:40:06.340 [  DEBUG]:mpegts: 12525V in 23.5 - started
2015-04-28 22:40:06.340 [   INFO]:opentv-skyit: registering mux 12525V in 23.5
2015-04-28 22:40:06.343 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0012 (18) [20/0x15741e0]
2015-04-28 22:40:06.343 [  DEBUG]:eit: installed table handlers
2015-04-28 22:40:06.343 [  DEBUG]:eit: grab started
2015-04-28 22:40:06.343 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0BBA (3002) [16/0x1575650]
2015-04-28 22:40:06.343 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0BBB (3003) [16/0x1576ac0]
2015-04-28 22:40:06.343 [  DEBUG]:uk_freesat: installed table handlers
2015-04-28 22:40:06.343 [  DEBUG]:uk_freesat: grab started
2015-04-28 22:40:06.343 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0039 (57) [16/0x1577f30]
2015-04-28 22:40:06.343 [  DEBUG]:viasat_baltic: installed table handlers
2015-04-28 22:40:06.343 [  DEBUG]:viasat_baltic: grab started
2015-04-28 22:40:06.343 [  DEBUG]:mpegts: 12525V in 23.5 - open PID tables subscription [0042/0x14ab3c0]
2015-04-28 22:40:06.343 [   INFO]:subscription: 0002: "scan" subscribing to mux "12525V", weight: 5, adapter: "Tmax TAS2101 : DVB-S #0", network: "23.5", service: "Raw PID Subscription"
2015-04-28 22:40:06.700 [  DEBUG]:linuxdvb: Tmax TAS2101 : DVB-S #0 - status    GOOD (SIGNAL | CARRIER | VITERBI | SYNC | SIGNAL)
2015-04-28 22:40:06.700 [  DEBUG]:linuxdvb: Tmax TAS2101 : DVB-S #0 - locked
2015-04-28 22:40:06.700 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0000 (0) [20/0x15793a0]
2015-04-28 22:40:06.700 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0001 (1) [16/0x157a810]
2015-04-28 22:40:06.700 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0010 (16) [16/0x157bc80]
2015-04-28 22:40:06.700 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0011 (17) [20/0x157d0f0]
2015-04-28 22:40:06.700 [  DEBUG]:mpegts: 12525V in 23.5 - open PID 0011 (17) [16/0x157e560]
2015-04-28 22:40:07.451 [WARNING]:linuxdvb: Tmax TAS2101 : DVB-S #0 - poll TIMEOUT
2015-04-28 22:40:08.412 [WARNING]:linuxdvb: Tmax TAS2101 : DVB-S #0 - retune nodata
2015-04-28 22:40:08.780 [  DEBUG]:linuxdvb: Tmax TAS2101 : DVB-S #0 - status    GOOD (SIGNAL | CARRIER | VITERBI | SYNC | SIGNAL)
2015-04-28 22:40:15.109 [WARNING]:linuxdvb: Tmax TAS2101 : DVB-S #0 - poll TIMEOUT
2015-04-28 22:40:15.484 [WARNING]:linuxdvb: Tmax TAS2101 : DVB-S #0 - retune nodata
2015-04-28 22:40:15.928 [  DEBUG]:linuxdvb: Tmax TAS2101 : DVB-S #0 - status    GOOD (SIGNAL | CARRIER | VITERBI | SYNC | SIGNAL)
2015-04-28 22:40:16.000 [  DEBUG]:service: 12525V in 23.5: Status changed to [Graceperiod expired] [Data timeout] 
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0011 (17) [20/0x157d0f0]
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0010 (16) [16/0x157bc80]
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0001 (1) [16/0x157a810]
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0000 (0) [20/0x15793a0]
2015-04-28 22:40:16.000 [   INFO]:mpegts: 12525V in 23.5 - scan no data, failed
2015-04-28 22:40:16.000 [   INFO]:subscription: 0002: "scan" unsubscribing
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID tables subscription [0042/0x14ab3c0]
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - stopping mux
2015-04-28 22:40:16.000 [  DEBUG]:linuxdvb: Tmax TAS2101 : DVB-S #0 - stopping 12525V in 23.5
2015-04-28 22:40:16.000 [  DEBUG]:linuxdvb: Tmax TAS2101 : DVB-S #0 - stopped dvr thread
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0011 (17) [16/0x157e560]
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0012 (18) [20/0x15741e0]
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0039 (57) [16/0x1577f30]
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0BBA (3002) [16/0x1575650]
2015-04-28 22:40:16.000 [  DEBUG]:mpegts: 12525V in 23.5 - close PID 0BBB (3003) [16/0x1576ac0]
2015-04-28 22:40:16.002 [  DEBUG]:mpegts: 23.5 - removing mux 12525V in 23.5 from scan queue
2015-04-28 22:40:16.002 [  DEBUG]:mpegts: 23.5 - adding mux 12525V in 23.5 to scan queue weight 5 flags 0100
2015-04-28 22:40:16.002 [  DEBUG]:epggrab: grab done for 12525V in 23.5 (stolen)
2015-04-28 22:40:16.002 [  DEBUG]:mpegts: 23.5 - removing mux 12525V in 23.5 from scan queue
```

beam, 2015-04-28:
FYI, original drivers now work without device restart (unplug and plug power)

beam, 2015-04-29:
Any idea? @crazycat69? Anything i can do to help?

ljalves, 2015-04-29:
Having lock means tunning should be fine.
No data means that TS stream path is broken somewhere.
I'll take a look once I have some time.

trsqr, 2015-04-29:
The lock indicates that the demodulator has locked in and the tuner has tuned correctly. Now the demodulator must pass the demodulated transport stream (TS) to the USB bridge.

Typical cases I've seen where no data comes in:
- the demod is configured incorrectly (TS mode incorrect or other TS related parameters) and thus the USB bridge does not receive the TS as expected
- the USB bridge is configured incorrectly (TS related parameters not correct)
- the sync byte is something else than what the kernel is expecting (0x47)
- there's some other component that needs initializing between the demod and the USB bridge (such as a CI module)

ljalves, 2015-04-30:
Just noticed that the init sequence on the tbs5922se has 2 extra register read/writes.
Please try adding the following:
eeb8979ad82e8011df5ad840c21421ff00052017

beam, 2015-04-30:
BINGO!

closed 2015-04-30 by beam