ljalves/linux_media #93: Installation Procedure

Farstucker, 2015-05-04:
Hiya

I am relatively new to Linux.

I am running Ubuntu 12.04. I recently upgraded the kernel to 3.16 and have decided to use this driver instead. I have 2x TBS 6891 and 1x TBS 6281. When i installed this driver i could only see the 6281 being picked up and it was showing in TvHeadEnd. However, dmesg reported missing firmware and the card did not work. Next, i then built and installed the v4l drivers and now both my 6891 cards work (Recognized as Conexant cards) and now i have each of them working properly. But now my 6281 has dissapeared. dmesg reports the following:

[    8.539191] saa716x_budget: Unknown symbol cx24117_get_i2c_adapter (err 0)
[    8.539232] saa716x_budget: disagrees about version of symbol cxd2820r_attach
[    8.539234] saa716x_budget: Unknown symbol cxd2820r_attach (err -22)

What is the proper procedure for this driver. Am i supposed to install this and v4l?

Thanks

Will

Xxyzzy, 2015-05-05:
Remove the in-tree media kernel modules:
  rm -rf /lib/modules/3.16/kernel/drivers/media
Install the v4l-drivers again:
  make install
Then reboot.
Firmware for the 6281 available here:
  https://github.com/OpenELEC/dvb-firmware/tree/master/firmware

Farstucker, 2015-05-06:
Hiya

Thanks for your reply.

After re-installing the drivers, the TBS 6281 is still not showing in TvHeadEnd, however i now get a different problem showing after using dmesg.

[    8.183550] 	1555f3bf5cc172e7d23c2b8db10d656d15bec13e [media] saa7164: fix compiler warning
[    8.183550] 	ebf984bb151e9952cccd060d3aba0b4d30a87e81 [media] v4l2: replace s_mbus_fmt by set_fmt in bridge drivers
[    8.183550] 	5eab4983777ba913efb42dd0f177b0dcdbd4f3d9 [media] v4l2: replace try_mbus_fmt by set_fmt in bridge drivers

I've done a bit of googling and it would seem i have to install a patch??

Thanks

Will

ljalves, 2015-05-11:
@Farstucker 
I see no problem in that dmesg paste.
It's just the "git log" of some git commits.

Farstucker, 2015-05-14:
Hiya

In the end i got it working by installing the latest TBS drivers. Everything now appears to be working OK...

Thanks

closed 2015-05-18 by ljalves