ljalves/linux_media #95: 2x TBS installed high BER on one and the other not working?

JohnnySSH, 2015-05-09:
Ok this is strange as I had the same issue running the "Closed Source Driver"....

I'm running TVHeadend Git on Arch Linux kernel version: 4.0.1-1-ARCH #1 SMP PREEMPT Wed Apr 29 12:00:26 CEST 2015 x86_64 GNU/Linux

Reading through the guide I ran this: rm -rf /lib/modules/`uname -r`/kernel/drivers/media/*

to clean out the old modules.

Then performed the install as suggested: make rmmod , make install

the driver built fine and installed fine!

02:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
        Subsystem: Device 6281:0001
        Kernel driver in use: SAA716x Budget
        Kernel modules: saa716x_budget

04:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
        Subsystem: Device 6991:0002
        Kernel driver in use: SAA716x Budget
        Kernel modules: saa716x_budget

I have one 6991SE DVB-S/S2 card and one 6281 DVB-T/T2 card in my system.

Running: lsdvb this shows up:

# lsdvb

                lsdvb: Simple utility to list PCI/PCIe DVB devices
                Version: 0.0.4
                Copyright (C) Manu Abraham

SAA716x Budget (1131:7160 6281:1) on PCI Domain:0 Bus:2 Device:0 Function:0
        DEVICE:0 ADAPTER:0 ERROR: Open frontend0 failed

I'm not sure what to do at this point! Both devices use the same IRQ:

cat /proc/interrupts gives this:

 16:         28          1          2          0  IR-IO-APIC  16-fasteoi   ehci_hcd:usb1, SAA716x Core, SAA716x Core

Could anyone suggest anything?

Many thanks!

JohnnySSH, 2015-05-09:
Ok, so looks like I didn't have the firmware installed....

I found it here:

http://article.gmane.org/gmane.linux.drivers.video-input-infrastructure/79636/match=dvb+tuner+si2158+a20+01+fw

but upon reboot this is what dmesg is giving me:

# dmesg |grep dvb
[    2.021741] DVB: registering new adapter (SAA716x dvb adapter)
[    2.184620] DVB: registering new adapter (SAA716x dvb adapter)
[    2.362256] DVB: registering new adapter (SAA716x dvb adapter)
[    2.467425] DVB: registering new adapter (SAA716x dvb adapter)
[   76.346535] si2168 0-0064: Direct firmware load for dvb-demod-si2168-b40-01.fw failed with error -2
[   76.346541] si2168 0-0064: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[   76.346542] si2168 0-0064: firmware file 'dvb-demod-si2168-02.fw' not found
[   76.349105] si2157 4-0060: Direct firmware load for dvb-tuner-si2158-a20-01.fw failed with error -2
[   76.349106] si2157 4-0060: firmware file 'dvb-tuner-si2158-a20-01.fw' not found
[   76.352682] si2168 0-0064: Direct firmware load for dvb-demod-si2168-b40-01.fw failed with error -2
[   76.352686] si2168 0-0064: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[   76.352687] si2168 0-0064: firmware file 'dvb-demod-si2168-02.fw' not found
[   76.352883] si2168 1-0064: Direct firmware load for dvb-demod-si2168-b40-01.fw failed with error -2
[   76.352887] si2168 1-0064: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[   76.352887] si2168 1-0064: firmware file 'dvb-demod-si2168-02.fw' not found
[   76.354992] si2157 3-0060: Direct firmware load for dvb-tuner-si2158-a20-01.fw failed with error -2
[   76.354993] si2157 3-0060: firmware file 'dvb-tuner-si2158-a20-01.fw' not found
[   76.355787] si2157 4-0060: Direct firmware load for dvb-tuner-si2158-a20-01.fw failed with error -2
[   76.355788] si2157 4-0060: firmware file 'dvb-tuner-si2158-a20-01.fw' not found
[   76.358386] si2168 1-0064: Direct firmware load for dvb-demod-si2168-b40-01.fw failed with error -2
[   76.358390] si2168 1-0064: Direct firmware load for dvb-demod-si2168-02.fw failed with error -2
[   76.358391] si2168 1-0064: firmware file 'dvb-demod-si2168-02.fw' not found
[   76.360942] si2157 3-0060: Direct firmware load for dvb-tuner-si2158-a20-01.fw failed with error -2
[   76.360942] si2157 3-0060: firmware file 'dvb-tuner-si2158-a20-01.fw' not found

I added the firmware into /lib/firmware

JohnnySSH, 2015-05-09:
Looks like I may have located the firmware: https://github.com/OpenELEC/dvb-firmware/tree/master/firmware

...just testing now!

JohnnySSH, 2015-05-09:
Managed to locate the proper firmware here: http://palosaari.fi/linux/v4l-dvb/firmware/

all loads fine!

JohnnySSH, 2015-05-09:
Can confirm that have picture now :-)

Unfortunately the other card still has high bit error rate, which is odd considering that it was working before??

JohnnySSH, 2015-05-09:
Weird!! The BER just disappeared and now am able to "lock" onto channels using the 6991SE DVB-S network??

JohnnySSH, 2015-05-10:
I have figured something out in context to the 6991SE card! It seems that only one tuner can work?? I have an F-Type satellite cable plugged into input B so tuner 2 works fine. Tuner 1 however gives high BER with TVHeadend complaining of "No Input Detected"! I have linked the input to Tuner 2 however it is still coming up with the same issue....

The 6281 after firmware is completely stable, it seems so far!! :-)

ljalves, 2015-05-11:
@JohnnySSH,

I'll double check the 6991SE init when I have the time. I might have missed something for the second tuner.

ljalves, 2015-05-11:
@JohnnySSH,
Might seem a stupid question but, how are you connecting the 6991SE card?

From your description I understand that:
 - you have 1 sat cable with an F type connector.
 - you connect the cable to input A and it works ok
 - you disconnect the cable from input A and connect the cable to input B and you have high BER

Is this correct?

JohnnySSH, 2015-05-11:
Thanks for response!

The 6991SE is connected as such:

Sat cable with F-type connector in Input B; the cable stays in Input B.

I am assuming that like the 6281, the 6991SE can use a single input for both tuners?? Perhaps it was my bad and I totally misunderstood, in that 2x Sat cables are required??

JohnnySSH, 2015-05-11:
Using method described I get good signal... on (other) tuner, 90%+ but BER is high.....

trsqr, 2015-05-11:
Satellite TV works in a different way than a DVB-T/C system. The tuner will need to set different voltage etc. to control the LNB - you cannot split a satellite signal. In other words, you will need to connect satellite cable from the both inputs of your DVB-S card to the satellite equipment (twin LNB or such).

JohnnySSH, 2015-05-11:
Oh oops so this was totally my misunderstanding!!!

Currently the setup is that I have a subscription box connected to the LNB, the Loop Out is connected to Input B....

JohnnySSH, 2015-05-11:
So to make this work I'd need a 2-way splitter?

ljalves, 2015-05-11:
No each sat input needs it's own cable up to the LNB or a multiswitch.
I'm not even sure if that Loop Out wont damage your card if it dowesn't have a DC block.

The sat inputs in your card will actually output voltage on the cables to power up the LNB (and do other  stuff like polarization).
Connecting 2 inputs to the same cable will make the voltages "colide" and you may end up burning one of the card inputs.

ljalves, 2015-05-11:
You can however connect 2 inputs to the same cable, but you will only be able to use the same polarization and band on both at the same time.

JohnnySSH, 2015-05-11:
I see.... so plugging the LNB Loop Out is not a good idea? I mean I can get channels fine.

But if it will damage the card then I will need to look at other method.

Also the dish is motorized but due to limitation on paid box can only use the dish as "universal LNB" as it will only connect to one satellite.

JohnnySSH, 2015-05-11:
Thanks for taking me through this btw!!

JohnnySSH, 2015-05-11:
Ok so this explains loop-out: http://forums.digitalspy.co.uk/showthread.php?t=1663645

JohnnySSH, 2015-05-11:
Since I got worried about "burning" the card out... I decided to buy a F-type switch;  this will allow me to use either the card or box by pressing either A or B button.

Then I should get motor control back too with card.....

ljalves, 2015-05-18:
Issue solved... Closing.

closed 2015-05-18 by ljalves