ljalves/linux_media #96: cx23885 anomaly

bas-t, 2015-05-13:
Hi Luis!

My tbs 6281/6285 cards still work like a dream, thanks again.
However, I also own a couple of DVBSky T982 cards, using a cx23885 bridge chip and si2157/si2168 tuner/demod chips. Of course I can get them working in a joint setup with my TBS cards, but not when I use your repo. So I guess something weird is going on here. Using your repo, the cx23885 module does not even load at all (compiled against a Debian Jessie stock kernel)

Not a biggy for me, just thought you should know.

Cheers,
Tycho.

RayCic, 2015-05-14:
It looks I have same problem.
I own cx23885 based TBS-6981 and when system boots I receive following error messages:
```
videobuf2_dma_sg: Unknown symbol dma_buf_export (err 0)
videobuf2_vmalloc: Unknown symbol dma_buf_export (err 0)
```
If I try to load cx23885 module manually I receive:
```
videobuf2_dma_sg: Unknown symbol dma_buf_export (err 0)
```
Tested on kernel 3.18.11 & 4.0.2 with latest ljalves linux_media tree
Kernel's built in driver works without problems but I need TBS-6285 support

MikeB2013, 2015-05-14:
This a bug and has been fixed upstream, see 
http://git.linuxtv.org/cgit.cgi/media_build.git/commit/?id=eda2744ce5711f99e2b37f2484e2727d6462e7c0

Mike

bas-t, 2015-05-14:
@MikeB2013 Thanks.
The issue will vanish with the next upstream merge, closing.

closed 2015-05-14 by bas-t