ljalves/linux_media #98: PCVC740K ToUcam webcam fails when USB-Live2 plugged in.

g8ecj, 2015-05-16:
Just installed the repo to enable a TBS6281 and a TBS6982 (which both work great!) but I also need svideo capture for an STB hence the USB-Live2. As soon as its plugged in the LED on the webcam illuminates and its video device (/dev/video0) gets taken by the USB-Live2.

Some confusing errors in syslog which may or maybe not relevant:
     videobuf2_vmalloc: Unknown symbol dma_buf_export (err 0)

This is on a new Debian stable (Jessie) install with kernel 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt9-3~deb8u1

All works fine with the standard set of modules (apart from the TurboSight cards of course)..

g8ecj, 2015-05-18:
Looks like its nothing to do with the USB-Live2, I get the same unknown symbol error with both the PWC webcam and a UVC webcam.

bas-t, 2015-05-18:
The unknown symbol error will disappear as soon as Luis pulls upstream updates. See #96

ljalves, 2015-05-18:
Updates done.
Closing issue

closed 2015-05-18 by ljalves