ljalves/linux_media #100: What part of work tree to apply PAL patch for saa7164

g8ecj, 2015-05-28:
I have been using a patch for the last 3-4 years to allow PAL on the analog side of a HVR2200. I wish to apply it to the linux_media tree to allow this card to work alongside a TBS 6281 and TBS6982.

I'm not sure whether to patch in the ~/media or ~/media_build directory having already downloaded, built and installed the tree.

Thanks

ljalves, 2015-05-28:
The saa7164 is supported in the official linux tree so you should send the patches to the linux media mailing list.
Once they get upstreamed, they are merged back to this tree.

closed 2015-05-28 by ljalves

bas-t, 2015-05-29:
@g8ecj However, in case your patch does not make it into the official linux tree, you can still use it by patching your ~/media directory.

@ljalves I guess g8ecj was not sugesting that you should pull the patch into your repo...

g8ecj, 2015-05-29:
Indeed - I tried putting the patch file into backports and running the 'build' script as per the README in media_build but I got errors.
I take it I should patch the copy in .../media/drivers/media/pci/saa7164 and do the build there...

bas-t, 2015-05-29:
You should indeed patch the copy in .../media/drivers/media/pci/saa7164.
Not sure what you mean by "do the build there".
Just build as per wiki: https://github.com/ljalves/linux_media/wiki/Installating
Only difference: before the 'cd media_build' line, you patch whatever you want to patch.

Cheers!

bas-t, 2015-05-29:
This may be confusing, so to get it straight: you patch in the directory you get from
git clone --depth=1 https://github.com/ljalves/linux_media.git -b latest ./media