ljalves/linux_media #101: TBS QBOX22 driver load error

jendik, 2015-05-30:
Hi Luis,

once again thanks a lot for all the work you've done...

I tried to use your latest driver versions with QBOX22 but got following error-(

Any idea? Could you help me please with that?
Thanks a lot
j.

###
sudo modprobe dvb_usb_tbsqbox22

dmesg:
[  256.675441] dvb-usb: found a 'TBS QBOX22 DVBS2 USB2.0' in cold state, will try to load a firmware
[  256.676473] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5922.fw'
[  256.676483] tbsqbox22: start downloading TBSQBOX firmware
[  256.796873] dvb-usb: found a 'TBS QBOX22 DVBS2 USB2.0' in warm state.
[  256.797302] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[  256.797832] dvb-usb: TBS QBOX22 DVBS2 USB2.0 error while loading driver (-23)
[  256.798030] usbcore: registered new interface driver tbsqbox22

lsusb:
Bus 001 Device 003: ID 734c:5922 TBS Technologies China
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x734c TBS Technologies China
  idProduct          0x5922
  bcdDevice            0.01
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

ljalves, 2015-06-09:
Do you have the dvb-usb-tbsqbox-id5922.fw file at /lib/firmware ?

mpmc, 2015-06-10:
Hi both, 

I haven't tried the 5922 in a while on my RPi, so today I decided to try and build the driver on my RPi2 & to my surprise (after a few hiccups) it works perfectly fine!

All, I did was make sure the kernel was up to date & grab the linux source made sure it was proper etc (so the driver builds correctly)

And heres the result!

```
[    7.978246] usb 1-1.5: new high-speed USB device number 5 using dwc_otg
[    8.098835] usb 1-1.5: New USB device found, idVendor=734c, idProduct=5922
[    8.107767] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    8.118183] usb 1-1.5: Product: TBS 5922
[    8.124096] usb 1-1.5: Manufacturer: TBS-Tech
[    8.131528] dvb-usb: found a 'TBS QBOX22 DVBS2 USB2.0' in cold state, will try to load a firmware
[    8.144441] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5922.fw'
[    8.153865] tbsqbox22: start downloading TBSQBOX firmware
[    8.278166] dvb-usb: found a 'TBS QBOX22 DVBS2 USB2.0' in warm state.
[    8.286830] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[    8.299315] DVB: registering new adapter (TBS QBOX22 DVBS2 USB2.0)
[    8.307772] usb 1-1.5: media controller created
[    8.410216] dvb-usb: MAC address: 00:22:ab:c0:20:34
[    8.417981] dvb_register_media_device: media device 'dvb-demux' registered.
[    8.418303] dvb_register_media_device: media device 'dvb-dvr' registered.
[    8.418557] dvb_register_media_device: media device 'dvb-net' registered.
[    8.418760] i2c i2c-0: Added multiplexed i2c bus 1
[    8.425735] i2c i2c-0: Added multiplexed i2c bus 2
[    8.574294] i2c i2c-2: av201x: Airoha Technology AV201x successfully attached
[    8.583455] QBOX22: TBS5922 attached.
[    8.590857] QBOX22: frontend attached
[    8.596343] usb 1-1.5: DVB: registering adapter 0 frontend 0 (Tmax TAS2101)...
[    8.605858] dvb_register_media_device: media device 'Tmax TAS2101' registered.
[    8.606134] dvb-usb: TBS QBOX22 DVBS2 USB2.0 successfully initialized and connected.
[    8.791454] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    8.800170] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   22.818270] usb 1-1.5: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
```
I remember having so many issue with this trying to get it to work on my RPi1 before, so I'm not sure what's changed but I'm tempted to see if it I can get it to run on that too!

It didn't take that long to build either!  =D

Although now I'm stuck between using my Technisat and the TBS 5922! :+1:  The TBS with @ljalves' driver seems to be slightly faster. If it's this good on an RPi1 I might continue to use it instead of the Technisat as thats becoming a collectors piece now (they no longer make pc tuners at all)!

ljalves, 2015-06-11:
@mpmc ,
Thanks for letting me know that it works!

mpmc, 2015-06-11:
@ljalves 

Do you know how I go about integrating your drivers into the Raspberry Pi Kernel? =P

jendik, 2015-06-11:
Hi Luis,

YES, I have it there...

~$ uname -va
Linux jendaserver 3.13.0-45-generic #74-Ubuntu SMP Tue Jan 13 19:36:28 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
~$ ls -lrtah /lib/firmware/dvb-usb-tbsqbox-id5922.fw
-rw-r--r-- 1 root root 8.0K May 30 14:30 /lib/firmware/dvb-usb-tbsqbox-id5922.fw
~$ md5sum /lib/firmware/dvb-usb-tbsqbox-id5922.fw
c88d4b13b573b9c4726c5f4d8e0a9d8f  /lib/firmware/dvb-usb-tbsqbox-id5922.fw

mpmc, 2015-06-11:
@jendik 

I'm getting the same hash too
```
mark@tvpi:~$ md5sum /lib/firmware/dvb-usb-tbsqbox-id5922.fw
c88d4b13b573b9c4726c5f4d8e0a9d8f  /lib/firmware/dvb-usb-tbsqbox-id5922.fw
```

But I'm using a newer kernel with the drivers, so you may want to upgrade & try again, IIRC I used 3.12 the last time I tried on my RPi1 & it completely crashed the Pi.

```
Linux tvpi 3.18.14-v7+ #794 SMP PREEMPT Sun Jun 7 12:50:51 BST 2015 armv7l GNU/Linux
```

jendik, 2015-06-11:
OK...thx for the info... I'll give it a try...

take care
j.

jendik, 2015-07-13:
Hi,

I've updated kernel, but unfortunately the same issue...

Probably smth. bad in my OS...Have you got any idea how to resolve it please?

thx a lot
j.

Linux 3.16.0-43-generic #58~14.04.1-Ubuntu SMP Mon Jun 22 10:21:20 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.2 LTS
Release:        14.04
Codename:       trusty

ljalves, 2016-06-24:
Closing old issue - reopen if needed.

closed 2016-06-24 by ljalves