ljalves/linux_media #102: No MAC address - 2x TBS6981 and TBS6985

similar75, 2015-06-04:
Driver for 6981 loads OK, firmware too. I can lock and watch tv`s. But the devices has no MAC address:
Jun 04 13:50:03: INFO: adapter = 7, device = 0
Jun 04 13:50:03: INFO:     mac = 00:00:00:00:00:00
Jun 04 13:50:03: INFO:     frontend = Conexant CX24117/CX24132
Jun 04 13:50:03: INFO: adapter = 6, device = 0
Jun 04 13:50:03: INFO:     mac = 00:00:00:00:00:00
Jun 04 13:50:03: INFO:     frontend = Conexant CX24117/CX24132
Jun 04 13:50:03: INFO: adapter = 5, device = 0
Jun 04 13:50:03: INFO:     mac = 00:00:00:00:00:00
Jun 04 13:50:03: INFO:     frontend = Conexant CX24117/CX24132
Jun 04 13:50:03: INFO: adapter = 4, device = 0
Jun 04 13:50:03: INFO:     mac = 00:00:00:00:00:00
Jun 04 13:50:03: INFO:     frontend = Conexant CX24117/CX24132
Jun 04 13:50:03: INFO: adapter = 3, device = 0
Jun 04 13:50:03: INFO:     mac = 00:00:00:00:00:00
Jun 04 13:50:03: INFO:     frontend = Tmax TAS2101
Jun 04 13:50:03: INFO: adapter = 2, device = 0
Jun 04 13:50:03: INFO:     mac = 00:00:00:00:00:00
Jun 04 13:50:03: INFO:     frontend = Tmax TAS2101
Jun 04 13:50:03: INFO: adapter = 1, device = 0
Jun 04 13:50:03: INFO:     mac = 00:00:00:00:00:00
Jun 04 13:50:03: INFO:     frontend = Tmax TAS2101
Jun 04 13:50:03: INFO: adapter = 0, device = 0
Jun 04 13:50:03: INFO:     mac = 00:00:00:00:00:00
Jun 04 13:50:03: INFO:     frontend = Tmax TAS2101
Jun 04 13:50:03: INFO: [main] exit

This is the dmesg (TBS6981): http://pastebin.com/MhmFdyBY

The next thing - driver for TBS6985 shows error: http://pastebin.com/0a74XpK6
I trying on stable kernel 4.0.4 (stock and custom) and 4.1.0-RC3 (from this git source) 

Everything works, but mac addresses are important to identify adapters, because sometimes they are messed up on load...

similar75, 2015-06-04:
Same on Debian Jessie with stock (debian) 3.16.0 kernel and without TBS6985 driver error - all works, but no mac addresses.

trsqr, 2015-06-04:
Do you know if the cards have a MAC address stored? Do you get one with the manufacturer-provided drivers?

similar75, 2015-06-04:
Yes, they have. When i use TBS proprietary drivers (closed-source) they have mac addresses and driver displays them on load.

ljalves, 2015-06-09:
@similar75,
I'll add the mac address assignment as soon as I can.