ljalves/linux_media #103: TBS 6281/6285 broken

bas-t, 2015-06-08:
Hi Luis, you don't have it in your repo yet, but with your next pull you'll get:
http://git.linuxtv.org/cgit.cgi/media_tree.git/patch/drivers/media/dvb-frontends/si2168.c?id=d2b72f6482b9a3c57f036c11786a2489dcc81176

This commit prevents TBS 6281/6285 to lock and tune.
My other adapters (DVBSky T982, cx23885,si2168,si2157) continue to work correct.
So I assume something has to be done in saa716x regarding the i2c logic.
Unfortunately, I know nothing about i2c logic. Can you fix this?

ljalves, 2015-06-09:
Hi,

I took a quick look at the patch, but I see no reason why it should break the i2c handling code.
What behavior do you get when applying the patch?

bas-t, 2015-06-09:
Well, as I stated: no lock, no tune.
Nothing in dmesg or other logs that sheds any light on this.

bas-t, 2015-06-15:
Not sure what caused the malfunction yet, but you are right, i2c handling code is ok.
Closing.

closed 2015-06-15 by bas-t

bas-t, 2015-06-26:
I don't know how I managed to screw up, that'll probably remain a mystery.
The good news: still going strong with TBS 6281/6285 in linux 4.1