ljalves/linux_media #104: TBS6280 Black Screen

FosterSM, 2015-06-24:
I am trying to get the latest saa716x_budget driver working on my TB6280. I have downloaded and built the code using the following script:

#! /bin/sh
cd /tmp
git clone git://linuxtv.org/media_build.git
git clone --depth=1 https://github.com/ljalves/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make distclean
make
make install
exit $?

The kernel I am using is the latest version 4.1

I get the following modules loaded with:

modprobe saa716x_budget int_type=1

saa716x_budget         21672  2 
tas2101                 7532  1 saa716x_budget
cxd2820r               13144  1 saa716x_budget
mb86a16                17438  1 saa716x_budget
cx24117                10003  1 saa716x_budget
saa716x_core           51360  1 saa716x_budget
dvb_core               72870  3 saa716x_core,saa716x_budget,cxd2820r

I get the following in /dev:
/dev/dvb/:
total 0
drwxr-xr-x 2 root root 120 Jun 24 10:21 adapter0
drwxr-xr-x 2 root root 120 Jun 24 10:21 adapter1

/dev/dvb/adapter0:
total 0
crw-rw---- 1 root video 212, 0 Jun 24 10:21 demux0
crw-rw---- 1 root video 212, 1 Jun 24 10:21 dvr0
crw-rw---- 1 root video 212, 3 Jun 24 10:21 frontend0
crw-rw---- 1 root video 212, 2 Jun 24 10:21 net0

/dev/dvb/adapter1:
total 0
crw-rw---- 1 root video 212, 4 Jun 24 10:21 demux0
crw-rw---- 1 root video 212, 5 Jun 24 10:21 dvr0
crw-rw---- 1 root video 212, 7 Jun 24 10:21 frontend0
crw-rw---- 1 root video 212, 6 Jun 24 10:21 net0

Every thing looks ok but if I try to use the device by running for example:
mplayer -vo vdpau:deint=4 dvb://1@"BBC TWO" -dr -cache 4096 -slave
I get the following error:

MPlayer SVN-r37344-4.9.2 (C) 2000-2014 MPlayer Team
183 audio & 397 video codecs

Playing dvb://1@BBC TWO.
dvb_tune Freq: 522000000
Not able to lock to the signal on the given frequency, timeout: 30
dvb_tune, TUNING FAILED
ERROR, COULDN'T SET CHANNEL  17: Failed to open dvb://1@BBC TWO.

Exiting... (End of file)

I have an earlier version of the saa716x_budget (two years plus old ??) code built into linux 3.14.45 and that is working fine on the same card with the same test. Could you suggest a way forward ? 

Thanks

FosterSM, 2015-06-27:
Turned on debugfs for modules cxd2820r and saa716x_budget and obtained the following trace (partial log of very long trace) when trying to tune to UK BBC2:

Jun 27 17:22:05 Andromeda kernel: [  278.665825] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:05 Andromeda kernel: [  278.666599] i2c i2c-9: cxd2820r_read_status_t: lock=00 00 00 00
Jun 27 17:22:05 Andromeda kernel: [  278.666612] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:05 Andromeda kernel: [  278.667394] i2c i2c-9: cxd2820r_read_status_t: lock=00 00 00 00
Jun 27 17:22:06 Andromeda kernel: [  278.924453] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  278.925222] i2c i2c-9: cxd2820r_read_status_t: lock=00 00 00 00
Jun 27 17:22:06 Andromeda kernel: [  278.925228] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  278.926006] i2c i2c-9: cxd2820r_read_status_t: lock=00 00 00 00
Jun 27 17:22:06 Andromeda kernel: [  279.079794] i2c i2c-9: cxd2820r_get_tune_settings: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.079870] i2c i2c-9: cxd2820r_search: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.079877] i2c i2c-9: cxd2820r_set_frontend: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.080073] i2c i2c-9: cxd2820r_set_frontend_t: frequency=522000000 bandwidth_hz=8000000
Jun 27 17:22:06 Andromeda kernel: [  279.080078] i2c i2c-9: cxd2820r_i2c_gate_ctrl: 1
Jun 27 17:22:06 Andromeda kernel: [  279.082197] i2c i2c-9: cxd2820r_i2c_gate_ctrl: 0
Jun 27 17:22:06 Andromeda kernel: [  279.088490] i2c i2c-9: cxd2820r_set_frontend_t: if_freq=0
Jun 27 17:22:06 Andromeda kernel: [  279.091528] i2c i2c-9: cxd2820r_search: loop=20
Jun 27 17:22:06 Andromeda kernel: [  279.143478] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.144253] i2c i2c-9: cxd2820r_read_status_t: lock=01 10 00 ee
Jun 27 17:22:06 Andromeda kernel: [  279.144258] i2c i2c-9: cxd2820r_search: loop=19
Jun 27 17:22:06 Andromeda kernel: [  279.196757] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.197543] i2c i2c-9: cxd2820r_read_status_t: lock=03 10 03 ee
Jun 27 17:22:06 Andromeda kernel: [  279.197548] i2c i2c-9: cxd2820r_search: loop=18
Jun 27 17:22:06 Andromeda kernel: [  279.250114] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.251568] i2c i2c-9: cxd2820r_read_status_t: lock=13 10 08 10
Jun 27 17:22:06 Andromeda kernel: [  279.251573] i2c i2c-9: cxd2820r_search: loop=17
Jun 27 17:22:06 Andromeda kernel: [  279.303323] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.305066] i2c i2c-9: cxd2820r_read_status_t: lock=03 10 08 10
Jun 27 17:22:06 Andromeda kernel: [  279.305070] i2c i2c-9: cxd2820r_search: loop=16
Jun 27 17:22:06 Andromeda kernel: [  279.356720] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.358475] i2c i2c-9: cxd2820r_read_status_t: lock=03 10 08 10
Jun 27 17:22:06 Andromeda kernel: [  279.358480] i2c i2c-9: cxd2820r_search: loop=15
Jun 27 17:22:06 Andromeda kernel: [  279.409964] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.411707] i2c i2c-9: cxd2820r_read_status_t: lock=03 10 08 10
Jun 27 17:22:06 Andromeda kernel: [  279.411712] i2c i2c-9: cxd2820r_search: loop=14
Jun 27 17:22:06 Andromeda kernel: [  279.463279] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.465022] i2c i2c-9: cxd2820r_read_status_t: lock=13 10 07 10
Jun 27 17:22:06 Andromeda kernel: [  279.465027] i2c i2c-9: cxd2820r_search: loop=13
Jun 27 17:22:06 Andromeda kernel: [  279.516628] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.518371] i2c i2c-9: cxd2820r_read_status_t: lock=03 10 08 10
Jun 27 17:22:06 Andromeda kernel: [  279.518376] i2c i2c-9: cxd2820r_search: loop=12
Jun 27 17:22:06 Andromeda kernel: [  279.569899] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.571643] i2c i2c-9: cxd2820r_read_status_t: lock=15 10 09 10
Jun 27 17:22:06 Andromeda kernel: [  279.571648] i2c i2c-9: cxd2820r_search: loop=11
Jun 27 17:22:06 Andromeda kernel: [  279.623232] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.624975] i2c i2c-9: cxd2820r_read_status_t: lock=15 10 09 10
Jun 27 17:22:06 Andromeda kernel: [  279.624980] i2c i2c-9: cxd2820r_search: loop=10
Jun 27 17:22:06 Andromeda kernel: [  279.676492] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.678236] i2c i2c-9: cxd2820r_read_status_t: lock=15 10 09 10
Jun 27 17:22:06 Andromeda kernel: [  279.678241] i2c i2c-9: cxd2820r_search: loop=9
Jun 27 17:22:06 Andromeda kernel: [  279.729838] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:06 Andromeda kernel: [  279.731580] i2c i2c-9: cxd2820r_read_status_t: lock=15 10 09 10
Jun 27 17:22:06 Andromeda kernel: [  279.731585] i2c i2c-9: cxd2820r_search: loop=8
Jun 27 17:22:07 Andromeda kernel: [  279.783092] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:07 Andromeda kernel: [  279.784835] i2c i2c-9: cxd2820r_read_status_t: lock=15 10 09 10
Jun 27 17:22:07 Andromeda kernel: [  279.784840] i2c i2c-9: cxd2820r_search: loop=7
Jun 27 17:22:07 Andromeda kernel: [  279.836399] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:07 Andromeda kernel: [  279.838143] i2c i2c-9: cxd2820r_read_status_t: lock=05 10 09 10
[... cut ...]
Jun 27 17:22:07 Andromeda kernel: [  280.243363] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:07 Andromeda kernel: [  280.245107] i2c i2c-9: cxd2820r_read_status_t: lock=15 00 09 10
Jun 27 17:22:07 Andromeda kernel: [  280.255238] i2c i2c-9: cxd2820r_read_status: delsys=3
Jun 27 17:22:07 Andromeda kernel: [  280.256981] i2c i2c-9: cxd2820r_read_status_t: lock=15 00 09 10
Jun 27 17:22:07 Andromeda kernel: [  280.259508] i2c i2c-9: cxd2820r_search: delsys=3
Jun 27 17:22:07 Andromeda kernel: [  280.259513] i2c i2c-9: cxd2820r_sleep_t
root@Andromeda:~#

FosterSM, 2015-06-29:
With further investigation and code instrumentation I find that the tda18212 tuner is failing with error -22. It appears that the call to regmap_bulk_write is expecting the value of map->format.parse_inplace to be non zero:
[   66.376266] tda18212 9-0060: delivery_system=16 frequency=522000000 bandwidth_hz=8000000
[   66.378384] regmap_bulk_write: map->bus = 0xf8431460, map->format.parse_inplace = 0
[   66.378389] tda18212 9-0060: regmap_bulk_write returns -22
[   66.378393] tda18212 9-0060: failed=-22

What value should I get ?

Thanks

FosterSM, 2015-06-29:
Just for interest I replaced the regmap_bulk_write call with a call to the following code added to tda18212.c:

static int tda18212_wr_regs(struct regmap *map, u8 reg, u8 *val, int len)
{
    int ret,i;
    for(i=0;i<len;i++)
    {
        ret = regmap_write(map, reg , val[i]);
        if (ret)
        {
            printk(KERN_INFO "tda18212_wr_regs error %d, i= %d\n",ret,i);
            goto error;
        }
    }
error:
    return ret;
}

My tbs6280 is now working with this hack in place.

FosterSM, 2015-06-29:
My tbs6280 in not working the card appears to be remembering the tuning setup from a previous boot to the kernel with the older software. The tda18212 driver reports as follows when trying to run my test after a power down reboot.

[  400.617982] tda18212 9-0060: delivery_system=3 frequency=522000000 bandwidth_hz=8000000
[  402.196520] tda18212 9-0060: delivery_system=16 frequency=522000000 bandwidth_hz=8000000
[  404.852800] tda18212 9-0060: delivery_system=3 frequency=522000000 bandwidth_hz=8000000
[  406.434089] tda18212 9-0060: delivery_system=16 frequency=522000000 bandwidth_hz=8000000
[  409.090474] tda18212 9-0060: delivery_system=3 frequency=522000000 bandwidth_hz=8000000
[  410.671702] tda18212 9-0060: delivery_system=16 frequency=522000000 bandwidth_hz=8000000
[  413.328046] tda18212 9-0060: delivery_system=3 frequency=522000000 bandwidth_hz=8000000
[  414.909330] tda18212 9-0060: delivery_system=16 frequency=522000000 bandwidth_hz=8000000
[  417.565664] tda18212 9-0060: delivery_system=3 frequency=522000000 bandwidth_hz=8000000
[  419.146995] tda18212 9-0060: delivery_system=16 frequency=522000000 bandwidth_hz=8000000
[  421.803260] tda18212 9-0060: delivery_system=3 frequency=522000000 bandwidth_hz=8000000
[  423.384595] tda18212 9-0060: delivery_system=16 frequency=522000000 bandwidth_hz=8000000
[  426.040925] tda18212 9-0060: delivery_system=3 frequency=522000000 bandwidth_hz=8000000
[  427.622211] tda18212 9-0060: delivery_system=16 frequency=522000000 bandwidth_hz=8000000

FosterSM, 2015-07-01:
After further investigation I have found that the following patch does allow my tbs6280 to initialise correctly and tune to channels:

--- /usr/src/linux-4.1.1.old/drivers/base/regmap/regmap-i2c.c	2015-06-22 06:05:43.000000000 +0100
+++ /usr/src/linux-4.1.1.new/drivers/base/regmap/regmap-i2c.c	2015-07-01 08:38:55.376924285 +0100
@@ -47,9 +47,15 @@
 	return i2c_smbus_write_byte_data(i2c, reg, val);
 }
 
+static int regmap_i2c_write(void *context, const void *data, size_t count);
+
+static int regmap_i2c_read(void *context, const void *reg, size_t reg_size, void *val, size_t val_size);
+
 static struct regmap_bus regmap_smbus_byte = {
 	.reg_write = regmap_smbus_byte_reg_write,
 	.reg_read = regmap_smbus_byte_reg_read,
+	.write = regmap_i2c_write,
+	.read = regmap_i2c_read,
 };
 
 static int regmap_smbus_word_reg_read(void *context, unsigned int reg,

The regmap_bulk_write call in the tda81212 code now appears to work as intended.

timmydog, 2015-07-08:
Hi FosterSM,

Firstly - nice work in finding a fix for this - I've had this issue for a bit but am also running an older kernel.

The patch appears to have got a bit mangled after being pasted in here by containing some valid markdown which has been rendered as a bulleted list.

Could you re-post the text but surround it in 3 backticks (see https://help.github.com/articles/markdown-basics/#multiple-lines) so I can copy and paste it, or create a pull-request or something?

I'll try and make some time (I don't have much) to download the kernel source and test it if you can.

Once again, thanks,

Tim

FosterSM, 2015-07-09:
Ok here is my patch again.

```
diff -uNr linux-4.1.old/drivers/base/regmap/regmap-i2c.c linux-4.1.new/drivers/base/regmap/regmap-i2c.c
--- linux-4.1.old/drivers/base/regmap/regmap-i2c.c	2015-06-22 06:05:43.000000000 +0100
+++ linux-4.1.new/drivers/base/regmap/regmap-i2c.c	2015-07-01 18:03:01.856758716 +0100
@@ -47,9 +47,15 @@
 	return i2c_smbus_write_byte_data(i2c, reg, val);
 }
 
+static int regmap_i2c_write(void *context, const void *data, size_t count);
+
+static int regmap_i2c_read(void *context, const void *reg, size_t reg_size, void *val, size_t val_size);
+
 static struct regmap_bus regmap_smbus_byte = {
 	.reg_write = regmap_smbus_byte_reg_write,
 	.reg_read = regmap_smbus_byte_reg_read,
+	.write = regmap_i2c_write,
+	.read = regmap_i2c_read,
 };
 
 static int regmap_smbus_word_reg_read(void *context, unsigned int reg,
```

jdenson12, 2015-09-10:
I've just created a pull request to fix this - there's no need to hack regmap like that. RegMap is incorrectly trying to use SMBUS and it can't use that for more that a single write - hence the lack of .write and .read functions, and the -EINVAL return from bulk_write. Tell it that I2C is available and the bulk_write call will work fine.

    diff --git a/drivers/media/pci/saa716x/saa716x_i2c.c b/drivers/media/pci/saa716x/saa716x_i2c.c index 255c639..d65edf8 100644
    --- a/drivers/media/pci/saa716x/saa716x_i2c.c
    +++ b/drivers/media/pci/saa716x/saa716x_i2c.c
    @@ -610,7 +610,7 @@ bail_out:
 
     static u32 saa716x_i2c_func(struct i2c_adapter *adapter)
     {
    -       return I2C_FUNC_SMBUS_EMUL;
    +       return I2C_FUNC_I2C | I2C_FUNC_SMBUS_EMUL;
     }

ljalves, 2015-09-10:
Thanks @jdenson12 - PR merged.
Marking as fixed.

closed 2015-09-10 by ljalves