ljalves/linux_media #105: TBS 6981 don't have "SNR" status in Tvheadend

zinonino, 2015-06-26:
I use firmware from your WIKI: https://github.com/ljalves/linux_media/wiki/CX24117-firmware
Debian 64 bit 8.1.0 with 3.16 kernel

[image]

With original driver from TBS have  "SNR" .

Andy-ABTec, 2015-06-26:
Same problem here with the TBS6985 (though I must admin I hadn't noticed until I tried the TBS drivers during my ground-up rebuild on Ubuntu 14.04 64 bit)

nickb937, 2015-07-02:
There is a SNR result but it is between 0000 and 00ff rather than 0000 and ffff. The problem is that tvheadend doesn't scale up the range provided.

fenickb@satellite:~$ femon -a 3
FE: Tmax TAS2101 (DVBS)
status SCVYL | signal ded8 | snr 0074 | ber 00000000 | unc 00000000 | FE_HAS_LOCK
status SCVYL | signal e012 | snr 0074 | ber 00000000 | unc 00000000 | FE_HAS_LOCK

Andy-ABTec, 2015-07-02:
Fair comment, however it doesn't explain why the standard TBS drivers can apparently provide the functionality with no change to my system other than swapping the drivers for my 6985.

Now I'm really confused...

zinonino, 2015-07-02:
@nickb937 
Here also have SNR in TVH with official driver of TBS.
Are this is problem in TVH where can't get signal from V4L driver  ? to know to search prob in TVH

Andy-ABTec, 2015-07-03:
Unfortunately I'm away at the moment and don't have access to my tvheadend box (vpn is down for some reason).

I'll double check and take some screen grabs and report back when I have definite proof one way or the other...

nickb937, 2015-07-03:
The TBS-provided drivers represented SNR in the range 0x0000 - 0xFFFF range; the TBS drivers here use 0x0000 - 0x00FF.  There's no standard on the range: some other TV cards use 0x0000 - 0x00FF and some use 0x0000 - 0xFFFF.

trsqr, 2015-07-03:
Ideally you want to implement DVBv5 statistics instead of the old DVBv3 in the driver as this problem is covered there...

ljalves, 2016-06-24:
I upstreamed the tbs6981 (cx24117) driver, the SNR fix (DVBv5 stats) has to go to the official media_tree, then merged to here...
I'll do it when I get the chance...

closed 2016-06-24 by ljalves