ljalves/linux_media #107: TBS6991SE: Can't attach to frontend

bw35, 2015-09-05:
Hi

I have TBS moi+ streaming box (ARM based embedded device containing a TBS6991SE card) I wish to update to the current linux_media kernel. With some tiny modifications I succeeded to build the kernel and the TBS opensource drivers from this repo.

While the saa716x_budget module can be loaded ok, it fails to attach to the frontend:

```
# modprobe saa716x_budget
DVB: registering new adapter (SAA716x dvb adapter)
i2c i2c-2: Added multiplexed i2c bus 3
i2c i2c-2: Added multiplexed i2c bus 4
i2c i2c-2: tas2101: i2c rdm err(-5) @0x00 (len=2)
i2c i2c-2: tas2101: Error attaching frontend
SAA716x Budget 0000:01:00.0: TurboSight TBS 6991SE frontend 0 attach failed
DVB: registering new adapter (SAA716x dvb adapter)
i2c i2c-1: Added multiplexed i2c bus 3
i2c i2c-1: Added multiplexed i2c bus 4
i2c i2c-1: tas2101: i2c rdm err(-5) @0x00 (len=2)
i2c i2c-1: tas2101: Error attaching frontend
SAA716x Budget 0000:01:00.0: TurboSight TBS 6991SE frontend 1 attach failed
```

The loaded modules are:

```
# lsmod
Module                  Size  Used by    Not tainted
saa716x_budget         22101  0 
tas2101                 7816  1 saa716x_budget
i2c_mux                 2162  1 tas2101
cxd2820r               13940  1 saa716x_budget
mb86a16                18926  1 saa716x_budget
saa716x_core           55272  1 saa716x_budget
dvb_core               65924  3 saa716x_budget,cxd2820r,saa716x_core
cx24117                10241  1 saa716x_budget
```

It seems that the I2C communication to the frontend isn't working. So I've tried scanning the I2C busses with i2cdetect, but no device is detected at any address: 

```
# i2cdetect -l
i2c-1   smbus           SAA716x I2C Core 0                      SMBus adapter
i2c-2   smbus           SAA716x I2C Core 1                      SMBus adapter
```

```
# i2cdetect -r 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

# i2cdetect -r 2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     
```

I've tried swapping the I2C busses in `saa716x_tbs6991se_frontend_attach()` and also disabled I2C mux usage by undefining `define TAS2101_USE_I2C_MUX` in tas2101.c

Any ideas how to fix the problem?
Has anybody got the TBS6991SE successfully working?

Peter

crazycat69, 2015-09-05:
First check with x86 linux

bw35, 2015-09-05:
Works ok with Kernel 3.16.7-24 on my PC. I can see the four muxed I2C buses, while they are missing on the ARM. I have to investigate in the I2C subsystem.

```
[  882.419902] DVB: registering new adapter (SAA716x dvb adapter)
[  882.426174] i2c i2c-9: Added multiplexed i2c bus 10
[  882.427136] i2c i2c-9: Added multiplexed i2c bus 11
[  882.528192] i2c i2c-11: av201x: Airoha Technology AV201x successfully attached
[  882.528200] SAA716x Budget 0000:28:00.0: DVB: registering adapter 0 frontend 0 (Tmax TAS2101)...
[  882.528425] DVB: registering new adapter (SAA716x dvb adapter)
[  882.529062] i2c i2c-8: Added multiplexed i2c bus 12
[  882.530105] i2c i2c-8: Added multiplexed i2c bus 13
[  882.631212] i2c i2c-13: av201x: Airoha Technology AV201x successfully attached
[  882.631220] SAA716x Budget 0000:28:00.0: DVB: registering adapter 1 frontend 0 (Tmax TAS2101)...

# i2cdetect -l
i2c-0   i2c             Radeon i2c bit bus 0x90                 I2C adapter
i2c-1   i2c             Radeon i2c bit bus 0x91                 I2C adapter
i2c-2   i2c             Radeon i2c bit bus 0x92                 I2C adapter
i2c-3   i2c             Radeon i2c bit bus 0x93                 I2C adapter
i2c-4   i2c             Radeon i2c bit bus 0x94                 I2C adapter
i2c-5   i2c             Radeon i2c bit bus 0x95                 I2C adapter
i2c-6   i2c             Radeon i2c bit bus 0x96                 I2C adapter
i2c-7   i2c             Radeon i2c bit bus 0x97                 I2C adapter
i2c-8   smbus           SAA716x I2C Core 0                      SMBus adapter
i2c-9   smbus           SAA716x I2C Core 1                      SMBus adapter
i2c-10  smbus           i2c-9-mux (chan_id 0)                   SMBus adapter
i2c-11  smbus           i2c-9-mux (chan_id 1)                   SMBus adapter
i2c-12  smbus           i2c-8-mux (chan_id 0)                   SMBus adapter
i2c-13  smbus           i2c-8-mux (chan_id 1)                   SMBus adapter
```

bw35, 2015-09-07:
I got it working. 

The PCIe bus wasn't running properly because of a bug introduced in kernel 4.x . The guilty commit is [a0b5cd4ac2d6](https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=a0b5cd4ac2d6) which has been fixed later in [8c9e06e647686](https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=8c9e06e64768665503e778088a39ecff3a6f2e0c). The fix is contained in the current stable 4.1.6 and mainline 4.2 kernel, but not in linux_media yet.

closed 2015-09-07 by bw35

enavarrocu, 2016-03-09:
@bw35 I have a TBS MOI, not the plus. I'm trying to do the same as you but I'm not able to compile the kernel. It's possible you can help me to do it?

Thanks in advance

bw35, 2016-03-13:
@enavarrocu  You can find my modifications for the MOI+ at [bw35/kernel-tbs-moiplus](https://github.com/bw35/kernel-tbs-moiplus). Besides some tiny fixes for the build to succeed I had to add the device tree for the box.

AFAIK the MOI is based on a i.MX6 which is supported in the main line kernel. You should probably get along by writing a suitable configuration and device tree.

enavarrocu, 2016-03-17:
@bw35 Thanks for your answer. The problem here is that I'm new in this so I'm having a lot of issue to make it compile. 

Did you modified the original u-boot? It's safe to upgrade it? I'm not sure about that but I want to do netboot because eventually the NAND is going to fail. Or also use the USB to boot but in the current version of the u-boot I think is not possible.

There is any way I can't contact you for some advice?

Thanks in advance

bw35, 2016-03-17:
@enavarrocu I have the MOI+ which is different from your MOI box. Mine has a Marvell Armada 300 SoC while yours has a NXP i.MX6.

On my box it's possible  to boot from USB or TFTP with the stock u-boot. It also features a fail safe mechanism to boot from the UART (kwboot) in case u-boot doesn't work anymore.

I don't know the i.MX6 in detail and I can't answer your questions. But the i.MX6 is well documented and supported as it is used on many commercial grade SOM or SBC. You should find the the necessary information on the web. 

Good Luck!