ljalves/linux_media #109: Machine freeze

markfoodyburton, 2015-10-04:
I am running Debian Squeeze with the 4.2.0-1 kernel on an Asrock N3150M, with a TechniSat SkyStar 2 eXpress HD.
I have followed the instructions on http://www.linuxtv.org/wiki/index.php/TBS6280 to build and install the modules, which are correctly loading.

02:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
        Subsystem: Device 1ae4:0700
        Flags: bus master, fast devsel, latency 0, IRQ 18
        Memory at 91300000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [40] MSI: Enable- Count=1/32 Maskable- 64bit+
        Capabilities: [50] Express Endpoint, MSI 00
        Capabilities: [74] Power Management version 2
        Capabilities: [80] Vendor Specific Information: Len=50 <?>
        Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>
        Kernel driver in use: SAA716x Budget

However, once the card is being used (e.g. tvheadend is scanning for channels), after a relatively short period of time (5-10 minutes), the machine 'freezes', and my only choice is the Big Red Button (it's small and greenish, but thats a super micro-case for you :-)). I see nothing in any logs.

Can anybody recommend a way to debug this issue?

Cheers
Mark.

markfoodyburton, 2015-10-05:
By getting the driver to take an exit on the first try,  rather than trying 100 times, then I can get some debug information out. Shortly after this is printed, the machine hangs, and it's BRB... 

 saa716x_i2c_send (0): FIFO full or Blocked
 saa716x_i2c_send (0): I2C Send failed (Err=-5)
 saa716x_i2c_write_msg (0): Address write failed
 saa716x_i2c_write_msg (0): Error writing data, err=-5
 saa716x_i2c_xfer (0): Error in Transfer, try 0
 saa716x_i2c_xfer (0): msg 0, addr = 0x68, len=2, flags=0x0
 saa716x_i2c_xfer (0):     <W 0000> 0xf4
 saa716x_i2c_xfer (0):     <W 0001> 0x12
 saa716x_i2c_xfer (0): msg 1, addr = 0x68, len=1, flags=0x1

ljalves, 2016-06-24:
Closing old issue.
Please re-open if still have this issue or updates about it.

closed 2016-06-24 by ljalves