ljalves/linux_media #110: Failed compiling the latest V4L-DVB

MrMerlino, 2015-11-05:
Hello,
i'm following this guide:
http://linuxtv.org/wiki/index.php/How_to_Obtain,_Build_and_Install_V4L-DVB_Device_Drivers

But i receive this error:

  CC [M]  /home/utente/media_build/v4l/v4l2-async.o
  CC [M]  /home/utente/media_build/v4l/v4l2-compat-ioctl32.o
  CC [M]  /home/utente/media_build/v4l/vb2-trace.o
/home/utente/media_build/v4l/vb2-trace.c:4:30: fatal error: trace/events/vb2.h: No such file or directory
compilation terminated.
scripts/Makefile.build:258: set di istruzioni per l'obiettivo "/home/utente/media_build/v4l/vb2-trace.o" non riuscito
make[3]: *** [/home/utente/media_build/v4l/vb2-trace.o] Errore 1
Makefile:1398: recipe for target '_module_/home/utente/media_build/v4l' failed
make[2]: *** [_module_/home/utente/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-4.2.0-16-generic'
Makefile:51: set di istruzioni per l'obiettivo "default" non riuscito
make[1]: *** [default] Errore 2
make[1]: uscita dalla directory "/home/utente/media_build/v4l"
Makefile:26: set di istruzioni per l'obiettivo "all" non riuscito
make: *** [all] Errore 2
build failed at ./build line 491.
utente@linuxtv:~/media_build$

Thanks!

ljalves, 2016-06-07:
Can you please retry?
I've merged the latest official media_tree and media_build.

ljalves, 2016-06-07:
Also:
Follow these instructions:
https://github.com/ljalves/linux_media/wiki/Installating

ljalves, 2016-06-24:
Re-open if needed

closed 2016-06-24 by ljalves