ljalves/linux_media #111: TBS5220

orrpan, 2015-11-08:
Any possibility to make the TBS5220 dvb-T2/T/C work?

nereeb, 2016-02-25:
The TBS 5220 is working with https://github.com/crazycat69/linux_media. I have tested DVB-C on a rpi2.

orrpan, 2016-03-01:
latest kernel?

ljalves, 2016-03-01:
I've merged CrazyCat PR #112 
Should be working now...

closed 2016-03-01 by ljalves

nereeb, 2016-03-01:
@orrpan  yes with 4.1.18-v7+

orrpan, 2016-03-02:
@nereeb Nice, trying to do it on rpi1 4.1.18+ make and make install works fine, but when I try to reboot to load it doesn't load the driver 

dvb-usb: found a 'TBS 5220 USB2.0' in cold state, will try to load a firmware
[  180.843801] usb 1-1.3: Direct firmware load for dvb-usb-tbsqbox-id5220.fw failed with error -2
[  180.843829] dvb-usb: did not find the firmware file. (dvb-usb-tbsqbox-id5220.fw) Please see linux/Documentation/dvb/ for more details on firmware-problems. (-2)

$ lsmod
Module                  Size  Used by
dvb_usb_tbs5220         7889  0 
dvb_usb                23647  1 dvb_usb_tbs5220
dvb_core               98263  1 dvb_usb
rc_core                24690  1 dvb_usb
media                  15964  2 dvb_usb,dvb_core
si2168                  7102  0 
i2c_mux                 2822  1 si2168
si2157                  4656  0 
cfg80211              479279  0 
rfkill                 21508  2 cfg80211
8192cu                556175  0 
snd_bcm2835            22502  0 
bcm2835_gpiomem         3703  0 
bcm2835_rng             2207  0 
snd_pcm                92829  1 snd_bcm2835
snd_timer              22164  1 snd_pcm
snd                    67534  3 snd_bcm2835,snd_timer,snd_pcm
uio_pdrv_genirq         3526  0 
uio                    10078  1 uio_pdrv_genirq
i2c_dev                 6386  0 
fuse                   86348  1 
ipv6                  358702  24

Any ideas?

trsqr, 2016-03-02:
Do you have the file dvb-usb-tbsqbox-id5220.fw in your /lib/firmware directory? It says that it failed to find the firmware file, so I believe you haven't placed it in the correct folder... (These firmwares cannot be distributed usually in standard Linux distributions due to licensing issues, so you'll need to add it manually).

orrpan, 2016-03-02:
@trsqr true, found it on https://github.com/beam/linux-tbs-drivers/blob/master/dvb-usb-tbsqbox-id5220.fw now it seems to be loaded. Thanks

orrpan, 2016-03-06:
problem, can't get it to work with dvbt2. I think i works with dvbc and dvbt but can't test it. I got it working in december with crazycats driver on dvbt2, but now when setting it to dvb-fe-tool -d DVBT2 it allways change when starting tvheadend :/

trsqr, 2016-03-06:
Tvheadend supports DVB-T2 natively - no need to touch dvb-fe-tool. You'll just need to make sure that when you define your muxes the delivery system is set to DVB-T2.

orrpan, 2016-03-06:
okey, strange thing is. I can't find any services/channels with tvheadend. But if I install the official drivers on windows, I find them :/

orrpan, 2016-03-24:
my problem was a to weak powersupply for the raspberry pi... Works lika a charm on rpi1 and rpi2 now