ljalves/linux_media #113: TBS6281 not working reliably

Jalle19, 2015-12-25:
I've tested this card with both DVB-T/T2 and DVB-C. In both situations, only one of the tuners actually work, and in the case of DVB-C a tuner can work once (tested with w_scan), then stop working immediately after. When it doesn't work, w_scan doesn't find anything and there is no output in dmesg. I've tried using both Linux 3.19 and 4.20 (Ubuntu repositories) and all variations of the int_type module parameter.

I have to revert to using the official driver for now. Any hints at what could be going on?

Jalle19, 2015-12-25:
I think it could be a generic issue with the whole DVB subsystem with these drivers because I had another tuner (dvb_usb_anysee module) which exhibited the same problems. I'll report back once I've tried the official drivers.

crazycat69, 2015-12-25:
options dvb_core dvb_powerdown_on_sleep=0

Jalle19, 2015-12-25:
@crazycat69 what the hell, it works! Still nothing on the second tuner (adapters 0 and 1 works, 2 and 3 don't) but I suspect that's a cabling issue since the tuner that is chained to that one doesn't work either. I'll report back if I get it to work.

Jalle19, 2015-12-26:
Yep, it was the cable.

closed 2015-12-26 by Jalle19

bas-t, 2016-06-20:
@Jalle19  Do you think 958e05a260e7f80e5cc25795fd7d9cf93c56b4a9 solves your issue without having to use
options dvb_core dvb_powerdown_on_sleep=0?

Mind trying?

Jalle19, 2016-06-23:
The only machines I have access to that use this card are mission critical so I can't test it I'm afraid. Perhaps someone else from this thread can give it a try?

bas-t, 2016-06-24:
That would be @crazycat69, I never experienced any reliability issues with my tbs6281.

I'm just curious if 958e05a has any positive effect on the matter.

crazycat69, 2016-06-24:
Maybe this si2168-b40 firmware issue. Look like 4.0.11 firmware patch work ok, but later builds have some sleep issue.

bas-t, 2016-06-24:
And I use the 4.0.11 version, so that might explain why I never experienced any reliability issues with my tbs6281.

Jalle19, 2016-06-24:
I think I'm on the newest firmware version too.