ljalves/linux_media #115: Problem compiling with kernel 4.3.3

zofiel, 2016-01-02:
I have the following error when perform the make command:

 CC [M]  /home/userssh/media_build/v4l/adp1653.o
/home/userssh/media_build/v4l/adp1653.c: In function 'adp1653_of_init':
/home/userssh/media_build/v4l/adp1653.c:468:20: error: too few arguments to function 'devm_gpiod_get'
  pd->enable_gpio = devm_gpiod_get(&client->dev, "enable");
                    ^
In file included from /home/userssh/media_build/v4l/adp1653.c:39:0:
include/linux/gpio/consumer.h:73:32: note: declared here
 struct gpio_desc *__must_check devm_gpiod_get(struct device *dev,
                                ^
scripts/Makefile.build:264: fallo en las instrucciones para el objetivo '/home/userssh/media_build/v4l/adp1653.o'
make[3]: *** [/home/userssh/media_build/v4l/adp1653.o] Error 1
Makefile:1378: recipe for target '_module_/home/userssh/media_build/v4l' failed
make[2]: *** [_module_/home/userssh/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/4.3.3-2-ARCH/build'
Makefile:51: fallo en las instrucciones para el objetivo 'default'
make[1]: *** [default] Error 2
make[1]: se sale del directorio '/home/userssh/media_build/v4l'
Makefile:26: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2

BBUK, 2016-01-04:
Hi there.

On looking at the kernel sources, it seems that a preprocessor directive that was inserted for compatibility reasons was removed in 4.3.  if you are comfortable in editing source files, there is something you could try as an experiment.  Navigate to line 468 in the file (from your system) /home/userssh/media_build/v4l/adp1653.c and change:

>pd->enable_gpio = devm_gpiod_get(&client->dev, "enable");

to

>pd->enable_gpio = devm_gpiod_get(&client->dev, "enable", GPIOD_ASIS);

If it does not work, let me know here (with any reported errors) and I'll try to build/fix myself (I will have to create a test system, so it will not be quick).

If it does work, please post back here so that everyone (including me) can benefit.

Bear in mind that I have *no* experience of the innards of this driver - I am just a *very* satisfied user shortly to commence my own 4.3 upgrade.  If this works, I'll submit a patch to ljalves (who deserves a load of credit here).

HTH

BBUK

pslijkhuis, 2016-01-05:
This works for me.

zofiel, 2016-02-07:
sorry for the delay.

I tried with last kernel and now the error is:

/home/userssh/media_build/v4l/cx23885-core.c: In function 'cx23885_initdev':
/home/userssh/media_build/v4l/cx23885-core.c:1993:7: error: implicit declaration of function 'pci_dma_supported' [-Werror=implicit-function-declaration]
  if (!pci_dma_supported(pci_dev, 0xffffffff)) {
       ^
cc1: some warnings being treated as errors
scripts/Makefile.build:258: fallo en las instrucciones para el objetivo '/home/userssh/media_build/v4l/cx23885-core.o'

So This issue is solved with the workaround

ljalves, 2016-03-01:
PR #119 merged. Should fix this issue.

closed 2016-03-01 by ljalves