ljalves/linux_media #116: reopen the tbs5990 ?

highkick05, 2016-01-19:
Hi I've installed a working openelec of the ljalves build , not working with tbs5990 ?

can I help get this working ?

ljalves, 2016-06-24:
Does that build include the tbs5990 driver?
Do you have a link to the repo used as source?

highkick05, 2016-06-24:
hi @ljalves 

Best to look @ [this link](https://github.com/crazycat69/linux_media/issues/1) on crazycats media_build, the furthest achieved in open source tbs5990 linux driver

tuner A & B working correctly on standard low resolution channels... but testing revealed does not work with MPEG-4/HD resolution yet. (all channels become green/pixelated even if appear normal initially)

TBS's closed linux drivers do however correctly decode MPEG-4/HD 
(UHD/4K unknown as I don't have a setup that supports it)

ljalves, 2016-06-24:
long thread!! no time to read!!
I think I don't even have that card added on my repo.

I could take a look if someone provides remote access to a machine with that card.

highkick05, 2016-06-25:
it would be worth the time, I could provide this machine.

just give me some time to set it up and I will message you details to ur email

any preference on setup ? ubuntu / version / kernel / etc

ljalves, 2016-06-26:
ubuntu is fine

highkick05, 2016-06-27:
ok I'll set this up 
will go with latest 16.04 w/ 4.4 kernel 

watch for email