ljalves/linux_media #118: Kernel 4.4

BBUK1, 2016-02-03:
This does not build as-is on kernel 4.4.  A few files need to be changed due to the removal of the function pci_dma_supported(struct pci_dev *hwdev, u64 mask).   A similar function dma_supported is, however, available (for the moment).

Firstly, a change that is required to support kernel 4.3 (as previously reported).  From the media_build directory, edit ..media/drivers/media/i2c/adp1653.c and change 

`pd->enable_gpio = devm_gpiod_get(&client->dev, "enable");`

to

`pd->enable_gpio = devm_gpiod_get(&client->dev, "enable", GPIOD_ASIS);`

The next set of changes are all similar, the files I had to change were (again from the media_build directory)

../media/drivers/media/pci/cx23885/cx23885-core.c
../media/drivers/media/pci/cx25821/cx25821-core.c
../media/drivers/media/pci/cx88/cx88-video.c
../media/drivers/media/pci/cx88/cx88-mpeg.c
../media/drivers/media/pci/cx88/cx88-alsa.c
../media/drivers/media/pci/saa7134/saa7134-core.c
../media/drivers/media/pci/saa7164/saa7164-core.c

In each of these files, search for a line of the form

`if (!pci_dma_supported({X}, {Y})) {`

where {X} and {Y} may differ and replace with

`if (!dma_supported({X} == NULL ? NULL : &{X}->dev, {Y})) {`

substituting the {X} and {Y} appropriately so for example:

`if (!pci_dma_supported(pci_dev, 0xffffffff)) {`

means {X} is "pci_dev" and {Y} is "0xffffffff" and the resulting changed line becomes

`if (!dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, 0xffffffff)) {`

and (another example)

`if (!pci_dma_supported(dev->pci,DMA_BIT_MASK(32))) {`

means {X} is "dev->pci"  and {Y} is "DMA_BIT_MASK(32)" and the resulting line becomes

`if (!dma_supported(dev->pci == NULL ? NULL : &dev->pci->dev, DMA_BIT_MASK(32))) {`

After all that follow the usual make installation steps, so

make dir DIR=../media
make distclean
make
sudo make install

(you may also need to remove duplicated modules manually as described in the wiki)

This is untested on any hardware other than my own (hence I should not really make a fork of this repo with these changes).  Please report back if these changes worked for you.

BBUK

JohnnySSH, 2016-02-06:
I tried the above on Arch Linux 64bit kernel with 2x TBS cards; one DVB-T/T2 and one DVB-S/S2 unfortunately the result was as follows:

/home/htpc/Downloads/tbs-opensource/media_build/v4l/adp1653.c: In function 'adp1653_of_init':
/home/htpc/Downloads/tbs-opensource/media_build/v4l/adp1653.c:468:20: error: too few arguments to function 'devm_gpiod_get'
  pd->enable_gpio = devm_gpiod_get(&client->dev, "enable");
                    ^
In file included from /home/htpc/Downloads/tbs-opensource/media_build/v4l/adp1653.c:39:0:
include/linux/gpio/consumer.h:73:32: note: declared here
 struct gpio_desc *__must_check devm_gpiod_get(struct device *dev,
                                ^
scripts/Makefile.build:264: recipe for target '/home/htpc/Downloads/tbs-opensource/media_build/v4l/adp1653.o' failed
make[3]: *** [/home/htpc/Downloads/tbs-opensource/media_build/v4l/adp1653.o] Error 1
Makefile:1384: recipe for target '_module_/home/htpc/Downloads/tbs-opensource/media_build/v4l' failed
make[2]: *** [_module_/home/htpc/Downloads/tbs-opensource/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/4.4.1-2-ARCH/build'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2

JohnnySSH, 2016-02-06:
Ok... so the edits are made to be made in the media/ directory, not the media_build/ directory! :-)

Unfortunately still doesn't build:

`/home/htpc/Downloads/tbs-opensource/media_build/v4l/cx23885-core.c: In function 'cx23885_initdev':
/home/htpc/Downloads/tbs-opensource/media_build/v4l/cx23885-core.c:1993:7: error: implicit declaration of function 'pci_dma_supported' [-Werror=implicit-function-declaration]
  if (!pci_dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, 0xffffffff)) {
       ^
cc1: some warnings being treated as errors
scripts/Makefile.build:258: recipe for target '/home/htpc/Downloads/tbs-opensource/media_build/v4l/cx23885-core.o' failed
make[3]: *** [/home/htpc/Downloads/tbs-opensource/media_build/v4l/cx23885-core.o] Error 1
Makefile:1384: recipe for target '_module_/home/htpc/Downloads/tbs-opensource/media_build/v4l' failed
make[2]: *** [_module_/home/htpc/Downloads/tbs-opensource/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/4.4.1-2-ARCH/build'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l'
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2`

JohnnySSH, 2016-02-06:
After fixing:

../media/drivers/media/pci/tw68/tw68-core.o

the drivers build and install fine!!

zofiel, 2016-02-07:
Could you explain the actions to be performed please?

I have the same problem

JohnnySSH, 2016-02-07:
@zofiel if you go into the root directory of the OpenSource drivers:

`$PATH/tbs-opensource`

There should be two folders in there... :

`media  media_build  update.sh`

So with a text editor like vi, emacs, nano etc... 

Follow the instructions above.

First open the file: media/drivers/media/i2c/adp1653.c

Then do a search (Find in graphical editors) for this line:

pd->enable_gpio = devm_gpiod_get(&client->dev, "enable");

and modify it to:

pd->enable_gpio = devm_gpiod_get(&client->dev, "enable", GPIOD_ASIS);

Save and Close :-)

Then for all these files:

../media/drivers/media/pci/cx23885/cx23885-core.c
../media/drivers/media/pci/cx25821/cx25821-core.c
../media/drivers/media/pci/cx88/cx88-video.c
../media/drivers/media/pci/cx88/cx88-mpeg.c
../media/drivers/media/pci/cx88/cx88-alsa.c
../media/drivers/media/pci/saa7134/saa7134-core.c
../media/drivers/media/pci/saa7164/saa7164-core.c

+

../media/drivers/media/pci/tw68/tw68-core.o

Search for this line:

if (!pci_dma_supported

....there should only be one instance per file!!

Then simply modify as stated:

> In each of these files, search for a line of the form
>
> if (!pci_dma_supported({X}, {Y})) {
>
> where {X} and {Y} may differ and replace with
>
> if (!dma_supported({X} == NULL ? NULL : &{X}->dev, {Y})) { [...]

So this line as an example:

if (!pci_dma_supported(pci_dev, 0xffffffff)) {

becomes this:

if (!dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, 0xffffffff)) {

comparing the two lines; remove "pci" from the front of the line and add "== NULL ? NULL : &" after the variable behind the bracket, then add the same variable after the "&"

After all relevant files have been modified go into the "media_build" directory and run:

make distclean
make
sudo make install

and provided no errors come up, it should be working.

lomion0815, 2016-02-12:
A patch file for it:
```
diff --git a/drivers/media/i2c/adp1653.c b/drivers/media/i2c/adp1653.c
index c70abab..f101279 100644
--- a/drivers/media/i2c/adp1653.c
+++ b/drivers/media/i2c/adp1653.c
@@ -465,7 +465,7 @@ static int adp1653_of_init(struct i2c_client *client,

        of_node_put(child);

-       pd->enable_gpio = devm_gpiod_get(&client->dev, "enable");
+       pd->enable_gpio = devm_gpiod_get(&client->dev, "enable", GPIOD_ASIS);
        if (!pd->enable_gpio) {
                dev_err(&client->dev, "Error getting GPIO\n");
                return -EINVAL;
diff --git a/drivers/media/pci/cx23885/cx23885-core.c b/drivers/media/pci/cx23885/cx23885-core.c
index 7aee76a..48fcb2d 100644
--- a/drivers/media/pci/cx23885/cx23885-core.c
+++ b/drivers/media/pci/cx23885/cx23885-core.c
@@ -1990,7 +1990,7 @@ static int cx23885_initdev(struct pci_dev *pci_dev,
                (unsigned long long)pci_resource_start(pci_dev, 0));

        pci_set_master(pci_dev);
-       if (!pci_dma_supported(pci_dev, 0xffffffff)) {
+        if (!dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, 0xffffffff)) {
                printk("%s/0: Oops: no 32bit PCI DMA ???\n", dev->name);
                err = -EIO;
                goto fail_context;
diff --git a/drivers/media/pci/cx25821/cx25821-core.c b/drivers/media/pci/cx25821/cx25821-core.c
index 559f829..64085ea 100644
--- a/drivers/media/pci/cx25821/cx25821-core.c
+++ b/drivers/media/pci/cx25821/cx25821-core.c
@@ -1319,7 +1319,7 @@ static int cx25821_initdev(struct pci_dev *pci_dev,
                dev->pci_lat, (unsigned long long)dev->base_io_addr);

        pci_set_master(pci_dev);
-       if (!pci_dma_supported(pci_dev, 0xffffffff)) {
+        if (!dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, 0xffffffff)) {
                pr_err("%s/0: Oops: no 32bit PCI DMA ???\n", dev->name);
                err = -EIO;
                goto fail_irq;
diff --git a/drivers/media/pci/cx88/cx88-alsa.c b/drivers/media/pci/cx88/cx88-alsa.c
index 7f8dc60..7ad1f6a 100644
--- a/drivers/media/pci/cx88/cx88-alsa.c
+++ b/drivers/media/pci/cx88/cx88-alsa.c
@@ -890,7 +890,7 @@ static int snd_cx88_create(struct snd_card *card, struct pci_dev *pci,
                return err;
        }

-       if (!pci_dma_supported(pci,DMA_BIT_MASK(32))) {
+        if (!dma_supported(pci == NULL ? NULL : &pci->dev, DMA_BIT_MASK(32))) {
                dprintk(0, "%s/1: Oops: no 32bit PCI DMA ???\n",core->name);
                err = -EIO;
                cx88_core_put(core, pci);
diff --git a/drivers/media/pci/cx88/cx88-mpeg.c b/drivers/media/pci/cx88/cx88-mpeg.c
index 34f5057..c767e74 100644
--- a/drivers/media/pci/cx88/cx88-mpeg.c
+++ b/drivers/media/pci/cx88/cx88-mpeg.c
@@ -393,7 +393,7 @@ static int cx8802_init_common(struct cx8802_dev *dev)
        if (pci_enable_device(dev->pci))
                return -EIO;
        pci_set_master(dev->pci);
-       if (!pci_dma_supported(dev->pci,DMA_BIT_MASK(32))) {
+        if (!dma_supported(dev->pci == NULL ? NULL : &dev->pci->dev, DMA_BIT_MASK(32))) {
                printk("%s/2: Oops: no 32bit PCI DMA ???\n",dev->core->name);
                return -EIO;
        }
diff --git a/drivers/media/pci/cx88/cx88-video.c b/drivers/media/pci/cx88/cx88-video.c
index cebb07d..7244cab 100644
--- a/drivers/media/pci/cx88/cx88-video.c
+++ b/drivers/media/pci/cx88/cx88-video.c
@@ -1311,7 +1311,7 @@ static int cx8800_initdev(struct pci_dev *pci_dev,
               dev->pci_lat,(unsigned long long)pci_resource_start(pci_dev,0));

        pci_set_master(pci_dev);
-       if (!pci_dma_supported(pci_dev,DMA_BIT_MASK(32))) {
+        if (!dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, DMA_BIT_MASK(32))) {
                printk("%s/0: Oops: no 32bit PCI DMA ???\n",core->name);
                err = -EIO;
                goto fail_core;
diff --git a/drivers/media/pci/saa7134/saa7134-core.c b/drivers/media/pci/saa7134/saa7134-core.c
index 72d7f99..01f0397 100644
--- a/drivers/media/pci/saa7134/saa7134-core.c
+++ b/drivers/media/pci/saa7134/saa7134-core.c
@@ -949,7 +949,7 @@ static int saa7134_initdev(struct pci_dev *pci_dev,
               pci_name(pci_dev), dev->pci_rev, pci_dev->irq,
               dev->pci_lat,(unsigned long long)pci_resource_start(pci_dev,0));
        pci_set_master(pci_dev);
-       if (!pci_dma_supported(pci_dev, DMA_BIT_MASK(32))) {
+        if (!dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, DMA_BIT_MASK(32))) {
                pr_warn("%s: Oops: no 32bit PCI DMA ???\n", dev->name);
                err = -EIO;
                goto fail1;
diff --git a/drivers/media/pci/saa7164/saa7164-core.c b/drivers/media/pci/saa7164/saa7164-core.c
index 3206a82..3e2a67e 100644
--- a/drivers/media/pci/saa7164/saa7164-core.c
+++ b/drivers/media/pci/saa7164/saa7164-core.c
@@ -1264,7 +1264,7 @@ static int saa7164_initdev(struct pci_dev *pci_dev,

        pci_set_master(pci_dev);
        /* TODO */
-       if (!pci_dma_supported(pci_dev, 0xffffffff)) {
+        if (!dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, 0xffffffff)) {
                printk("%s/0: Oops: no 32bit PCI DMA ???\n", dev->name);
                err = -EIO;
                goto fail_irq;
diff --git a/drivers/media/pci/tw68/tw68-core.c b/drivers/media/pci/tw68/tw68-core.c
index c135165..642793d 100644
--- a/drivers/media/pci/tw68/tw68-core.c
+++ b/drivers/media/pci/tw68/tw68-core.c
@@ -256,7 +256,7 @@ static int tw68_initdev(struct pci_dev *pci_dev,
                dev->name, pci_name(pci_dev), dev->pci_rev, pci_dev->irq,
                dev->pci_lat, (u64)pci_resource_start(pci_dev, 0));
        pci_set_master(pci_dev);
-       if (!pci_dma_supported(pci_dev, DMA_BIT_MASK(32))) {
+        if (!dma_supported(pci_dev == NULL ? NULL : &pci_dev->dev, DMA_BIT_MASK(32))) {
                pr_info("%s: Oops: no 32bit PCI DMA ???\n", dev->name);
                err = -EIO;
                goto fail1;
```

tkurbad, 2016-03-01:
@lomion0815 the patch doesn't work anymore if posted this way, since all tabs are converted to spaces. You should instead fork this repository, create a branch of your own fork, do the changes there and then do a pull request. This way, the original maintainer could merge your changes easily. Also, all other people could download your patch directly from your github commit...

ljalves, 2016-03-01:
PR #119 merged.
Thanks for the contribution.

closed 2016-03-01 by ljalves