ljalves/linux_media #121: unable to build driver - kodibuntu

acoloss, 2016-03-07:
Hi,

i'm receiving these errors with the latest source when building on kodibuntu

Linux koditvh 3.13.0-79-generic 

```
  CC [M]  /home/acoloss/media_build/v4l/adp1653.o
/home/acoloss/media_build/v4l/adp1653.c: In function 'adp1653_of_init':
/home/acoloss/media_build/v4l/adp1653.c:468:59: error: 'GPIOD_ASIS' undeclared (first use in this function)
  pd->enable_gpio = devm_gpiod_get(&client->dev, "enable", GPIOD_ASIS);
                                                           ^
/home/acoloss/media_build/v4l/adp1653.c:468:59: note: each undeclared identifier is reported only once for each function it appears in
/home/acoloss/media_build/v4l/adp1653.c:468:2: error: too many arguments to function 'devm_gpiod_get'
  pd->enable_gpio = devm_gpiod_get(&client->dev, "enable", GPIOD_ASIS);
  ^
In file included from /home/acoloss/media_build/v4l/adp1653.c:39:0:
include/linux/gpio/consumer.h:29:32: note: declared here
 struct gpio_desc *__must_check devm_gpiod_get(struct device *dev,
                                ^
make[3]: *** [/home/acoloss/media_build/v4l/adp1653.o] Error 1
make[2]: *** [_module_/home/acoloss/media_build/v4l] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-3.13.0-79-generic'
make[1]: *** [default] Error 2
make[1]: Leaving directory `/home/acoloss/media_build/v4l'
make: *** [all] Error 2
```

acoloss, 2016-03-07:
i managed to fix it myself my reverting the last changes (6 days ago), but i guess it will happen with others as well?

stephen47, 2016-03-22:
I can confirm this issue as well. Reverting https://github.com/ljalves/linux_media/commit/5f7a3131c3090270c348166ff7575b8e6ce58c0c fixed the issue.

xraynorm, 2016-03-26:
I am seeing this too. How do I revert  5f7a313?

jpg73, 2016-04-03:
I'm concerning too ... How do I revert 5f7a313?

Thanks in advance
JP

ljalves, 2016-06-24:
This should be fixed with the latest media_build / media_tree

Re-open if not.

closed 2016-06-24 by ljalves