ljalves/linux_media #125: Kernel 4.5

BBUK, 2016-04-25:
Hi all.

This does not build on kernel 4.5.  I made the following edit to line 734 of ../media/drivers/media/dvb-frontends/cxd2820r_core.c and changed
` priv->gpio_chip.dev = &priv->i2c->dev;`
to
` priv->gpio_chip.parent = &priv->i2c->dev;`
(note that you are editing a file in the "media" directory not the "media_build" directory)
This change will break compatibility with kernels earlier than 4.5 so don't apply it unless you have a specific issue with 4.5.  Note that I don't really know what I am doing here.  The change works on my setup but please report back if this also works for you.

BBUK

JohnnySSH, 2016-05-03:
Hi looks like this works for me fine..... yet to test, but does build and install on Arch Linux updated as of today ;-)

ljalves, 2016-06-07:
Updated to latest media_tree.
Should compile fine at least up to kernel 4.7

closed 2016-06-24 by ljalves