ljalves/linux_media #126: Error loading driver

albertdelcapio, 2016-05-04:
Hi People, I could install driver successfully on RPI2 with ubuntu 14.04, and another RPI2 with libreELEC (openelec with[ embedded drivers](https://forum.libreelec.tv/thread-130.html)) is with the same issue, driver gets installed OK and when you reboot, at booting there is an error

here is the log:

> dmesg | grep -i dvb
[    7.947885] dvb-usb: found a 'TBS 5881 CI USB2.0' in cold state, will try to                                                                                         load a firmware
[    8.046236] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5881.f                                                                                        w'
[    8.164696] dvb-usb: found a 'TBS 5881 CI USB2.0' in warm state.
[    8.165251] dvb-usb: will pass the complete MPEG2 transport stream to the sof                                                                                        tware demuxer.
[    8.165469] DVB: registering new adapter (TBS 5881 CI USB2.0)
[    8.278670] dvb-usb: MAC address: xxxxxxxxxxxxxxx
[    8.280338] dvb_create_media_entity: media entity 'dvb-demux' registered.
[    8.391562] dvb_create_media_entity: media entity 'dvb-ca-en50221' registered                                                                                        .
[    8.392546] usb 1-1.3: DVB: registering adapter 0 frontend 0 (Silicon Labs Si                                                                                        2168)...
[    8.392588] dvb_create_media_entity: media entity 'Silicon Labs Si2168' regis                                                                                        tered.
[    8.405014] dvb-usb: TBS 5881 CI USB2.0 error while loading driver (-12)
[   13.391394] Modules linked in: 8021q si2157(O) si2168(O) i2c_mux dvb_usb_tbs5                                                                                        881(O) dvb_usb(O) rc_core(O) dvb_core(O) media(O) bcm2835_gpiomem bcm2835_wdt
[   13.391489] CPU: 2 PID: 297 Comm: kdvb-ca-0:0 Tainted: G           O    4.4.7                                                                                         #1
[   13.391609] LR is at dvb_ca_en50221_check_camstatus.part.0+0x28/0xbc [dvb_cor                                                                                        e]
[   13.391773] Process kdvb-ca-0:0 (pid: 297, stack limit = 0xa9cfe210)
[   13.392249] [<7f018484>] (dvb_ca_en50221_check_camstatus.part.0 [dvb_core]) f                                                                                        rom [<7f018744>] (dvb_ca_en50221_thread+0xfc/0x700 [dvb_core])
[   13.392373] [<7f018648>] (dvb_ca_en50221_thread [dvb_core]) from [<8003827c>]                                                                                         (kthread+0xe4/0xfc)
[   21.205250] Modules linked in: 8021q si2157(O) si2168(O) i2c_mux dvb_usb_tbs5                                                                                        881(O) dvb_usb(O) rc_core(O) dvb_core(O) media(O) bcm2835_gpiomem bcm2835_wdt
[   21.205442] PC is at dvb_frontend_open+0x34/0x32c [dvb_core]
[   21.205478] LR is at dvb_device_open+0xa0/0xdc [dvb_core]
[   21.206601] [<7f01a9a8>] (dvb_frontend_open [dvb_core]) from [<7f011b08>] (dv                                                                                        b_device_open+0xa0/0xdc [dvb_core])
[   21.206730] [<7f011a68>] (dvb_device_open [dvb_core]) from [<800ea80c>] (chrd                                                                                        ev_open+0xd4/0x194)

problem is with TBS5881 as seen, any ideas?

ljalves, 2016-06-24:
Not sure I can help you here... could do some debug with remote access to a machine with the card.

albertdelcapio, 2016-06-27:
no problem! just contact me and we take a look on it

Date: Fri, 24 Jun 2016 09:49:25 -0700

closed 2016-07-05 by albertdelcapio