ljalves/linux_media #129: TBS6905 - any success?

gedl, 2016-06-02:
Hi,

TBS6905 is not listed in the supported cards table, but as this list hasn't been updated for a year, I've decided to ask:

Has anybody had any success with TBS6905 with this or any other driver?

Thanks

crazycat69, 2016-06-02:
See my fork of this repo. Recently added tbspcie driver from Luis with 6205, 6904, 6909 support. From me added fixed stv6120 driver from Cris Lee and stv0910 from DD. And finally added support for 6183 in saa716x_budget and 6903/5/8 in tbspcie. But i test only 6983 and 6908. But 6905 is same 6908 with broadcast version stv0910.10:30, 2 июня 2016 г., "Gonçalo Luiz" <notifications@github.com>: TBS6905 is not listed in the supported cards table, but as this list hasn't been updated for a year, I've decided to ask: Has anybody had any success with TBS6905 with this or any other driver?

ljalves, 2016-06-02:
Hi guys,

I've been working on the new tbs cards "offline" but I plan to update the github repo really soon.

It will add open source drivers for the following cards as a start:
DVB-C/T
TBS6205 (quad)

DVB-S
TBS6903 (dual)
TBS6904 (quad)
TBS6905 (quad)
TBS6909 (octo)
TBS6910 (dual+CI)

DVB-T/C/S
TBS5520 (single - usb)

A lot of other new cards will be add later.

Best regards,
Luis

gedl, 2016-06-02:
Obrigado.

Can't wait.

Gonçalo Luiz

ovidiu31, 2016-06-02:
excellent job @ljalves 

much appreciating the whole effort

bpbastos, 2016-06-02:
Awesome @ljalves!

crazycat69, 2016-06-02:
> I've been working on the new tbs cards "offline" but I plan to update the github repo really soon.

Welcome back to github :-)

stailus, 2016-06-03:
Hi ljalves
I try new TBS open source drivers on Ubuntu 16.04 with kernel 4.4.0 with a TBS6205 card on DVB-T2 and DVB-C. Work but sometimes don't get lock when start app and I must restart my dvbapp (try with minisatip and dvblast and both have the same behaivour especially with DVB-C signal maybe signal have noise I don't know). With dvblast I try to get signal information, but no success (last dvblast from git)

type: QAM
name: Silicon Labs Si2168
frequency_min: 42000000
frequency_max: 870000000
frequency_stepsize: 0
frequency_tolerance: 0
symbol_rate_min: 1000000
symbol_rate_max: 7200000
symbol_rate_tolerance: 0
notifier_delay: 0

capability list:
CAN_INVERSION_AUTO
CAN_FEC_1_2
CAN_FEC_2_3
CAN_FEC_3_4
CAN_FEC_5_6
CAN_FEC_7_8
CAN_FEC_AUTO
CAN_QPSK
CAN_QAM_16
CAN_QAM_32
CAN_QAM_64
CAN_QAM_128
CAN_QAM_256
CAN_QAM_AUTO
CAN_TRANSMISSION_MODE_AUTO
CAN_GUARD_INTERVAL_AUTO
CAN_HIERARCHY_AUTO
CAN_MUTE_TS
CAN_2G_MODULATION
CAN_MULTISTREAM

status:
HAS_SIGNAL
HAS_CARRIER
HAS_VITERBI
HAS_SYNC
HAS_LOCK

Bit error rate: 0
Signal strength: 0
SNR: 0

I try with femon but the same. Can you fix this issue?

ljalves, 2016-06-03:
That issue has been fixed in the driver already.
The driver CrazyCat submitted to his repo is unfinished and from a very early stage.

Hopefully I'll be able to update my git with all the new stuff during this weekend!

Alfredo-, 2016-06-08:
Thank you for continuing working on drivers.
I enjoy my TBS6922 because of you

ljalves, 2016-06-24:
TBS6905 supported now.
Closing!

closed 2016-06-24 by ljalves