ljalves/linux_media #131: Driver no longer works

robmcc1983, 2016-06-13:
Hi,
I have used these drivers for over a year now and since the update they no longer work for me.
i have reinstalled several times without any issue except this time it doesnt install the driver for my tbs6982.

i also have a hauppage hvr 3000 which i why i used open source drivers so that both cards work together  but what happens now is that the driver just does not install tbs6982 driver at all.
my hvr-3000 works in kernel so no problems there.

Rob

ljalves, 2016-06-13:
Can you be more specific about "doesnt install the driver for my tbs6982"?

robmcc1983, 2016-06-13:
I'm not a pro with Linux I'm still leaning lol, but I'm the past I've followed the instructions and both my cards have shown up in Tvheadend. I've just replaced my hard drive with a ssd installed the exact way I normally do but I cannot get the tbs card to appear in Tvheadend.
I use lubuntu 14.04 which is update to 4.2.x kernel. At first I though maybe it could be the kernel but I put my old hard drive in and that is using the same kernel but it's using you previous release of open source drivers (I never I updated as I only noticed today you had a recent update). Since posting earlier I have installed lubuntu 16.04 and used your open source drivers but that installs my tbs card ok but then my hauuppage hvr-3000 no longer shows in tv headend. Like I say I'm not a expert, if things go wrong I take the cowards way out and just reinstall everything lol

ljalves, 2016-06-13:
Can you please paste the output of your "dmesg" or /var/log/kern.log (since last boot).
Please use http://pastebin.com/

robmcc1983, 2016-06-14:
hi,
so i have done a fresh install of lubuntu 14.04 i will post kern.log and dmesg before and after your open source driver installation.

here is the kern.log text before driver installation http://pastebin.com/uFwvpRhT

Here is the dmesg before driver installation http://pastebin.com/VKnUbCX3

thanks

robmcc1983, 2016-06-14:
here is kern.log after opensource driver installation http://pastebin.com/xPAq4NWC

and here is dmesg after driver installation http://pastebin.com/pStXADeG

i must add too that when i ran make i had some sort of error2 regarding usb speed super plus.

then when i ran sudo make install it looks like its install tt_usb and not sa7160_budget like it use to.

Cheers

ljalves, 2016-06-14:
Couple of things:
-The kern.log paste is incomplete  (not sure about dmesg)
- I can't see any reference to the saa716x module, so try the following:
> sudo modprobe saa716x_budget

robmcc1983, 2016-06-14:
i get the same error whilst running make on debian http://pastebin.com/ZNNg7R1c

robmcc1983, 2016-06-14:
Also sudo modprobe saa716x_budget gives me error "fatal: module saa716x_budget not found.

ljalves, 2016-06-14:
Something is wrong...
Your build process is trying to use headers from kernel 3.16.0-4 while your running kernel is 4.2.0

Did you followed the install steps correctly?

In your case I'm sure you missed the "make distclean" step.

robmcc1983, 2016-06-14:
That's because the last pastebin was from Debian instead of Ubuntu 
Debian uses 3.16 where as lubuntu uses 4.2.
So far after having this issue I have tried lubuntu 14.04, lubuntu 16.04, Ubuntu server 14.04 and Debian.
All of these give me the exact same error when trying to make from your open source driver. So something has had to of changed in the update and it no longer works for me.
I have put my old hard drive back in now which is running lubuntu 14.04 with the 4.2 kernel. That works fine as the drivers were made before your latest update.
Thanks anyway

ljalves, 2016-06-14:
kernel 3.16 is too old but 4.2 works without any problem.

robmcc1983, 2016-06-14:
no kernels work for me, a clean install of lubuntu 14.04 with kernel 4.2 is when i first noticed the problem. driver will not install.
lubuntu 16.04 with kernel 4.4 driver will not install.

I have tried many different fresh installs of different various linux distros and every one give me the same error since you open source drivers have been updated.

like i said in my earlier post i have reinstalled a couple of times over the last 12 months with reinstalling your old version without no problems whatsoever, so for me your drivers are broken.

ljalves, 2016-06-14:
If you want, give me remote access to your machine and I'll try to debug what's wrong.
Email me if you're interested.

ljalves, 2016-06-15:
The "usb speed super plus" error should be fixed now with the latest update to media_build.

ljalves, 2016-06-24:
If you still have issues, retry with the latest media_build / media_tree

Re-open issue if needed

closed 2016-06-24 by ljalves