ljalves/linux_media #132: Kernel 4.6 ln missing file operand

JohnnySSH, 2016-06-17:
Hi, I'm trying to update my drivers to work with kernel 4.6 but I am running into an issue with a complaint about the "ln" command.

Update process is per documentation:

 cd media
 git remote update
 git pull
 cd ../media_build
 git remote update
 git pull
 make distclean
 make

Getting response:

Applying patches for kernel 4.6.2-1-ARCH
patch -s -f -N -p1 -i ../backports/api_version.patch
2 out of 2 hunks FAILED
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
make[2]: *** [Makefile:138: apply_patches] Error 1
make[2]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/linux'
Preparing to compile for kernel version 4.6.2

***WARNING:*** You do not have the full kernel sources installed.
This does not prevent you from building the v4l-dvb tree if you have the
kernel headers, but the full kernel source may be required in order to use
make menuconfig / xconfig / qconfig.

If you are experiencing problems building the v4l-dvb tree, please try
building against a vanilla kernel before reporting a bug.

Vanilla kernels are available at http://kernel.org.
On most distros, this will compile a newly downloaded kernel:

cp /boot/config-`uname -r` <your kernel dir>/.config
cd <your kernel dir>
make all modules_install install

Please see your distro's web site for instructions to build a new kernel.

WARNING: This is the V4L/DVB backport tree, with experimental drivers
	 backported to run on legacy kernels from the development tree at:
		http://git.linuxtv.org/media-tree.git.
	 It is generally safe to use it for testing a new driver or
	 feature, but its usage on production environments is risky.
	 Don't use it in production. You've been warned.
Created default (all yes) .config file
./scripts/make_myconfig.pl
perl scripts/make_config_compat.pl /lib/modules/4.6.2-1-ARCH/build ./.myconfig ./config-compat.h
creating symbolic links...
ln: missing file operand
Try 'ln --help' for more information.
make[1]: *** [Makefile:261: links] Error 123
make[1]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l'
make: *** [Makefile:26: all] Error 2

If it's something to do with Kernel headers I'll check that I have them installed as I have been building fine for a while now with previous kernels, so maybe something hasn't properly got updated yet??

ljalves, 2016-06-24:
Try with a full clean - before the "make distclean", also do:

make clean
make dir DIR=../media
make distclean
make

JohnnySSH, 2016-06-24:
So I ran this:

 cd media
 git remote update
 git pull
 cd ../media_build
 git remote update
 git pull
 make clean
 make dir DIR=../media
 make distclean
 make

and I still get same issue:

rm -f *~ *.o *.ko .*.o.cmd .*.ko.cmd *.mod.c av7110_firm.h fdump \
	config-compat.h Module.symvers Module.markers modules.order \
	*.unsigned .*.ko.unsigned.cmd
make -C firmware clean
make[2]: Entering directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l/firmware'
rm -f ihex2fw
rm -f vicam/firmware.fw ttusb-budget/dspbootcode.bin cpia2/stv0672_vp4.bin av7110/bootcode.bin
make[2]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l/firmware'
rm -f .version .*.o.flags .*.o.d *.mod.gcno Makefile.media \
	Kconfig Kconfig.kern .config .config.cmd .myconfig \
	.kconfig.dep
rm -rf .tmp_versions .tmp*.ver .tmp*.o .*.gcno
rm -f scripts/lxdialog scripts/kconfig
make -C firmware distclean
make[2]: Entering directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l/firmware'
rm -f ihex2fw
rm -f vicam/firmware.fw ttusb-budget/dspbootcode.bin cpia2/stv0672_vp4.bin av7110/bootcode.bin
for i in av7110/ cpia2/ ttusb-budget/ vicam/; do if [ -d $i ]; then rm -rf $i; fi; done
make[2]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l/firmware'
make[1]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l'
make -C /home/htpc/Downloads/tbs-opensource/media_build/v4l 
make[1]: Entering directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l'
No version yet, using 4.6.2-1-ARCH
scripts/make_makefile.pl
Updating/Creating .config
make[2]: Entering directory '/home/htpc/Downloads/tbs-opensource/media_build/linux'
Syncing with dir ../../media
Applying patches for kernel 4.6.2-1-ARCH
patch -s -f -N -p1 -i ../backports/api_version.patch
2 out of 2 hunks FAILED
1 out of 1 hunk FAILED
1 out of 1 hunk FAILED
make[2]: *** [Makefile:138: apply_patches] Error 1
make[2]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/linux'
Preparing to compile for kernel version 4.6.2

***WARNING:*** You do not have the full kernel sources installed.
This does not prevent you from building the v4l-dvb tree if you have the
kernel headers, but the full kernel source may be required in order to use
make menuconfig / xconfig / qconfig.

If you are experiencing problems building the v4l-dvb tree, please try
building against a vanilla kernel before reporting a bug.

Vanilla kernels are available at http://kernel.org.
On most distros, this will compile a newly downloaded kernel:

cp /boot/config-`uname -r` <your kernel dir>/.config
cd <your kernel dir>
make all modules_install install

Please see your distro's web site for instructions to build a new kernel.

WARNING: This is the V4L/DVB backport tree, with experimental drivers
	 backported to run on legacy kernels from the development tree at:
		http://git.linuxtv.org/media-tree.git.
	 It is generally safe to use it for testing a new driver or
	 feature, but its usage on production environments is risky.
	 Don't use it in production. You've been warned.
Created default (all yes) .config file
./scripts/make_myconfig.pl
perl scripts/make_config_compat.pl /lib/modules/4.6.2-1-ARCH/build ./.myconfig ./config-compat.h
creating symbolic links...
ln: missing file operand
Try 'ln --help' for more information.
make[1]: *** [Makefile:261: links] Error 123
make[1]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l'
make: *** [Makefile:26: all] Error 2

JohnnySSH, 2016-07-09:
Any update on this as the TBS drivers are still not building??

JohnnySSH, 2016-07-16:
Well... I found a solution though it wasn't through updating the source but cleaning out the tbs-opensource/ directory then following the instructions here:

https://github.com/ljalves/linux_media/wiki/Installating

then it built without any issues.....