ljalves/linux_media #138: Compilation error

karthikpatkar, 2016-07-14:
Hi,

I am getting following error while compiling.

make[2]: Leaving directory `/usr/src/linux-headers-3.13.0-79-generic'
./scripts/rmmod.pl check
Can't locate Proc/ProcessTable.pm in @INC (you may need to install the Proc::ProcessTable module) (@INC contains: /etc/perl /usr/local/lib/perl/5.18.2 /usr/local/share/perl/5.18.2 /usr/lib/perl5 /usr/share/perl5 /usr/lib/perl/5.18 /usr/share/perl/5.18 /usr/local/lib/site_perl .) at ./scripts/rmmod.pl line 4.
BEGIN failed--compilation aborted at ./scripts/rmmod.pl line 4.
make[1]: *** [default] Error 2
make[1]: Leaving directory `/home/karthik/Documents/dvb-t2/media_build/v4l'
make: *** [all] Error 2

ovidiu31, 2016-07-18:
you need a newer kernel

ljalves, 2016-07-19:
Actually that one is just a missing module:

apt-get install libproc-processtable-perl

Fomtom24, 2024-04-14:
You can mark it as solved.
All I had to do is, a "make clean" and a "make distclean".
Now all is working again.
Thanks for your patience!
Tom