ljalves/linux_media #139: Breaks loading of cx88-dvb driver related to "frame_vector_create" symbol

SimonIremonger, 2016-08-04:
Trying to rebuild/use  media_driver  up-to-date version  --  and a combination of DVB hardware that once used to work together, no longer does (definitely had it all together on an older version of media_driver).

Using e.g. Ubuntu Kernel 4.4.0-31-generic or Debian Kernel 4.6.0-1-amd64 seems to fail similarly with media_driver as of 2016-08-02.

Specifically, The PCI-E DVB-T2 card (requiring the driver), and USB dvb adapters load fine
[modules dvb_core,saa716x_core,saa716x_budget,cxd2820r,dvb_usb_v2]   fine.

HOWEVER, the cx88-dvb (PCI DVB card, very long standing linux support) module can no longer be loaded, some symbol conflict...  dmesg gives:-

"frame_vector: exports duplicate symbol frame_vector_create (owned by kernel)"

trying to load cx88_dvb via insmod then gives a whole load of different 'unknown symbol' errors of course due to dependencies ...

I think this must be a regression or some kernel change that needs importing into this tree?  -- happy to help with some changes/experiments to try to fix this.

With many thanks,

pinkylinc, 2016-08-04:
Hello there,

I'm no expert but I think I can help you with this as I battled with the same issue. I believe the frame_vector_create function used to be included in the module but is now included with the kernel. To solve this I recompiled the driver without the frame_vector module included, but I did also have to manually clean the module out of the modules tree.

To purge it from my tree I manually removed /lib/modules/`uname -r`/kernel/mm/frame_vector.ko
(I also removed by entire media tree at one point but probably just try this first).

I mucked around a lot getting it to compile and work with both my hauppauge dvb-s pci cards and my tbs ones but I think this was my final successful proceedure:

```
git clone https://github.com/ljalves/media_build.git
git clone --depth=1 https://github.com/ljalves/linux_media.git -b latest ./media
cd media_build

make cleanall && make distclean
make dir DIR=../media
make download untar
make prepare
```
(make prepare will generate the .config file for editing - now disable FRAME_VECTOR module in v4l/.config by opening it in a text editor and set to "=y")

`make && make install 
`
(or perhaps also try using the ./build script now and skip the make download untar step, I can't remember which I found worked).

Hope this helps.

SimonIremonger, 2016-08-04:
"make download untar" definitely 'causes' a new problem (downloads some incompatible change).  There seems to be no need to use separate ./build script or doing extra 'clean' steps either.  Key is simply the 'make prepare' and editing v4l.config FRAME_VECTOR=y  before make && make install  ...

Which then (seemingly) 'fixes' the problem, all DVB adapters now startup ok! =).

pinkylinc, 2016-08-04:
Thanks? Sure you're welcome, glad it helped.

In regard to the extra clean steps - the make cleanall can solve problems if patches have been applied to the source tree (when not using a fresh checkout) and it can also help to remove old module configurations. 

The build script is useful for getting the latest build tree and firmware for other (not tbs) media tree drivers but you probably already have this installed thus its not required in your case.

SimonIremonger, 2016-08-05:
OK, on further experimentation (when more awake!!) this does indeed seem to also be working!

Indeed, much thankyou is indeed in order (sorry, I hadn't got to that point previous night etc).

I'm wondering what of the clean steps, build script, "make download untar" etc. should be added to the "installing" wiki for this project?

In any case this tagged issue will hopefully be picked up and fixed in the tree somehow-or-other...  =).]

stuartm, 2016-10-31:
Thank you for this, saved me a lot of time.

The main required step would be running make prepare (to generate the config) and then editing the config. make clean isn't necessary on a new build, only for subsequent builds, and I didn't find the make download step to be necessary either.