ljalves/linux_media #143: TBS6984: TBS 6984 frontend 0 doesn't seem to have a isl6422b on the i2c bus

tonhead, 2016-08-30:
Hi,

My problem seems to be similar in many respects to issue #3 posted by jendik which has now been closed.

I have previously (up until about a week ago) had my TBS6984 v20 card working with a Mythbuntu installation based on Ubuntu 14.04 and Linux Kernel 3.13.0-43-generic. This was using the TBS closed-source drivers.

However, due to a blown PC PSU which corrupted the drive (which then failed to boot) I decided to take the plunge of reinstalling the latest 16.04 based mythbuntu on a SSD and try these open-source drivers. Since doing this I have been unable to get the card tuners working correctly, even after subsequently trying both the TBS official open-source drivers and finally the TBS closed-source drivers.

A symptom reported by these drivers (and incidentally the TBS open source drivers too) is that the frontends do not seem to have a isl6422b on the 12c bus.

Some more info:

dmesg | grep -Ei '(cx|saa|firmware|i2c)'

```
[    0.203265] pci 0000:00:00.0: [Firmware Bug]: reg 0x1c: invalid BAR (can't size)
[    2.489263] i2c /dev entries driver
               	026a8a36fed7af47bc7acbbebf0f3174024f6ccf stv0910: Fix gate_ctrl exit path on i2c errors
[   14.281303] DVB: registering new adapter (SAA716x dvb adapter)
[   17.896098] cx24117 1-0055: creating new instance
[   17.896108] i2c i2c-1: cx24117: Attaching frontend 0
[   17.907202] SAA716x Budget 0000:02:00.0: TurboSight TBS 6984 frontend 0 doesn't seem to have a isl6422b on the i2c bus.
[   17.907940] SAA716x Budget 0000:02:00.0: DVB: registering adapter 0 frontend 0 (Conexant CX24117/CX24132)...
[   17.910095] DVB: registering new adapter (SAA716x dvb adapter)
[   17.910695] cx24117 1-0055: attaching existing instance
[   17.910701] i2c i2c-1: cx24117: Attaching frontend 1
[   17.921009] SAA716x Budget 0000:02:00.0: TurboSight TBS 6984 frontend 1 doesn't seem to have a isl6422b on the i2c bus.
[   17.921810] SAA716x Budget 0000:02:00.0: DVB: registering adapter 1 frontend 0 (Conexant CX24117/CX24132)...
[   17.922101] DVB: registering new adapter (SAA716x dvb adapter)
[   17.922779] cx24117 0-0005: creating new instance
[   17.922785] i2c i2c-0: cx24117: Attaching frontend 0
[   17.933107] SAA716x Budget 0000:02:00.0: TurboSight TBS 6984 frontend 2 doesn't seem to have a isl6422b on the i2c bus.
[   17.933982] SAA716x Budget 0000:02:00.0: DVB: registering adapter 2 frontend 0 (Conexant CX24117/CX24132)...
[   17.934272] DVB: registering new adapter (SAA716x dvb adapter)
[   17.934999] cx24117 0-0005: attaching existing instance
[   17.935004] i2c i2c-0: cx24117: Attaching frontend 1
[   17.945679] SAA716x Budget 0000:02:00.0: TurboSight TBS 6984 frontend 3 doesn't seem to have a isl6422b on the i2c bus.
[   17.955782] SAA716x Budget 0000:02:00.0: DVB: registering adapter 3 frontend 0 (Conexant CX24117/CX24132)...
[   20.791188] i2c i2c-1: cx24117_load_firmware: FW version 1.44.95.2
[   20.791205] i2c i2c-1: cx24117_firmware_ondemand: Firmware upload complete
[   25.853992] i2c i2c-0: cx24117_load_firmware: FW version 1.44.95.2
[   25.854009] i2c i2c-0: cx24117_firmware_ondemand: Firmware upload complete

```

uname -a:

`Linux MythPVR 4.4.0-34-generic #53-Ubuntu SMP Wed Jul 27 16:06:39 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
`

I have created 3 separate config files with the same content:

cat /etc/modprobe.d/options-dvb.conf
cat /etc/modprobe.d/saa716x_budget.conf
cat /etc/modprobe.d/tbs.conf

```
#original drivers
options saa716x_tbs-dvb int_type=1

#alternative drivers
options saa716x_budget int_type=1

```

lspci | grep -i saa

`02:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 02)`

sudo lsmod:
[... 133 lines cut ...]
rmmod: ERROR: Module saa716x_core is in use by: saa716x_budget
/sbin/rmmod dvb_core
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module dvb_core is in use by: saa716x_core saa716x_budget
/sbin/rmmod media
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module media is in use by: dvb_core
/sbin/rmmod cx24117
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module cx24117 is in use by: saa716x_budget
/sbin/rmmod tas2101
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
rmmod: ERROR: Module tas2101 is in use by: saa716x_budget
Couldn't unload: tas2101 cx24117 media dvb_core saa716x_core saa716x_budget dvb_core
make[1]: Leaving directory '/home/glenn/Desktop/media_build/v4l'

```

Any help or guidance you can provide is really appreciated!

Many Thanks in advance.

ljalves, 2016-08-30:
This message "TurboSight TBS 6984 frontend 0 doesn't seem to have a isl6422b on the i2c bus." is not an error or issue.

There are at least 2 hardware versions of the TBS6984.
One with the isl6422b and another without.
That message just informs that yours does't have it...

tonhead, 2016-08-31:
Thanks for the quick reply. 

Is there any particular further log information I can provide then to help establish the cause of my non-working adapters? MythTV reports an error with the card so I am unable to scan for channels.

Thanks.

ljalves, 2016-08-31:
"Pulseaudio" is blocking rmmod to succeed.
If the modprobe doesn't produce output in kern.log or dmesg then ir means that the module was still loaded.
Stop pulseaudio service and retry rmmod, or just do a reboot
If after a reboot it still doesn't work, paste dmesg (better to use a paste service - like pastebin.com)

tonhead, 2016-08-31:
Thanks for the advice!

I finally had to completely uninstall pulseaudio after I still couldn't get rid of a Pulseaudio rmodule reference with a blank UUID!?

Anyway after doing this I was finally able to run sudo make rmmod twice to remove all of the modules, followed by the modprobe commands.

The relevant output from dmesg is:-

```
[  308.539432] WARNING: You are using an experimental version of the media stack.
               	As the driver is backported to an older kernel, it doesn't offer
               	enough quality for its usage in production.
               	Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
               	026a8a36fed7af47bc7acbbebf0f3174024f6ccf stv0910: Fix gate_ctrl exit path on i2c errors
[  308.564614] DVB: registering new adapter (SAA716x dvb adapter)
[  312.068071] cx24117 1-0055: creating new instance
[  312.068089] i2c i2c-1: cx24117: Attaching frontend 0
[  312.068109] isl6422_write: write reg 45
[  312.073315] isl6422_write: I/O error <-5>
[  312.073320] isl6422_set_current: I/O error <-5>
[  312.073334] SAA716x Budget 0000:02:00.0: TurboSight TBS 6984 frontend 0 doesn't seem to have a isl6422b on the i2c bus.
[  312.073345] SAA716x Budget 0000:02:00.0: DVB: registering adapter 0 frontend 0 (Conexant CX24117/CX24132)...

```

Frontends 1 - 3 had similar output.

This looks very similar to the errors hrodenburg posted on 13th Jan 2014 for issue #3 

Full dmesg output at http://pastebin.com/AvyfAK4s 

Thanks for your continued help and time!

ljalves, 2016-08-31:
Forget about those errors - It's a non-issue.

Can you provide dmesg after trying to play a channel?
(I want to see the firmware loading messages)

tonhead, 2016-09-01:
I'll give this a go tomorrow night and post the results thanks.

tonhead, 2016-09-01:
I don't have any channels to play yet as I unable to scan for them in MythTV due to the non-working adapters.

I therefore tried a channel scan using:

scan -x0 /usr/share/dvb/dvb-s/Astra-28.2E | tee channels.conf

with dmesg output:

```
[  336.842344] i2c i2c-1: cx24117_load_firmware: FW version 1.44.95.2
[  336.842370] i2c i2c-1: cx24117_firmware_ondemand: Firmware upload complete

```
on then reinstalling mythtv and all of its dependencies I received a very slightly different dmesg output:

```
[ 1173.193961] i2c i2c-0: cx24117_load_firmware: FW version 1.44.95.2
[ 1173.193976] i2c i2c-0: cx24117_firmware_ondemand: Firmware upload complete

```

The downside to this is that I am again now unable to remove the modules due to their inter-dependencies, including the UUID-less Pulseaudio module:

```
scripts/rmmod.pl unload
found 649 modules
/sbin/rmmod dvb_core
Pulseaudio is running with UUID(s):
rmmod: ERROR: Module dvb_core is in use by: saa716x_core saa716x_budget
/sbin/rmmod saa716x_budget
...
...

```
Is this of any use in understanding my problem?

Many Thanks!

pjacobs-eu, 2024-10-18:
> This message "TurboSight TBS 6984 frontend 0 doesn't seem to have a isl6422b on the i2c bus." is not an error or issue.
> 
> There are at least 2 hardware versions of the TBS6984. One with the isl6422b and another without. That message just informs that yours does't have it...

Due to a change from Kernel 6.5, this message also occurs with cards that have the isl6422b chip. Without the chip the signal is, of course, not usable (only a few muxes, in my case).

After a small edit, it works again.
See: https://github.com/tbsdtv/linux_media/pull/361