ljalves/linux_media #144: No support for TBS6281SE

speedyrazor, 2016-09-03:
Hi, I have been using the drivers for about 1 year with a TBS6984 DVB-S2 card with great success,. I just also got a TBS6281SE DVB-T/C card and am using that alongside the TBS6984, cleaned out all drivers and done a fresh install, but the TBS6281SE is not being picked up, only the TBS6984. If I use the official drivers then both cards drivers install correctly, but would like to use these open source ones. Is there anything I need to do to get both cards working alongside each other?

Kind regards.

lukastribus, 2017-03-31:
Looks like its there since commit 989e2fe3c894330ede73c44f98e431c68a0f2824 ?

speedyrazor, 2017-04-01:
That's good news, thanks. But how do you specify if you want the card to function in DVB-T or DVB-C ?

lukastribus, 2017-04-01:
That's the applications job, not the drivers, The driver will provide access to both technologies, it is the dvb-application that picks adapter and technology.

speedyrazor, 2017-04-01:
Hmmm, that is not how the official drivers work, as the official drivers are installed with a specific command, and will either run in DVB-T or DVB-C. Changing from DVB-T to DVB-C or visa-versa requires a driver reinstall.

lukastribus, 2017-04-15:
Yeah well, this isn't the official driver, is it?

ObviusOwl, 2017-04-28:
Does the driver work for you? I'm loading the driver with `modprobe tbsecp3` but the driver doesn't seem to pick up the card. There are no nodes in `/dev/dvb` created. My kernel is 4.4.0-75-lowlatency on ubuntu 14.04 (mint 17).

lukastribus, 2017-04-28:
No, I got the TBS6281 non-SE.

You do need at least a 4.6 kernel, and the proper firmware files (check dmesg).

crazycat69, 2017-04-29:
You can reduce Cxd2820r requipment to 4.4 (this really work). See v4l/version.txt in media_build.For <4.4 kernels possible remove cxd2820r from saa7160 adapter driver depends. See my media_build and and lijux_media fork.