ljalves/linux_media #146: TBS6985 no frontend on adapter0 and adapter1

wociscz, 2016-09-07:
Hi, i have similar issue like https://github.com/ljalves/linux_media/issues/107

Modules built ok (followed local wiki howto), rebooted and there are frontends only at adapter2&3
Actual setup:
ubuntu 16.04, kernel 4.4.0-36 (ubuntu), mini-itx ipx7-ion (atom 330 with nvidia chips all around)
Tried kernel 4.6 - same behaviour, 4.8 - modules wont build.

Didn't find solution (clean install not tried)
One note - before tbs6985, i have tbs6980 in same machine (works fine both tuners)

Adapter2&3 works fine, but 4 adapters will be better :-)

From kern.log i2c errors:
```
Sep  7 00:34:08 tvheadend kernel: [    6.954281] i2c i2c-9: tas2101: Error attaching frontend
Sep  7 00:34:08 tvheadend kernel: [    6.955124] SAA716x Budget 0000:02:00.0: TurboSight TBS 6985 frontend 0 attach failed
Sep  7 00:34:08 tvheadend kernel: [    6.959258] DVB: registering new adapter (SAA716x dvb adapter)
Sep  7 00:34:08 tvheadend kernel: [    6.965143] i2c i2c-9: Added multiplexed i2c bus 10
Sep  7 00:34:08 tvheadend kernel: [    6.969570] i2c i2c-9: Added multiplexed i2c bus 11
Sep  7 00:34:08 tvheadend kernel: [    7.265296] i2c i2c-9: tas2101: i2c rdm err(-5) @0x00 (len=2)
Sep  7 00:34:08 tvheadend kernel: [    7.267212] i2c i2c-9: tas2101: Error attaching frontend
Sep  7 00:34:08 tvheadend kernel: [    7.268206] SAA716x Budget 0000:02:00.0: TurboSight TBS 6985 frontend 1 attach failed
Sep  7 00:34:08 tvheadend kernel: [    7.271729] DVB: registering new adapter (SAA716x dvb adapter)
Sep  7 00:34:08 tvheadend kernel: [    7.274283] i2c i2c-8: Added multiplexed i2c bus 10
Sep  7 00:34:08 tvheadend kernel: [    7.274686] i2c i2c-8: Added multiplexed i2c bus 11
```

and kern.log sysfs errors:
```
Sep  7 00:34:08 tvheadend kernel: [    7.571707] sysfs: cannot create duplicate filename '/devices/pci0000:00/0000:00:0c.0/0000:02:00.0/i2c-8/channel-0'
Sep  7 00:34:08 tvheadend kernel: [    7.572497] can't create symlink for channel 0
Sep  7 00:34:08 tvheadend kernel: [    7.575607] sysfs: cannot create duplicate filename '/devices/pci0000:00/0000:00:0c.0/0000:02:00.0/i2c-8/channel-1'
Sep  7 00:34:08 tvheadend kernel: [    7.576544] can't create symlink for channel 1
```

Full kern.log from last boot:
http://pastebin.com/r61R6tyj

crazycat69, 2016-09-07:
i think need disable using I2C_MUX adapter inside tas2101 driver.

triagony, 2017-01-18:
I have the exat same issue now (ubuntu, kernel 4.4.0-59) with TBS6982. Only 1 adapter out of two is registered:

dmesg -H | grep -i dvb
```
[  +0,018591] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[  +0,000013] SAA716x Budget 0000:01:00.0: DVB: registering adapter 0 frontend 0 (Tmax TAS2101)...
[  +0,000411] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[  +4,288957] SAA716x Budget 0000:01:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
```

ronny@vivian:~$ dmesg -H | grep -i i2c
```
[  +0,000006] i2c /dev entries driver
[  +0,000256] i2c i2c-6: Added multiplexed i2c bus 8
[  +0,000042] i2c i2c-6: Added multiplexed i2c bus 9
[  +0,188307] i2c i2c-9: av201x: Airoha Technology AV201x successfully attached
[  +0,000982] i2c i2c-7: Added multiplexed i2c bus 10
[  +0,000240] i2c i2c-7: Added multiplexed i2c bus 11
[  +0,093898] i2c i2c-7: tas2101: i2c rdm err(-5) @0x00 (len=2)
[  +0,000796] i2c i2c-7: tas2101: Error attaching frontend
```

dmesg -H | grep -i budget
```
[  +0,002152] SAA716x Budget 0000:01:00.0: enabling device (0000 -> 0002)
[  +0,000013] SAA716x Budget 0000:01:00.0: DVB: registering adapter 0 frontend 0 (Tmax TAS2101)...
[  +0,002160] SAA716x Budget 0000:01:00.0: TurboSight TBS 6982SE frontend 1 attach failed
[  +4,288957] SAA716x Budget 0000:01:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)

```

Is there any known solution to this?

Thanks in advance!

-triagony

pbjudby, 2018-03-04:
Well - I cannot get my TBS6985 on Ubuntu 16.04 will not load - Keeps failing like this in dmesg: 
`
[    3.848165] tbs6290fe: module license 'TurboSight Proprietary' taints kernel.
[    3.848167] Disabling lock debugging due to kernel taint
[    3.849483] snd_hda_intel 0000:00:1b.0: enabling device (0000 -> 0002)
[    3.850862] IR JVC protocol handler initialized
[    3.853542] IR Sony protocol handler initialized
[    3.856048] saa716x_core: disagrees about version of symbol dvb_dmxdev_init
[    3.856052] saa716x_core: Unknown symbol dvb_dmxdev_init (err -22)
[    3.856077] saa716x_core: disagrees about version of symbol dvb_register_adapter
[    3.856079] saa716x_core: Unknown symbol dvb_register_adapter (err -22)
[    3.856115] saa716x_core: disagrees about version of symbol dvb_dmx_release
[    3.856117] saa716x_core: Unknown symbol dvb_dmx_release (err -22)
[    3.857413] saa716x_core: disagrees about version of symbol dvb_net_init
[    3.857416] saa716x_core: Unknown symbol dvb_net_init (err -22)
[    3.857421] saa716x_core: disagrees about version of symbol dvb_dmxdev_release
[    3.857422] saa716x_core: Unknown symbol dvb_dmxdev_release (err -22)
[    3.857429] saa716x_core: disagrees about version of symbol dvb_frontend_detach
[    3.857430] saa716x_core: Unknown symbol dvb_frontend_detach (err -22)
[    3.857434] saa716x_core: disagrees about version of symbol dvb_net_release
[    3.857435] saa716x_core: Unknown symbol dvb_net_release (err -22)
[    3.857440] saa716x_core: disagrees about version of symbol dvb_unregister_frontend
[    3.857441] saa716x_core: Unknown symbol dvb_unregister_frontend (err -22)
[    3.857447] saa716x_core: disagrees about version of symbol dvb_register_frontend
[    3.857448] saa716x_core: Unknown symbol dvb_register_frontend (err -22)
[    3.857452] saa716x_core: disagrees about version of symbol dvb_unregister_adapter
[    3.857453] saa716x_core: Unknown symbol dvb_unregister_adapter (err -22)
[    3.857456] saa716x_core: disagrees about version of symbol dvb_dmx_init
[    3.857457] saa716x_core: Unknown symbol dvb_dmx_init (err -22)

`
Regarding using the TBS or opensource drivers. Tried severalt times now - but ending up with the same error message each time