ljalves/linux_media #147: Mix cards TBSECP3 frontend 0 attach failed

nightshadow36, 2016-09-12:
Hello.
I have just upgraded with 2 x 6909 and 2 x 6905
With the 6905 I get TBSECP3 frontend 0 attach failed - TBSECP3 frontend 1 attach failed
I tried with a spare 6905 card and same error so i know the card is not faulty.
Please see screen shot, any idea how to fix?
Thank you.
[image]

crazycat69, 2016-09-12:
Please attach full dmesg output

nightshadow36, 2016-09-12:
@crazycat69 
Here is the log I hope its the correct one.
[dmesg.tar.gz](https://github.com/ljalves/linux_media/files/467936/dmesg.tar.gz)

steven1758, 2016-09-18:
from the log it is after load the firmware still need some time to init for 6909 need more time to load the tuner ,please wait more time to check whether is ok