ljalves/linux_media #149: Unable to build in Kernel 4.8.2 "incomplete type   struct dma_attrs attrs"

JohnnySSH, 2016-10-19:
Getting this error, standard build process:

/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:26:19: error: field 'attrs' has incomplete type
  struct dma_attrs attrs;
                   ^~~~~
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:35:20: error: field 'attrs' has incomplete type
  struct dma_attrs  attrs;
                    ^~~~~
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c: In function 'vb2_dc_alloc':
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:163:7: error: implicit declaration of function 'dma_get_attr' [-Werror=implicit-function-declaration]
  if (!dma_get_attr(DMA_ATTR_NO_KERNEL_MAPPING, &buf->attrs))
       ^~~~~~~~~~~~
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c: In function 'vb2_dc_put_userptr':
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:429:3: error: implicit declaration of function 'DEFINE_DMA_ATTRS' [-Werror=implicit-function-declaration]
   DEFINE_DMA_ATTRS(attrs);
   ^~~~~~~~~~~~~~~~
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:429:20: error: 'attrs' undeclared (first use in this function)
   DEFINE_DMA_ATTRS(attrs);
                    ^~~~~
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:429:20: note: each undeclared identifier is reported only once for each function it appears in
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:431:3: error: implicit declaration of function 'dma_set_attr' [-Werror=implicit-function-declaration]
   dma_set_attr(DMA_ATTR_SKIP_CPU_SYNC, &attrs);
   ^~~~~~~~~~~~
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c: In function 'vb2_dc_get_userptr':
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:493:19: error: 'attrs' undeclared (first use in this function)
  DEFINE_DMA_ATTRS(attrs);
                   ^~~~~
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c: In function 'vb2_dma_contig_init_ctx_attrs':
/home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.c:743:17: error: dereferencing pointer to incomplete type 'struct dma_attrs'
   conf->attrs = *attrs;
                 ^~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:296: /home/htpc/Downloads/tbs-opensource/media_build/v4l/videobuf2-dma-contig.o] Error 1
make[2]: *** [Makefile:1471: _module_/home/htpc/Downloads/tbs-opensource/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/4.8.2-1-ARCH/build'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/htpc/Downloads/tbs-opensource/media_build/v4l'
make: *** [Makefile:26: all] Error 2

davidrnewman, 2016-10-26:
I tried editing videobuf2-dma-config.c to include dma-attrs.h but it gets overwritten on every make.

Xxyzzy, 2016-10-30:
Patch for kernel 4.8.*

/Håkan

[diff-4.8.X.txt](https://github.com/ljalves/linux_media/files/560809/diff-4.8.X.txt)

davidrnewman, 2016-10-31:
Is that a patch to apply in media_build, media or the linux headers on my machine?

Xxyzzy, 2016-10-31:
Changes in 3 files under the media dir, and one file under media_build.

davidrnewman, 2016-10-31:
patch -p0 -i diff-4.8.C.txt doesn't work ("patch: **** Only garbage was found in the patch input."). Need to generate the diff using -c or -u as well as -r, as the normal output doesn't list the filenames on lines 2 and 3.

LubosD, 2016-11-02:
I've manually reworked the patch, here it is:
[media_build-linux-4.8.x.txt](https://github.com/ljalves/linux_media/files/566109/media_build-linux-4.8.x.txt)
[media-linux-4.8.x.txt](https://github.com/ljalves/linux_media/files/566110/media-linux-4.8.x.txt)

thecosmicslug, 2016-11-04:
I am having this problem too, using ubuntu 16.10...
I am not too sure how to apply this patch, can someone explain it to me?

thecosmicslug, 2016-11-08:
Nobody can tell me how/where to apply the above patch file? I really didnt want to downgrade back to kernel 4.4

LubosD, 2016-11-08:
Just learn to use the `patch` command (e.g. `patch -p1 < thepatchfile`).

davidrnewman, 2016-11-08:
It took me some time to sort out where to put the patch files. One goes in the media_build directory the other in the media directory. Then 

patch -p1 -i media_build-linux-4.8.x.txt in the media_build directory.

The patch source actually compiles, but it isn't sensitive enough to pick up channels in a DTV T2 scan where I live. I think the closed source version actually got the auto gain control working.

thecosmicslug, 2016-11-11:
Thanks david, I never got past compile errors even with the patch, but luckily the closed-source version provided by TBS has been updated for kernel 4.8 so I'll be using that.

davidrnewman, 2016-11-11:
Note that there is a fork of this open source driver. One of the authors there works for TBS.

https://github.com/tbsdtv/linux_media/wiki

ljalves, 2017-01-11:
I've just updated the media tree and media_build with the latest official linux media sources.
Should build fine now for latest kernels

closed 2017-01-11 by ljalves