ljalves/linux_media #150: saa716x_budget.ko not created with kernel 4.4.0-59 (ubuntu)?

triagony, 2017-01-17:
Hello everybody,

today I tried to update the driver for my tbs-card 6982 with this instructions https://github.com/ljalves/linux_media/wiki/Installating but the saa716x_budget.ko seems not to be created anymore (other than with previous kernel versions). I also tried a fresh install after removing the old files from my system. So (obviously) I get the following error when I try to load the module:

`sudo modprobe saa716x_budget int_type=1`
`modprobe: FATAL: Module saa716x_budget not found in directory /lib/modules/4.4.0-59-generic`

As you can see, in previous kernel versions the module has been installed properly: 

`ronny@vivian:~$ locate *.ko |grep saa716x_budget`
`[...]`
`/lib/modules/3.19.0-56-generic/kernel/drivers/media/pci/saa716x/saa716x_budget.ko`
`/lib/modules/3.19.0-59-generic/kernel/drivers/media/pci/saa716x/saa716x_budget.ko`
`/lib/modules/4.4.0-51-generic/kernel/drivers/media/pci/saa716x/saa716x_budget.ko`
`/lib/modules/4.4.0-57-generic/kernel/drivers/media/pci/saa716x/saa716x_budget.ko`
`ronny@vivian:~$ `

My system is:

`ronny@vivian:~$ lsb_release -a`
`No LSB modules are available.`
`Distributor ID:	Ubuntu`
`Description:	Ubuntu 16.04.1 LTS`
`Release:	16.04`
`Codename:	xenial`
`ronny@vivian:~$`

`ronny@vivian:~$ uname -a`
`Linux vivian 4.4.0-59-generic #80-Ubuntu SMP Fri Jan 6 17:47:47 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux`
`ronny@vivian:~$ `

`ronny@vivian:/$ sudo lspci -v | grep --after-context=10 7160`
`01:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)`
	`Subsystem: Device 6982:0002`
	`Flags: fast devsel, IRQ 11`
	`Memory at f7d00000 (64-bit, non-prefetchable) [disabled] [size=1M]`
	`Capabilities: [40] MSI: Enable- Count=1/32 Maskable- 64bit+`
	`Capabilities: [50] Express Endpoint, MSI 00`
	`Capabilities: [74] Power Management version 2`
	`Capabilities: [80] Vendor Specific Information: Len=50 <?>`
	`Capabilities: [100] Vendor Specific Information: ID=0000 Rev=0 Len=088 <?>`

Has anybody an idea how to fix this?

Thanks in advance!

Triagony

crazycat69, 2017-01-18:
https://bitbucket.org/CrazyCat/media_build/commits/cc56f0830a0e1ca91cd930d5a9b6fe7b2486db0e

triagony, 2017-01-18:
Thank you very much! Problem solved!

closed 2017-01-18 by triagony

dafrasaga, 2017-01-26:
Hi All,
I have slackware64 3.10.17  kernel and I have saa716x_budget.ko not created issue too.
I have tbs6281 and I done 

git clone https://github.com/ljalves/media_build.git
 git clone --depth=1 https://github.com/ljalves/linux_media.git -b latest ./media
 cd media_build
 make dir DIR=../media
 make distclean
 make
 sudo make install

.. and the solution in [https://bitbucket.org/CrazyCat/media_build/commits/cc56f0830a0e1ca91cd930d5a9b6fe7b2486db0e](url) has not worked.

I'm very noob into linux.. Any hints ??

Cheers
Gabriele

crazycat69, 2017-01-26:
In this repo required 4.6+ kernel for saa716x module. Depend from cxd2820r driver, which use regmap API.P.S. "noobs" must use *buntu LTS with latest 4.4 lts kernel :-) 18:41, 26 января 2017 г., dafrasaga <notifications@github.com>:I have slackware64 3.10.17  kernel and I have saa716x_budget.ko not created issue too.I have tbs6281 and I donegit clone https://github.com/ljalves/media_build.gitgit clone --depth=1 https://github.com/ljalves/linux_media.git -b latest ./mediacd media_buildmake dir DIR=../mediamake distcleanmakesudo make install.. and the solution in https://bitbucket.org/CrazyCat/media_build/commits/cc56f0830a0e1ca91cd930d5a9b6fe7b2486db0e has not worked.I'm very noob into linux.. Any hints ??

dafrasaga, 2017-01-27:
Hi crazycat69,
excuse me for the question.. only to understand..
If I find cxd2820r files in 3.10 release  [..](https://git.kernel.org/cgit/linux/kernel/git/stable/linux-stable.git/tree/drivers/media/dvb-frontends?id=refs/tags/v3.10.104)  it is not suffice, isn't it?
We do need 4.4+ release which has more code. Is it right?

Cheers
Gabriele